rbonghi/jetson_stats #144: jtop only works with sudo jetpack 4.5.1 xavier NX

issamsaid, 2021-03-11:
**Describe the bug**
jtop does not work unless used with sudo on the latest jetpack version

**To Reproduce**
Steps to reproduce the behavior:
1. install 4.5.1 on xavier NX
2. IMPORTANT - copy the rootfs to nvme (the board is booting from nvme)
3. install jteston-stats
4. try jtop 

**Screenshots**
- when running without sudo :
you can see the ncurse window created for a second and then 
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/jtop/__main__.py", line 148, in main
    curses.wrapper(JTOPGUI, jetson, [ALL, GPU, CPU, MEM, CTRL, INFO], init_page=args.page, loop=args.loop, seconds=LOOP_SECONDS)
  File "/usr/lib/python2.7/curses/wrapper.py", line 43, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python2.7/dist-packages/jtop/gui/jtopgui.py", line 109, in __init__
    self.run(loop, seconds)
  File "/usr/local/lib/python2.7/dist-packages/jtop/gui/jtopgui.py", line 136, in run
    while not self.events() and self.jetson.ok(spin=True):
  File "/usr/local/lib/python2.7/dist-packages/jtop/jtop.py", line 1056, in ok
    raise ex_value
IndexError: list index out of range

- with sudo it works fine

PLEASE NOTE THAT the board had been rebooted many times after the installation.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

**Board**
 - jetson xavier NX

**Jetpack**
 - Jetpack: 4.5.1
 - L4T:  32.5.1

**Jetson-Stats**
 - Version: 3.1.0

issamsaid, 2021-03-11:
and sometimes it returns:
jtop
I can't access to jetson_stats.service.
Please logout or reboot this board.

NOTE that the board have been rebooted multiple times.

kouta-kun, 2021-03-22:
+1 on a Jetson Nano w/ Jetpack 4.5.1

rbonghi, 2021-03-22:
I will work on it, soon as possible

kouta-kun, 2021-03-22:
Forcing reinstall jetson_stats seems to fix it for a couple of reboots on my devices

issamsaid, 2021-03-22:
> Forcing reinstall jetson_stats seems to fix it for a couple of reboots on my devices

It did not solve it on my end.

issamsaid, 2021-03-29:
I found the problem: when parsing for JETSON env in  `/usr/local/lib/python3.6/dist-packages/jtop/core/common.py` sometimes the entries in the env files has tokens that expend into multiple lines and therefor the `value = tup[1].strip()` might fail. The fix would be to only check for `tup[1]` when `tup[0]` is related to `JETSON env`:
```
def import_os_variables(SOURCE, PATTERN):
    if os.path.isfile(SOURCE):
        logger.debug("Open source file {source}".format(source=SOURCE))
        proc = sp.Popen(['bash', '-c', 'source {source} && env'.format(source=SOURCE)], stdout=sp.PIPE, stderr=sp.PIPE)
        # Load variables
        source_env = {}
        for tup in map(lambda s: s.decode("utf-8").strip().split('=', 1), proc.stdout):
            name = tup[0].strip()
            if PATTERN in name:
                value = tup[1].strip()
                source_env[name] = value
        return source_env
    else:
        logger.error("File does not exist")
        return {}
```

zinwalin, 2021-04-07:
same here. 

I can't access to jetson_stats.service.
Please logout or reboot this board.

Jetpack 4.5

FreemanX, 2021-07-19:
Same here with jetson nano 4GB, Jetpack 4.5.1

Tiryoh, 2021-11-04:
> I found the problem: when parsing for JETSON env in `/usr/local/lib/python3.6/dist-packages/jtop/core/common.py` sometimes the entries in the env files has tokens that expend into multiple lines and [...]

This hotfix worked on my Jetson Nano + JetPack 4.5.1. Thanks @issamsaid!

huajng, 2022-04-22:
> 
if i get I can't access to jetson_stats.service.
Please logout or reboot this board.

and then use 
sudo jtop
it can run for me

rbonghi, 2022-06-11:
Lat reply, but I will fix in the next release

rbonghi, 2023-01-04:
Is this all fixed with this new release candidate?
If you don't read any new error, I will close this issue

```
sudo -H pip3 install jetson-stats==4.0.0rc3
````

rbonghi, 2023-01-05:
New release **4.0**
```
sudo -H pip install -U jetson-stats
```

closed 2023-01-05 by rbonghi

f3rm4rf3r, 2023-08-03:
Hello!
We are using `jtop` together with `netdata` to pull statistics about the temperature/current/voltage, etc...
We saw this problem too and it worked when we run `sudo jtop`.
The issues was that the `netdata` user didn't have access to the jtop socket in `/run/jtop.sock`.
We fixed this by adding the user to the `jtop` group with:
```
sudo usermod -a -G jtop netdata
```
Replace `netdata` for the user you want to give access to `jtop`.