rbonghi/jetson_stats #147: Cannot install with "sudo -H pip install -U jetson-stats"

SteffenEpp, 2021-04-27:
I get the following error when trying to:

WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f8ed62cc0>: Failed to establish a new connection: [Errno -2] Name or service not known',)': /simple/jetson-stats/

Can somebody help me with that?

SteffenEpp, 2021-04-27:
The full message is:

WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f8ae719b0>: Failed to establish a new connection: [Errno -2] Name or service not known',)': /simple/jetson-stats/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f8ae71a90>: Failed to establish a new connection: [Errno -2] Name or service not known',)': /simple/jetson-stats/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f8ae719e8>: Failed to establish a new connection: [Errno -2] Name or service not known',)': /simple/jetson-stats/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f8ae71780>: Failed to establish a new connection: [Errno -2] Name or service not known',)': /simple/jetson-stats/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f8ae713c8>: Failed to establish a new connection: [Errno -2] Name or service not known',)': /simple/jetson-stats/
ERROR: Could not find a version that satisfies the requirement jetson-stats (from versions: none)
ERROR: No matching distribution found for jetson-stats

huajng, 2022-04-22:
I use sudo python3 -m pip install jetson-stats -i http://pypi.douban.com/simple/ --trusted-host pypi.douban.com
but I meet another issue "ModuleNotFoundError: No module named '_curses'"

rbonghi, 2023-01-05:
Hi @SteffenEpp 
I have never seen your error, but looking online seems like a user error, not a jetson_stats error. Try this link to fix 
https://stackoverflow.com/questions/52815784/python-pip-raising-newconnectionerror-while-installing-libraries

about @huajng 
Are you working with which NVIDIA Jetson and which Jetpack?

rbonghi, 2023-01-05:
This error should now be fixed with the latest release **4.0** try to install and check again
```
sudo -H pip install -U jetson-stats
```

closed 2023-01-05 by rbonghi