rbonghi/jetson_stats #148: This mode  does not exists

paulds8, 2021-04-29:
**Describe the bug**
Unable to run jtop.

All I get is "This mode  does not exists" which seems like it's not picking up a mode at all.

**To Reproduce**
Try launch jtop.

**Expected behavior**
jtop should start running.

**Additional context**

 - NVIDIA Jetson TX2
   * Jetpack 4.4.1 [L4T 32.4.4]
   * NV Power Mode: NVPM - Type: WARN:
   * jetson_stats.service: active
 - Board info:
   * Type: TX2
   * SOC Family: tegra186 - ID:24
   * Module: P3310-1000 - Board: UNKNOWN
   * Code Name: quill
   * Boardids: 3310:0000:A0
   * CUDA GPU architecture (ARCH_BIN): 6.2
   * Serial Number: _redacted_
 - Libraries:
   * CUDA: 10.2.89
   * cuDNN: 8.0.0.180
   * TensorRT: 7.1.3.0
   * Visionworks: 1.6.0.501
   * OpenCV: 4.1.1 compiled CUDA: NO
   * VPI: 4.4.1-b50
   * Vulkan: NOT_INSTALLED
 - jetson-stats:
   * Version 3.1.0
   * Works on Python 3.6.9

paulds8, 2021-04-29:
Never mind. I explicitly set the mode with `sudo nvpmodel -m 3` and now it's working after a reboot. Perhaps a clearer error message would help others in future.

neilyoung, 2021-09-22:
@paulds8  Thanks for sharing the solution

leonpano2006, 2022-04-09:
> @paulds8  Thanks for sharing the solution

Fixed? 
Then close this issue as solved

neilyoung, 2022-04-09:
At the time of writing this last year I was satisfied with the solution suggested by @paulds8