rbonghi/jetson_stats #152: Exception when NVJPG is active

astar26, 2021-06-08:
**Describe the bug**
Access to jtop.stats raises an exception when nvjpeg is active. Does not reproduce in GUI.
Occurs on AGX Xavier.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a jtop object and start it.
2. Run the following command to activate nvjpeg:
`gst-launch-1.0 videotestsrc ! nvvidconv ! nvjpegenc ! fakesink`
3. Attempt access to jtop.stats
4. See error

`j = jtop.jtop()
j.start()
j.stats
`

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Expected behavior**
Return of current stats.

**Additional context**
Removing the subscript appears to solve the issue, tested inside a docker image so any effect on GUI behavior has not been tested.

**Board**
 - SOC: tegra194
 - ID: 25
 - Board(s): P2822-0000
 - Code Name: [e.g. test]

**Jetpack**
 - Jetpack: 4.4
 - L4T: 32.4.3

**Jetson-Stats**
 - Version: 3.1.0

astar26, 2021-06-08:
Exception is raised from the following line:
https://github.com/rbonghi/jetson_stats/blob/d1a2118cd5742c8b44b471e3000aa8f97a9a6f9e/jtop/jtop.py#L556

Replacing `self.engine.nvjpg['rate']` with `self.engine.nvjpg` appears to solve the issue when used programmatically, however before submitting a fix - this seems to break the current code style and may break GUI behavior - am I missing something?

rbonghi, 2023-01-05:
This error should now be fixed with the latest release **4.0** try to install and check again
```
sudo -H pip install -U jetson-stats
```

closed 2023-01-05 by rbonghi