rbonghi/jetson_stats #154: Selection of the over-current limit on Xavier NX

Myzhar, 2021-06-21:
**Is your feature request related to a problem? Please describe.**
With JP4.5 when the Xavier NX is under heavy load there are a lot of warnings about the "over-current" status.

**Describe the solution you'd like**
The solution proposed on devtalk is to raise the over-current limit from the default value of 3.6A to 5.0A:
https://forums.developer.nvidia.com/t/system-throttled-due-to-over-current/167029/58

Adding this option to `jtop` could be very useful

rbonghi, 2022-06-11:
Do you know for the JP5.0 how to read this data?

I was looking my Xavier NX and I don't see any folder like:
`/sys/devices/c250000.i2c/i2c-7/7-0040/iio\:device0/crit_current_limit_0`

rbonghi, 2023-01-04:
This data is now not available from Jetpack 5.x
I am still investigating where I can find it.

If you know where is located let me know

rbonghi, 2023-03-17:
From jtop 4.2.0 is available the input voltage in page CTRL, area 6

[image]

closed 2023-03-17 by rbonghi