rbonghi/jetson_stats #155: Environment Variables (jetson_variables)

santaimpersonator, 2021-06-22:
**Describe the bug**
Sometimes I have issues with the [environment variables](https://github.com/rbonghi/jetson_stats/wiki/jetson_variables)

**To Reproduce**
Steps to reproduce the behavior:
1. Using the pre-built image ([jetbot-043_nano-4gb-jp45.zip](https://drive.google.com/file/d/1o08RPDRZuDloP_o76tCoSngvq1CVuCDh/view?usp=sharing)) for Jetpack 4.5 on the 4GB board from the [jetbot.org website](http://jetbot.org/master/software_setup/sd_card.html)
2. Install the jetson_stats package (install `pip3` then the package)
3. Call Jetson variables:
    * `export | grep JETSON`
4. No values displayed:
  [image]
  Sometimes, I can get the values to display after a reboot.
  [image]
  Otherwise, I have to run the `jetson_env.sh` script, before trying to retrieve the environment variables:
    * `source /etc/profile.d/jetson_env.sh`

**Jetson-Stats**
 - Version: 3.1.0

rbonghi, 2021-07-05:
Thx,

I notice that error time ago on my platform, but I'm extremely busy in this period.
I hope to fix soon :-)

rbonghi, 2023-01-05:
Install the latest release **4.0** try to install and check again
```
sudo -H pip install -U jetson-stats
```

closed 2023-01-05 by rbonghi