rbonghi/jetson_stats #172: Jtop causes desktop operation jam

carlyangdong, 2021-09-10:
jetpack version: Jetpack4.4/Jetpack4.5.1
HW: Agx Xavier
jtop version: v3.1.1
cmd: sudo jtop

When I open jtop, all operations on the desktop become periodically stuck. moving the mouse, dragging the window and entering commands are not smooth, return to normal after shutdown jtop.
The phenomenon is like executing "watch -n 3 nvpmodel -m 0".
I guess jtop has a thread calling "nvpmodel" periodically, it can cause serious problems.

I'm not sure if everyone has the same problem.

Myzhar, 2021-09-23:
I can confirm this issue. Sometimes it's like if the daemon takes too many resources. If you kill it and restart everything turns back to normal.

rbonghi, 2023-01-04:
Found a bug on nvpmodel access, this issue seems now fixed, if you can try let me know:

```
sudo -H pip3 install jetson-stats==4.0.0rc3
```

rbonghi, 2023-01-05:
This error should now be fixed with the latest release **4.0** try to install and check again
```
sudo -H pip install -U jetson-stats
```

closed 2023-01-05 by rbonghi