rbonghi/jetson_stats #178: jtop connection error if secureboot is enabled on Jetson device?

SharpZKing, 2021-10-13:
Jtop **connection error** if secureboot is enabled on Jetson device?

Describe:
I installed jetson_state with **sudo -H pip install -U jetson-stats**, but got error of "connection error" when run **jtop**.

Log shows **jetson_clocks show infos error** like ** No such file**  to:

/sys/kernel/debug/bpmp/debug/clk/emc/min_rate 
/sys/kernel/debug/bpmp/debug/clk/emc/max_rate
/sys/kernel/debug/bpmp/debug/clk/emc/rate
/sys/kernel/debug/bpmp/debug/clk/emc/mrq_rate

I can only see the **/sys/kernel/debug/bpmp** directory, but not the **/sys/kernel/debug/bpmp/debug** directory.

oldgalileo, 2023-05-23:
@rbonghi is there an update on this and/or a workaround?

rbonghi, 2024-01-09:
Hi @oldgalileo and @SharpZKing 

After a long time, I looked at the new Jetson-Stats and I think it's mostly fixed to work in a secure boot release.
If jtop doesn't can you please share the output from these commands. I think if few iterations will be able to run
```
journalctl -u jtop.service -n 100 --no-pager
```
```
jtop --error-log
```