rbonghi/jetson_stats #183: Error connection on jtop

simue, 2021-11-29:
When executing jtop after a couple of seconds it stops with the only output being ``` Error connection ```

Steps to reproduce the behavior:
1. Connect to Jetson via ssh
2. Run jtop
3. Wait for error message

```
simue@ubuntu:~$ jtop
Error connection
simue@ubuntu:~$
```

**jetson_stats service logs**
```
simue@ubuntu:~$ sudo journalctl -u jetson_stats
[sudo] password for simue:
-- Logs begin at Tue 2021-11-30 01:58:15 CST, end at Tue 2021-11-30 02:16:51 CST. --
Nov 30 01:58:15 ubuntu systemd[1]: Started jetson_stats service.
Nov 30 01:58:16 ubuntu jtop[4905]: [INFO] jtop.core.common - fan loaded on /sys/devices/pwm-fan
Nov 30 01:58:16 ubuntu jtop[4905]: [INFO] jtop.core.common - jetson_clocks loaded on /usr/bin/jetson_clocks
Nov 30 01:58:17 ubuntu jtop[4905]: [INFO] jtop.core.common - tegrastats loaded on /usr/bin/tegrastats
Nov 30 01:58:17 ubuntu jtop[4905]: [INFO] jtop.__main__ - jetson_stats server loaded
Nov 30 01:58:18 ubuntu jtop[4905]: [INFO] jtop.core.fan - Mode set default status=False
Nov 30 01:58:48 ubuntu jtop[4905]: [INFO] jtop.service - tegrastats started 500ms
Nov 30 01:58:51 ubuntu jtop[4905]: [INFO] jtop.service - tegrastats close
Nov 30 01:58:51 ubuntu jtop[4905]: [INFO] jtop.service - jetson_clocks show closed
Nov 30 02:08:38 ubuntu jtop[4905]: [INFO] jtop.service - tegrastats started 500ms
Nov 30 02:08:41 ubuntu jtop[4905]: [INFO] jtop.service - tegrastats close
Nov 30 02:08:41 ubuntu jtop[4905]: [INFO] jtop.service - jetson_clocks show closed
Nov 30 02:09:04 ubuntu jtop[4905]: [INFO] jtop.service - tegrastats started 500ms
Nov 30 02:09:07 ubuntu jtop[4905]: [INFO] jtop.service - tegrastats close
Nov 30 02:09:07 ubuntu jtop[4905]: [INFO] jtop.service - jetson_clocks show closed
Nov 30 02:09:09 ubuntu jtop[4905]: [INFO] jtop.service - tegrastats started 500ms
Nov 30 02:09:12 ubuntu jtop[4905]: [INFO] jtop.service - tegrastats close
Nov 30 02:09:12 ubuntu jtop[4905]: [INFO] jtop.service - jetson_clocks show closed
```

**Expected behavior**
jtop gui showing or at least a more concise error message on what has gone wrong

**Additional context**
I used jetson_config's health scripts to check the jetson_stats installation and it appears to be ok.
I did reinstall jetson_stats multiple times, with corresponding restarts.
The jetson_stats service appears to be running.

```
- NVIDIA Jetson AGX Xavier [16GB]
   * Jetpack 4.6 [L4T 32.6.1]
   * NV Power Mode: MODE_30W_4CORE - Type: 5
   * jetson_stats.service: active
 - Board info:
   * Type: AGX Xavier [16GB]
   * SOC Family: tegra194 - ID:25
   * Module: P2888-0001 - Board: P2822-0000
   * Code Name: galen
   * CUDA GPU architecture (ARCH_BIN): 7.2
   * Serial Number: 1425020081660
 - Libraries:
   * CUDA: NOT_INSTALLED
   * cuDNN: NOT_INSTALLED
   * TensorRT: NOT_INSTALLED
   * Visionworks: NOT_INSTALLED
   * OpenCV: NOT_INSTALLED compiled CUDA: NO
   * VPI: NOT_INSTALLED
   * Vulkan: 1.2.70
 - jetson-stats:
   * Version 3.1.1
   * Works on Python 3.6.9
```

**Jetson-Stats**
 - Version: 3.1.1

simue, 2022-01-24:
Hi @rbonghi, do you have any hint on what this behaviour could be caused by?
I realized that this is only the case when another app I am using is accessing the tegra+nvidia devices.
It is running in a docker container, so maybe that's the issue?

lurenlym, 2022-08-11:
same error. 
`- NVIDIA Jetson Xavier NX (Developer Kit Version)
   * Jetpack UNKNOWN [L4T 34.1.1]
   * jetson_stats.service: active
 - Board info:
   * Type: Xavier NX (Developer Kit Version)
   * SOC Family: tegra194 - ID:25
   * Module: P3668 - Board: P3509-000
   * Code Name: jakku
   * CUDA GPU architecture (ARCH_BIN): 7.2
   * Serial Number: 1425021054005
 - Libraries:
   * CUDA: NOT_INSTALLED
   * cuDNN: 8.3.2.49
   * TensorRT: 8.4.0.11
   * Visionworks: 1.6.0.501
   * OpenCV: 4.1.1 compiled CUDA: NO
   * VPI: 0.3.7
   * Vulkan: 1.1.70
 - jetson-stats:
   * Version 3.1.4
   * Works on Python 3.6.9`

rbonghi, 2023-01-04:
I think this issue is fixed now, if you want to try in preview:

```
sudo -H pip3 install jetson-stats==4.0.0rc3
```

rbonghi, 2023-01-05:
This error should now be fixed with the latest release **4.0** try to install and check again
```
sudo -H pip install -U jetson-stats
```

closed 2023-01-05 by rbonghi