rbonghi/jetson_stats #187: jetson clocks has a wrong state

fayfive, 2021-12-20:
Jetpack 4.6 rev3

```sudo jetson_clocks```
I see ```jtop```, but it's inactive.

Here is tegrastats

Before jetson_clocks
```
RAM 1689/3825MB (lfb 339x4MB) SWAP 0/1913MB (cached 0MB) CPU [3%@345,off,off,0%@345,1%@345,1%@345] EMC_FREQ 0% GR3D_FREQ 0% PLL@50.5C MCPU@50.5C PMIC@50C GPU@50C BCPU@50.5C thermal@50.3C
RAM 1689/3825MB (lfb 339x4MB) SWAP 0/1913MB (cached 0MB) CPU [2%@345,off,off,1%@345,1%@345,0%@345] EMC_FREQ 0% GR3D_FREQ 0% PLL@50.5C MCPU@50.5C PMIC@50C GPU@50C BCPU@50.5C thermal@50.3C
RAM 1689/3825MB (lfb 339x4MB) SWAP 0/1913MB (cached 0MB) CPU [2%@345,off,off,0%@345,0%@345,0%@345] EMC_FREQ 0% GR3D_FREQ 0% PLL@50.5C MCPU@50.5C PMIC@50C GPU@50C BCPU@50.5C thermal@50.3C
RAM 1689/3825MB (lfb 339x4MB) SWAP 0/1913MB (cached 0MB) CPU [2%@345,off,off,0%@345,2%@345,0%@345] EMC_FREQ 0% GR3D_FREQ 0% PLL@50.5C MCPU@50.5C PMIC@50C GPU@50C BCPU@50.5C thermal@50.3C
RAM 1689/3825MB (lfb 339x4MB) SWAP 0/1913MB (cached 0MB) CPU [2%@345,off,off,0%@345,1%@345,0%@345] EMC_FREQ 0% GR3D_FREQ 0% PLL@50.5C MCPU@50.5C PMIC@50C GPU@50C BCPU@50.5C thermal@50.3C
RAM 1689/3825MB (lfb 339x4MB) SWAP 0/1913MB (cached 0MB) CPU [2%@345,off,off,2%@345,1%@345,1%@345] EMC_FREQ 0% GR3D_FREQ 0% PLL@50.5C MCPU@50.5C PMIC@50C GPU@50C BCPU@50.5C thermal@50.3C
RAM 1690/3825MB (lfb 339x4MB) SWAP 0/1913MB (cached 0MB) CPU [11%@2035,off,off,5%@2035,96%@2035,4%@2035] EMC_FREQ 0% GR3D_FREQ 0% PLL@52C MCPU@52C PMIC@50C GPU@50C BCPU@52C thermal@50.9C
RAM 1690/3825MB (lfb 339x4MB) SWAP 0/1913MB (cached 0MB) CPU [2%@2035,off,off,0%@2035,98%@2035,1%@2035] EMC_FREQ 0% GR3D_FREQ 0% PLL@51.5C MCPU@51.5C PMIC@50C GPU@50C BCPU@51.5C thermal@51.2C
RAM 1690/3825MB (lfb 339x4MB) SWAP 0/1913MB (cached 0MB) CPU [5%@345,off,off,1%@345,27%@345,1%@345] EMC_FREQ 0% GR3D_FREQ 0% PLL@50.5C MCPU@50.5C PMIC@50C GPU@50C BCPU@50.5C thermal@50.6C
RAM 1690/3825MB (lfb 339x4MB) SWAP 0/1913MB (cached 0MB) CPU [2%@345,off,off,0%@345,1%@345,1%@345] EMC_FREQ 0% GR3D_FREQ 0% PLL@51C MCPU@51C PMIC@50C GPU@50C BCPU@51C thermal@50.3C
```

After
```
RAM 1674/3825MB (lfb 334x4MB) SWAP 0/1913MB (cached 0MB) CPU [1%@2035,off,off,0%@2035,0%@2035,0%@2035] EMC_FREQ 0% GR3D_FREQ 0% PLL@50.5C MCPU@50.5C PMIC@50C GPU@50C BCPU@50.5C thermal@50.3C
RAM 1674/3825MB (lfb 334x4MB) SWAP 0/1913MB (cached 0MB) CPU [1%@2035,off,off,0%@2035,0%@2035,0%@2035] EMC_FREQ 0% GR3D_FREQ 0% PLL@50.5C MCPU@50.5C PMIC@50C GPU@50C BCPU@50.5C thermal@50.3C
RAM 1674/3825MB (lfb 334x4MB) SWAP 0/1913MB (cached 0MB) CPU [0%@2035,off,off,0%@2035,0%@2035,0%@2035] EMC_FREQ 0% GR3D_FREQ 0% PLL@50.5C MCPU@50.5C PMIC@50C GPU@50C BCPU@50.5C thermal@50.3C
RAM 1674/3825MB (lfb 334x4MB) SWAP 0/1913MB (cached 0MB) CPU [0%@2035,off,off,0%@2035,0%@2035,0%@2035] EMC_FREQ 0% GR3D_FREQ 0% PLL@50.5C MCPU@50.5C PMIC@50C GPU@50C BCPU@50.5C thermal@50.3C
RAM 1674/3825MB (lfb 334x4MB) SWAP 0/1913MB (cached 0MB) CPU [1%@2035,off,off,0%@2035,0%@2035,0%@2035] EMC_FREQ 0% GR3D_FREQ 0% PLL@50.5C MCPU@50.5C PMIC@50C GPU@50C BCPU@50.5C thermal@50.3C
RAM 1674/3825MB (lfb 334x4MB) SWAP 0/1913MB (cached 0MB) CPU [0%@2035,off,off,1%@2035,0%@2035,0%@2035] EMC_FREQ 0% GR3D_FREQ 0% PLL@51C MCPU@51C PMIC@50C GPU@50.5C BCPU@51C thermal@50.3C
RAM 1674/3825MB (lfb 334x4MB) SWAP 0/1913MB (cached 0MB) CPU [0%@2035,off,off,0%@2035,0%@2035,0%@2035] EMC_FREQ 0% GR3D_FREQ 0% PLL@50.5C MCPU@50.5C PMIC@50C GPU@50C BCPU@50.5C thermal@50.3C
RAM 1674/3825MB (lfb 334x4MB) SWAP 0/1913MB (cached 0MB) CPU [1%@2035,off,off,0%@2035,0%@2035,0%@2035] EMC_FREQ 0% GR3D_FREQ 0% PLL@50.5C MCPU@50.5C PMIC@50C GPU@50C BCPU@50.5C thermal@50.6C
RAM 1674/3825MB (lfb 334x4MB) SWAP 0/1913MB (cached 0MB) CPU [0%@2035,off,off,0%@2035,0%@2035,0%@2035] EMC_FREQ 0% GR3D_FREQ 0% PLL@50.5C MCPU@50.5C PMIC@50C GPU@50C BCPU@50.5C thermal@50.3C
RAM 1674/3825MB (lfb 334x4MB) SWAP 0/1913MB (cached 0MB) CPU [0%@2035,off,off,0%@2035,0%@2035,0%@2035] EMC_FREQ 0% GR3D_FREQ 0% PLL@50.5C MCPU@50.5C PMIC@50C GPU@50C BCPU@50.5C thermal@50.3C
RAM 1674/3825MB (lfb 334x4MB) SWAP 0/1913MB (cached 0MB) CPU [1%@2035,off,off,0%@2035,0%@2035,0%@2035] EMC_FREQ 0% GR3D_FREQ 0% PLL@50.5C MCPU@50.5C PMIC@50C GPU@50C BCPU@50.5C thermal@50.3C
```
I can see CPU frequency has boost to its maximum available frequency, but jtop show that ```Jetson Clocks: inactive```

rbonghi, 2022-06-11:
Has this bug appeared again?

If yes, can you download the latest jtop and reopen this issue.
Thank you in advance

closed 2022-06-11 by rbonghi

fayfive, 2022-06-12:
No, it works well.