rbonghi/jetson_stats #190: jetson-stats not supported for [L4T 32.5.2]

moran991231, 2022-01-13:
Please update jetson-stats with new jetpack

**Linux for Tegra**
 - L4T: 32.5.2

**Jetson-Stats**
 - Version: 3.1.1
 
[image]

moran991231, 2022-01-13:
Plz help me

clintlombard, 2022-01-13:
I get the same warning, but everything seems to work fine in `jtop`

clintlombard, 2022-01-13:
Duplicate of #185

rbonghi, 2022-01-13:
I read now. I fix soon :-)

closed 2022-01-13 by rbonghi

rbonghi, 2022-01-13:
Please update jetson-stats to **3.1.2**

```
sudo -H pip install -U jetson-stats
```

and reboot your board

moran991231, 2022-01-14:
It worked! Thanks!!! :)

moran991231, 2022-01-14:
Hi I have a question.
I have an account that I made at first when installing jetpack --> let this account `user0`  
And then, I made another one after installation jetpack --> let this `user`

With `user0` I can run `jtop` successfully.  
However with `user1`, `jtop` doesn't work, even though I try it after rebooting.
```bash
user1@my_board:~$ jtop
I can't access jetson_stats.service.
Please logout or reboot this board.
```
Fortunately, I can run through `sudo jtop` with `user1` (I can see a warning comment)
```bash
user1@my_board:~$ sudo jtop
[sudo] password for user1: 
[WARN] SUDO is no more required
```
Anyway I can run jtop with user1, but I wonder why it doesn't work through `jtop`.
I think you can give me a advice.
Thanks for reading

rbonghi, 2022-01-14:
because the second user is not in the same group and doesn't have access without sudo.

To enable it:
```
usermod -a -G jetson_stats user1
```
**remember to logout and login**

commit c3f5d9df (2022-01-13): Update jetson_variables

Fix #190 and #185