rbonghi/jetson_stats #193: Example jtop_logger.py throws exception in jtop.py

cap9qd, 2022-02-13:
**Describe the bug**
When I try to run the python examples I get the following error:
```
File "/usr/local/lib/python3.6/dist-packages/jtop/jtop.py", line 556, in stats
    stats['NVJPG'] = self.engine.nvjpg['rate'] if self.engine.nvjpg else 'OFF'
TypeError: 'int' object is not subscriptable
```

**To Reproduce**
Steps to reproduce the behavior:
1: Installed jtop via pip3 (currently shipped jetpack 4.6) I also tried pip/pip2 for python2.7.
2: Restart jetson
3: Verified jtop works.
4: Run the example jtop_logger.py.

**Expected behavior**
Output jtop data.

**Additional context**
This is a brand new install of jetpack without running any apt updates. I need to pin docker and containerd and haven't figure that out yet. 
[image]

cap9qd, 2022-02-13:
Something that I have noticed is that nvjpg is being utilized from another process off and on (Frigate beta) and toggles back and forth from "OFF" to showing running at 6xx.xxMHz. The example works fine until it switches state but that happens often.

cap9qd, 2022-02-13:
Seems this is the "offending" code in service.py (lines 421 and 422)
```
        if nvjpg_data:
            data['engines']['NVJPG'] = nvjpg_data['rate'] if nvjpg_data['status'] else {}
```

This returns an int or an empty dict. IDK what is the best option as changing this to match the other nvdec and nvenc will break jtop.  

For my purposes I updated jtop.py @ line 556 to handle the int for now:
`stats['NVJPG'] = (self.engine.nvjpg/100000) if type(self.engine.nvjpg) is int else 'OFF'`

rbonghi, 2023-01-05:
This error should now be fixed with the latest release **4.0** try to install and check again
```
sudo -H pip install -U jetson-stats
```

closed 2023-01-05 by rbonghi