rbonghi/jetson_stats #198: info is correct (updated?)?

leonpano2006, 2022-03-09:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. open minecraft on jetson and set -Xmx7400m and run
2. open terminal and open jtop
3. then ram is almost full but swap useage is not updated (stucked) but system monitor is keep to update info
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Expected behavior**
A clear and concise description of what you expected to happen.
update info 
**Additional context**
Add any other context about the problem here.

**Board**
 - SOC: [e.g. tegraXXX]
 - ID: [e.g. 10]
 - Board(s): [e.g. P1000-1000]
 - Code Name: [e.g. test]

**Jetpack**
 - Jetpack: [e.g. 4.3]
 - L4T: [e.g. 5.2.1]

**Jetson-Stats**
 - Version: [e.g. 1.8]

PrestonHack, 2022-04-15:
run jetson_config also set the usergroup for jtop to jetson_stats

PrestonHack, 2022-04-20:
5 is out? cool

rbonghi, 2022-04-21:
Hi @PrestonHack ,

I will start to work on jetson-stats soon! Just a few weeks, apologize for my delay

rbonghi, 2022-06-11:
@leonpano2006 
Does this bug appear on the JP 5.0? if this is true, can you check with the release candidate 4.0.0rc2
```
sudo -H pip install jetson-stats==4.0.0rc2
```

leonpano2006, 2022-06-12:
I need to fix my synology first

rbonghi, 2023-01-05:
Now Jetpack added with the latest released **4.0**
```
sudo -H pip install -U jetson-stats
```

closed 2023-01-05 by rbonghi