rbonghi/jetson_stats #203: jetson-stats not supported for [L4T 32.7.1]

leonpano2006, 2022-03-16:
Please update jetson-stats with new jetpack

**Linux for Tegra**
 - L4T: 32.7.1

**Jetson-Stats**
 - Version: 3.1.2

watersounds, 2022-03-25:
have same question

rbonghi, 2022-03-25:
Still releasing, there are some issues to release the documentation

rbonghi, 2022-03-25:
The bug is coming from this issue: https://github.com/sphinx-doc/sphinx/issues/10291
I'm finding a workaround, but right now I didn't find a solution

leonpano2006, 2022-03-26:
> The bug is coming from this issue: [sphinx-doc/sphinx#10291](https://github.com/sphinx-doc/sphinx/issues/10291) I'm finding a workaround, but right now I didn't find a solution

is usable but it will say is unsupported 
other the this everything works out of box

rbonghi, 2022-03-28:
Temporary skipped the documentation.
Now you can install the new jetson-stats version

To update:
```
sudo -H pip install -U jetson-stats
```
and reboot your board

closed 2022-03-28 by rbonghi