rbonghi/jetson_stats #210: jetson-stats not supported for [L4T 34.1.0] (Jetpack 5.0)

leonpano2006, 2022-04-08:
Please update jetson-stats with new jetpack

**Linux for Tegra**
 - L4T: 34.1.0

**Jetson-Stats**
 - Version: 3.1.3

leonpano2006, 2022-04-08:
and Fan is not working it will always show off even is runing on maxium speed

rbonghi, 2022-04-08:
Hi @leonpano2006 
I will focus on JP5.0 soon. I think from next month.

Let me know all issues you find, and I will fix all.

leonpano2006, 2022-04-09:
> Hi @leonpano2006 
> 
> I will focus on JP5.0 soon. I think from next month.
> 
> 
> 
> Let me know all issues you find, and I will fix all.

https://github.com/rbonghi/jetson_stats/issues/211
This is also one issue

leonpano2006, 2022-04-09:
i also find other issue
memory only shows ram not vram
[image]
unlike this one in jetpack 4.6.1
[image]
and meny info missing
[image]
on jetpack 4.6.1
[image]

rbonghi, 2022-04-09:
Aligned and renamed this issue

leonpano2006, 2022-04-09:
And FAN control function is failed

leonpano2006, 2022-04-09:
lavapipe
similar to (that is jetpack 4.x with ubuntu 20.04) https://github.com/rbonghi/jetson_stats/issues/173
[image]

phoenixmy, 2022-04-16:
I met same issue on Jetson Orin Dev Kit, maybe it is too new...

~$ jtop
[WARN] Board missing P3737-000 (press CTRL + Click)
[WARN] jetson-stats not supported for [L4T 34.1.0]
  Please, try: sudo -H pip install -U jetson-stats
  or open an issue on Github (press CTRL + Click)

[image]

leonpano2006, 2022-04-17:
Of course 
Orin is not exist in l4t 32.x

leonpano2006, 2022-04-21:
[image]
In jetpack 5.0 it can't detect ram use from gpu (only cpu)

rbonghi, 2022-06-11:
I think is only missing to fix the GPU memory read and all major bugs for JP5 are fixed.
The new pre-release is the 4.0.0rc2
```
sudo -H pip install jetson-stats==4.0.0rc2
```

dn1983, 2022-06-16:
Hi.
What about temperature?
[image]

leonpano2006, 2022-06-16:
> Hi.
> 
> What about temperature?
> 
> [image]
> 
> 

What is this absolute zero temperature

rbonghi, 2022-06-16:
I think you installed a script that emulates fake data. Are you working on a desktop?

rbonghi, 2022-06-16:
If you have an NVIDIA Jetson Orin, I cannot test, and I don't know this output.

rbonghi, 2023-01-05:
This error should now be fixed with the latest release, **4.0** try to install and check again.
```
sudo -H pip install -U jetson-stats
```

The temperatures come directly from the Jetson sensors. So it is not an error from jtop. I hope that future Jetpack releases will fix them.

closed 2023-01-05 by rbonghi