rbonghi/jetson_stats #212: OpenCV CUDA variables update

HoneyPatouceul, 2022-04-19:
Running L4T R32.6.1 on AGX Xavier. 
Seems recent opencv versions doesn't show "Use Cuda" in getBuildInformation() used by opencv_version --verbose.
Sorry I'm unable to make a PR, but it may be possible to get opencv CUDA availabilty with:
```
opencv_version --verbose | grep  "CUDA" | cut -f2 -d ':' | tr -s ' ' | cut -d' ' -f2
```
and CUDA version with :
```
opencv_version --verbose | grep  "CUDA" | cut -f2 -d ':' | tr -s ' ' | cut -d' ' -f4 | cut -d',' -f1
```

rbonghi, 2022-05-20:
Hi @HoneyPatouceul 

if I well understand I fixed using your hints:

https://github.com/rbonghi/jetson_stats/blob/develop/jtop/jetson_libraries#L35

and

https://github.com/rbonghi/jetson_stats/blob/develop/jtop/jetson_libraries#L49

Let me know if make sense for you as well

HoneyPatouceul, 2022-05-20:
@rbonghi yes, that's the suggested change. Not tested, though. Feel free to close the issue when verified.

rbonghi, 2022-05-20:
Thanks I will check for the new release

commit 2c040ffa (2022-05-20): Fix #212
closed 2022-06-10 by rbonghi