rbonghi/jetson_stats #218: Board missing UNKNOWN (Nintendo Switch (2017))

theofficialgman, 2022-05-08:
Please update jetson-stats with this board

**Board**
 - Board(s): UNKNOWN
 - Boardis: 2595:0000:A0
 - SOC: tegra210
 - ID: 33
 - Code Name: icosa

**Jetpack**
 - Jetpack: 4.3
 - L4T: 32.3.1

**Jetson-Stats**
 - version: 3.1.3

`/sys/firmware/devicetree/base/model` -> `Nintendo Switch (2017)`

leonpano2006, 2022-05-16:
@gman#3709
I think he are focused on jetpack 5.0 update 
and he is now work for Nvidia. idk if he will add switch root support 
if not then we need to make stop for switch root fork

rbonghi, 2022-05-31:
Interesting this variable! I have never seen it!
I will add the Jetson model name!

rbonghi, 2022-06-10:
If you want to test the new version [4.0.0rc1](https://github.com/rbonghi/jetson_stats/releases/tag/4.0.0rc1):

```
sudo -H pip install jetson-stats==4.0.0rc1
```

Please remember right now jtop is **unstable**, I'm still fixing other bugs

theofficialgman, 2022-06-10:
I'm not sure if it was correct for you to make this change
https://github.com/rbonghi/jetson_stats/commit/d64eca1abb461a74ffa9bf955ee2974b89103170

I do not think that every devicetree released by nvidia contains this model variable. If memory serves me correctly, I think that the TX2 devicetrees do not contain it. I could be wrong though, and they could have fixed that

theofficialgman, 2022-06-10:
regardless... I think you have made some sytax error @rbonghi 

```
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/jtop/__main__.py", line 148, in main
    curses.wrapper(JTOPGUI, jetson, [ALL, GPU, CPU, MEM, CTRL, INFO], init_page=args.page, loop=args.loop, seconds=LOOP_SECONDS)
  File "/usr/lib/python3.6/curses/__init__.py", line 94, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopgui.py", line 109, in __init__
    self.run(loop, seconds)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopgui.py", line 138, in run
    self.draw()
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/lib/common.py", line 80, in wrapped
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopgui.py", line 149, in draw
    self.header()
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/lib/common.py", line 120, in wrapped
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopgui.py", line 186, in header
    status = [self.jetson.board.info["model"]]
KeyError: 'model'
```

theofficialgman, 2022-06-10:
ah nevermind I just needed to reboot since the old service was still loaded

rbonghi, 2022-06-10:
Thx!
Yes, need a reboot, I changed some variables

theofficialgman, 2022-06-11:
@rbonghi you appear to be using a grep somewhere to remove the "N" when displaying the model under the Board: Model: category

so it shows up a "intendo Switch (2017)" there

rbonghi, 2022-06-11:
In the new pre-release will be fixed. I keep you posted

rbonghi, 2023-01-05:
Now works with release **4.0**
```
sudo -H pip install -U jetson-stats
```