rbonghi/jetson_stats #228: ImportError: cannot import name 'jtop'

AnamikaPaul, 2022-05-27:
**Describe the bug**
When i run this code it doesnt get jtop and shows error. i have already installed jtop

from jtop import jtop
import json, datetime

if __name__ == "__main__":

    with jtop() as jetson:
            # jetson.stats provides our system measurements as type dict.
            tmp = jetson.stats  
            # time and uptime are proved as time objects. These needed to be converted before passing as a JSON string,
            tmp["time"] = str(tmp["time"].strftime('%m/%d/%Y'))
            tmp["uptime"] = str(tmp["uptime"])
            # We then convert our dict -> Json string
            influx_json= {"jetson": tmp}
            print(json.dumps(influx_json), flush=True)
**To Reproduce**

**python3 jtop.py**
Traceback (most recent call last):
  File "jtop.py", line 8, in <module>
    from jtop import jtop
  File "/home/anamika/Downloads/yolo_5/jtop.py", line 8, in <module>
    from jtop import jtop
ImportError: cannot import name 'jtop'
anamika@anamika-desktop:~/Downloads/yolo_5$ python3 --version
Python 3.6.9

rbonghi, 2022-05-30:
Let me check, I'm working for a new pimped version :-)

rbonghi, 2022-06-11:
how did you install jtop?
If you used "pip install" (the library is only available on python 2.7)
In this case, you need to:
```
sudo -H pip uninstall jetson_stats
```
reboot and
```
sudo -H pip3 install jetson_stats
```

If you didn't fix it, reopen this issue

closed 2022-06-11 by rbonghi