rbonghi/jetson_stats #232: Board missing P3737-000

avideci, 2022-06-01:
Please update jetson-stats with this board

**Board**
 - Board(s): P3737-000
 - Boardis: 
 - SOC: tegra23x
 - ID: 
 - Code Name: concord

**Jetpack**
 - Jetpack: UNKNOWN
 - L4T: 34.1.1

**Jetson-Stats**
 - version: 3.1.4

avideci, 2022-06-01:
It works fine with sudo, 
But it still failed after I tried to re-install it using this prompt:
```bash
[WARN] SUDO is no more required
[WARN] Board missing P3737-000 (press CTRL + Click)
[WARN] jetson-stats not supported for [L4T 34.1.1]
  Please, try: sudo -H pip install -U jetson-stats
  or open an issue on Github (press CTRL + Click)
```
Thanks.

rbonghi, 2022-06-02:
I'm currently working on a new release for Jetpack 5.
A first pre-release should be out next week.

closed 2023-01-03 by rbonghi