rbonghi/jetson_stats #234: Do not have fan controller

cpymaple, 2022-06-09:
Hi, my device is Jetson NX and my environment is ubuntu 20.04. After I install the jetson_stats, it seems that I do not have the fan controller (jtop -> 5 ->but there is no button to control fan). And then I want to control the fan manually, but I do not have the pwm-fan folder in /sys/devices/pwm-fan. Do you know how to solve it ?

cpymaple, 2022-06-09:
[image]
[image]
[image]

rbonghi, 2022-06-10:
You are working with JP5.0 that is currently not supported. #224 
I will release soon a new version to manage it

rbonghi, 2022-06-10:
If you want to test the new version [4.0.0rc1](https://github.com/rbonghi/jetson_stats/releases/tag/4.0.0rc1):

```
sudo -H pip install jetson-stats==4.0.0rc1
```

Please remember right now jtop is **unstable**, I'm still fixing other bugs

rbonghi, 2022-06-10:
Hi @theofficialgman 
which Jetson are you using?

rbonghi, 2022-06-10:
BTW: I found the same bug on old JP. I'm still fixing :-)

Thank you

rbonghi, 2022-06-11:
This release should fix all these bugs
```
sudo -H pip install jetson-stats==4.0.0rc2
```
Please remember right now jtop is **unstable**, I'm still fixing other bugs

cpymaple, 2022-06-12:
> If you want to test the new version [4.0.0rc1](https://github.com/rbonghi/jetson_stats/releases/tag/4.0.0rc1):
> 
> ```
> sudo -H pip install jetson-stats==4.0.0rc1
> ```
> 
> Please remember right now jtop is **unstable**, I'm still fixing other bugs

OK, thank you. I find a way to solve it.  1. sudo /usr/bin/jetson_clocks --fan, this command can turn on the fan. 2. /sys/devices/platform/pwm-fan/hwmon/hwmon4. In this folder, we can find pwm1 document, change the number( from 0-255 ) can change the speed of fan.

closed 2022-06-12 by cpymaple