rbonghi/jetson_stats #238: jtop not displaying power consumption in Jetpack 5.0

agrija9, 2022-06-10:
Hi, I am running a Jetson Xavier NX (Jetpack 5.0). When I launch jtop I can not see power consumption (I could see these stats with Jetpack 4.6 for example). Do you recommend me re-installing jtop?
[image]

rbonghi, 2022-06-11:
I pre-released a new version of jetson_stats [4.0.0rc1](https://github.com/rbonghi/jetson_stats/releases/tag/4.0.0rc1)

If you want to test please follow the command below:

```
sudo -H pip install jetson-stats==4.0.0rc1
```

rbonghi, 2022-06-13:
Hi @agrija9 ,

1. Good point, I check why the numbers are not the same! I didn't notice
2. For the second question, you can use the jetson-stats API, like this example: https://github.com/rbonghi/jetson_stats/blob/master/examples/jtop_logger.py

rbonghi, 2023-01-05:
Try the new release **4.0**
```
sudo -H pip install -U jetson-stats
```