rbonghi/jetson_stats #254: Docker Deployment does not work - pull access denied for rbonghi/jetson-stats

subzero210, 2022-07-22:
**Describe the bug**
With docker deployment, the pull failed because the image seems not to exist.

**To Reproduce**
Steps to reproduce the behavior:
1. Try tu run the command "docker pull rbonghi/jetson-stats" as described in the README

**Expected behavior**
Pulling correctly the image in order to run a new container

**Jetpack**
 - Jetpack: [e.g. 4.3]
 - L4T: [e.g. 5.2.1]

**Jetson-Stats**
 - Version: [e.g. 1.8]

nielingyun, 2022-07-25:
run:  docker pull rbonghi/jetson_stats:master

mwindowshz, 2022-08-04:
On Tx2 jtop works, trying to use on docker

this downloaded the docker image, 
`docker pull rbonghi/jetson_stats:master`
but it cannot run there is an error:
`sudo docker run -it -v /run/jtop.sock:/run/jtop.sock rbonghi/jetson_stats:master`

> Traceback (most recent call last):
>   File "/usr/local/bin/jtop", line 8, in <module>
>     sys.exit(main())
>   File "/usr/local/lib/python3.10/site-packages/jtop/__main__.py", line 148, in main
>     curses.wrapper(JTOPGUI, jetson, [ALL, GPU, CPU, MEM, CTRL, INFO], init_page=args.page, loop=args.loop, seconds=LOOP_SECONDS)
>   File "/usr/local/lib/python3.10/curses/__init__.py", line 94, in wrapper
>     return func(stdscr, *args, **kwds)
>   File "/usr/local/lib/python3.10/site-packages/jtop/gui/jtopgui.py", line 109, in __init__
>     self.run(loop, seconds)
>   File "/usr/local/lib/python3.10/site-packages/jtop/gui/jtopgui.py", line 138, in run
>     self.draw()
>   File "/usr/local/lib/python3.10/site-packages/jtop/gui/lib/common.py", line 80, in wrapped
>     return func(self, *args, **kwargs)
>   File "/usr/local/lib/python3.10/site-packages/jtop/gui/jtopgui.py", line 149, in draw
>     self.header()
>   File "/usr/local/lib/python3.10/site-packages/jtop/gui/lib/common.py", line 120, in wrapped
>     return func(*args, **kwargs)
>   File "/usr/local/lib/python3.10/site-packages/jtop/gui/jtopgui.py", line 186, in header
>     status = [self.jetson.board.info["model"]]
> KeyError: 'model'

And when trying to install on my docker  the service needs to be restarted with a command that is not available in the docker

> Successfully built jetson-stats
> Installing collected packages: jetson-stats
> Successfully installed jetson-stats-3.1.4
> root@ubuntu:/# jtop
> The jetson_stats.service is not active. Please run:
> sudo systemctl restart jetson_stats.service
> 
> root@ubuntu:/# systemctl restart jetson_stats.service
> System has not been booted with systemd as init system (PID 1). Can't operate.
> Failed to connect to bus: Host is down

Hope there is a fix. 
Thanks

mwindowshz, 2022-08-08:
I managed to run 
`run -it -v /run/jtop.sock:/run/jtop.sock rbonghi/jetson_stats:master /bin/bash 
`so got terminal. 
then tried running jtop, it crashes as above. 
reinstalled using 
` pip3 uninstall jetson_stats
``pip3 install jetson_stats
`now jtop loads, but the last page of info crashes:

> Traceback (most recent call last):
>   File "/usr/local/bin/jtop", line 8, in <module>
>     sys.exit(main())
>   File "/usr/local/lib/python3.10/site-packages/jtop/__main__.py", line 148, in main
>     curses.wrapper(JTOPGUI, jetson, [ALL, GPU, CPU, MEM, CTRL, INFO], init_page=args.page, loop=args.loop, seconds=LOOP_SECONDS)
>   File "/usr/local/lib/python3.10/curses/__init__.py", line 94, in wrapper
>     return func(stdscr, *args, **kwds)
>   File "/usr/local/lib/python3.10/site-packages/jtop/gui/jtopgui.py", line 109, in __init__
>     self.run(loop, seconds)
>   File "/usr/local/lib/python3.10/site-packages/jtop/gui/jtopgui.py", line 138, in run
>     self.draw()
>   File "/usr/local/lib/python3.10/site-packages/jtop/gui/lib/common.py", line 80, in wrapped
>     return func(self, *args, **kwargs)
>   File "/usr/local/lib/python3.10/site-packages/jtop/gui/jtopgui.py", line 153, in draw
>     page.draw(self.key, self.mouse)
>   File "/usr/local/lib/python3.10/site-packages/jtop/gui/pinfo.py", line 96, in draw
>     if self.jetson.local_interfaces:
>   File "/usr/local/lib/python3.10/site-packages/jtop/jtop.py", line 818, in local_interfaces
>     return get_local_interfaces()
>   File "/usr/local/lib/python3.10/site-packages/jtop/core/common.py", line 179, in get_local_interfaces
>     namestr = names.tostring()
> AttributeError: 'array.array' object has no attribute 'tostring'
>

nielingyun, 2022-08-08:
> I managed to run was missing `-v /run/jtop.sock:/run/jtop.sock`
> 
> `run -it -v /run/jtop.sock:/run/jtop.sock rbonghi/jetson_stats:master /bin/bash `so got terminal. then tried running jtop, it crashes as above. reinstalled using `` pip3 uninstall jetson_stats ``pip3 install jetson_stats ``now jtop loads, but the last page of info crashes:
> 
> > Traceback (most recent call last):
> > File "/usr/local/bin/jtop", line 8, in 
> > sys.exit(main())
> > File "/usr/local/lib/python3.10/site-packages/jtop/**main**.py", line 148, in main
> > curses.wrapper(JTOPGUI, jetson, [ALL, GPU, CPU, MEM, CTRL, INFO], init_page=args.page, loop=args.loop, seconds=LOOP_SECONDS)
> > File "/usr/local/lib/python3.10/curses/**init**.py", line 94, in wrapper
> > return func(stdscr, *args, **kwds)
> > File "/usr/local/lib/python3.10/site-packages/jtop/gui/jtopgui.py", line 109, in **init**
> > self.run(loop, seconds)
> > File "/usr/local/lib/python3.10/site-packages/jtop/gui/jtopgui.py", line 138, in run
> > self.draw()
> > File "/usr/local/lib/python3.10/site-packages/jtop/gui/lib/common.py", line 80, in wrapped
> > return func(self, *args, **kwargs)
> > File "/usr/local/lib/python3.10/site-packages/jtop/gui/jtopgui.py", line 153, in draw
> > page.draw(self.key, self.mouse)
> > File "/usr/local/lib/python3.10/site-packages/jtop/gui/pinfo.py", line 96, in draw
> > if self.jetson.local_interfaces:
> > File "/usr/local/lib/python3.10/site-packages/jtop/jtop.py", line 818, in local_interfaces
> > return get_local_interfaces()
> > File "/usr/local/lib/python3.10/site-packages/jtop/core/common.py", line 179, in get_local_interfaces
> > namestr = names.tostring()
> > AttributeError: 'array.array' object has no attribute 'tostring'

https://github.com/rbonghi/jetson_stats/pull/153#issue-925284258

rbonghi, 2023-01-05:
This error should now be fixed with the latest release **4.0** try to install and check again
```
sudo -H pip install -U jetson-stats
```

closed 2023-01-05 by rbonghi