rbonghi/jetson_stats #284: Jetson Orin Developer Kit 中的错误 GUI

linzyjx, 2022-10-23:
**Describe the bug**
GUI wrong in Jetson Orin Developer Kit

1. Display order of CPU 10, 11, 12 is wrong.
2. Sensor CV0,1,2 display wrong.
3. Power display wrong in "VDDQ VDD2 1V8"?

**To Reproduce**

Not applicable.

**Screenshots**
[image]

**Expected behavior**
Not applicable.

**Additional context**
Not applicable.

**Board**
 - SOC:tegra23x
 - ID: [e.g. 10]
 - Board(s): Module: P3701-0000 - Carrier: P3737-0000
 - Code Name: concord

**Jetpack**
 - Jetpack:Jetpack 5.0.1 DP
 - L4T: 34.1.1

**Jetson-Stats**
 - Version: jtop 4.0.0rc2

rbonghi, 2022-11-01:
[image]
Still working to fix it. New RC I hope will be fix today :-)

rbonghi, 2023-01-05:
Fully support Orin with the latest released **4.0**
```
sudo -H pip install -U jetson-stats
```

closed 2023-01-05 by rbonghi