rbonghi/jetson_stats #290: jtop not working without sudo

harshad-0302, 2022-11-02:
when I am trying to run jtop without sudo it is giving me below error:
[image]

But with sudo it is working fine. Please tell me how can I run it without sudo rights.

engrjislam, 2022-11-02:
`sudo` will not require from the next boot. Restart your **jetson** device.

harshad-0302, 2022-11-03:
I rebooted the board using sudo shutdown -r now . But still getting same after reboot while running jtop

harshad-0302, 2023-01-04:
Hi @rbonghi 

Please find logs below:

__/Jetson_performance$ journalctl -u jetson_stats.service
-- Logs begin at Wed 2023-01-04 08:16:04 CET, end at Wed 2023-01-04 10:54:55 CET. --
Jan 04 08:16:06 platform-rename systemd[1]: Started jetson_stats service.
Jan 04 08:16:06 platform-rename jtop[5279]: [INFO] jtop.core.common - fan loaded on /sys/devices/pwm-fan
Jan 04 08:16:06 platform-rename jtop[5279]: [INFO] jtop.core.common - jetson_clocks loaded on /usr/bin/jetson_clocks
Jan 04 08:16:06 platform-rename jtop[5279]: [INFO] jtop.core.common - tegrastats loaded on /usr/bin/tegrastats
Jan 04 08:16:06 platform-rename jtop[5279]: [INFO] jtop.__main__ - jetson_stats server loaded
Jan 04 08:16:08 platform-rename jtop[5279]: [INFO] jtop.core.fan - Mode set default status=False
Jan 04 09:14:56 platform-rename jtop[5279]: [INFO] jtop.service - tegrastats started 500ms
Jan 04 09:15:58 platform-rename jtop[5279]: [INFO] jtop.service - tegrastats close
Jan 04 09:15:58 platform-rename jtop[5279]: [INFO] jtop.service - jetson_clocks show closed

platform@platform-rename:~/Harshad/jtop_script/Performance_data_to_BDC_app/Jetson_performance$ jtop
I can't access jetson_stats.service.
Please logout or reboot this board.

Also check when i run jtop without sudo it gives above error. I tried rebooting also but did'nt worked still same error.

rbonghi, 2023-01-04:
mmm weird, the jtop server is running.

Have you install jtop with different python version?

harshad-0302, 2023-01-04:
Hi,

I installed jtop with pip3 i.e python3 i guess.

rbonghi, 2023-01-04:
I apologize if I ask to do an uninstall and a new reinstall, but you help me understand the reason it doesn't work.

```
sudo pip uninstall jetson-stats
sudo pip3 uninstall jetson-stats
```
Let me know if this message appears and when, it's really helpful: `WARNING: Skipping jetson-stats as it is not installed.`

**REBOOT**

try now:
```
jtop
```
Now you should read an error that jtop is not installed

Check if not installed jtop service:
```
systemctl status jetson_stats.service
```

Reinstall jetson stats
```
sudo pip3 install jetson-stats
```

Should work now.
If doesn't work again, can you send me the logs please from these commands:
```
systemctl status jetson_stats.service
```

```
journalctl -u jetson_stats.service
```

rbonghi, 2023-01-05:
jtop can be installed only in one python distribution.

Try to open the python terminal and write `import jtop` if you don't read any error is installed jetson_stats also on python 2.7

harshad-0302, 2023-01-05:
It is giving error as "No module named jtop"

rbonghi, 2023-01-05:
Wait my new release this afternoon I add a feature that help me to understand the reason is not running on your jetson

rbonghi, 2023-01-05:
This error should now be fixed with the latest release **4.0** try to install and check again
```
sudo -H pip install -U jetson-stats
```

harshad-0302, 2023-01-06:
Shall I uninstall and then install or directly I can run this command??

harshad-0302, 2023-01-06:
Hi @rbonghi 

I installed the latest version of jtop and then I rebooted the system.
After that if I am trying to run jtop without sudo it is giving me below error now:
[image]

Please let me what shall i do now?

rbonghi, 2023-01-06:
Can you show me the output from these commands?

```
systemctl status jetson_stats.service
```

```
journalctl -u jetson_stats.service
```

harshad-0302, 2023-01-06:
Please find logs below:
systemctl status jetson_stats.service
● jetson_stats.service - jetson_stats service
   Loaded: loaded (/etc/systemd/system/jetson_stats.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Fri 2023-01-06 09:44:39 CET; 3h 1min ago
  Process: 4892 ExecStart=/usr/local/bin/jtop --force (code=exited, status=0/SUCCESS)
 Main PID: 4892 (code=exited, status=0/SUCCESS)

Jan 06 09:44:39 platform-rename jtop[4892]: Process JtopServer-1:
Jan 06 09:44:39 platform-rename jtop[4892]: Traceback (most recent call last):
Jan 06 09:44:39 platform-rename jtop[4892]:   File "/usr/lib/python3.6/multiprocessing/process.py", line 258, in _bootstrap
Jan 06 09:44:39 platform-rename jtop[4892]:     self.run()
Jan 06 09:44:39 platform-rename jtop[4892]:   File "/usr/local/lib/python3.6/dist-packages/jtop/service.py", line 188, in run
Jan 06 09:44:39 platform-rename jtop[4892]:     self.jetson_clocks.initialization(self.nvpmodel)
Jan 06 09:44:39 platform-rename jtop[4892]:   File "/usr/local/lib/python3.6/dist-packages/jtop/core/jetson_clocks.py", line 291, in init
Jan 06 09:44:39 platform-rename jtop[4892]:     raise JtopException("I cannot initialize jetson_clocks controller")
Jan 06 09:44:39 platform-rename jtop[4892]: jtop.core.exceptions.JtopException: I cannot initialize jetson_clocks controller
Jan 06 09:44:39 platform-rename jtop[4892]: [INFO] jtop.service - Service closed

journalctl -u jetson_stats.service
-- Logs begin at Tue 2022-09-06 05:18:05 CEST, end at Fri 2023-01-06 12:46:32 CET. --
Jan 06 09:44:30 platform-rename systemd[1]: Started jetson_stats service.
Jan 06 09:44:31 platform-rename jtop[4892]: [INFO] jtop.core.common - fan loaded on /sys/devices/platform
Jan 06 09:44:31 platform-rename jtop[4892]: [WARNING] jtop.service - Fan is not available on this board in paths ['/sys/devices/platform'
Jan 06 09:44:31 platform-rename jtop[4892]: [INFO] jtop.core.common - fan loaded on /sys/devices/pwm-fan
Jan 06 09:44:31 platform-rename jtop[4892]: [INFO] jtop.core.common - jetson_clocks loaded on /usr/bin/jetson_clocks
Jan 06 09:44:31 platform-rename jtop[4892]: [INFO] jtop.core.common - tegrastats loaded on /usr/bin/tegrastats
Jan 06 09:44:31 platform-rename jtop[4892]: [INFO] jtop.__main__ - jetson_stats server loaded
Jan 06 09:44:31 platform-rename jtop[4892]: [INFO] jtop.__main__ - Running on python: 3.6.9
Jan 06 09:44:32 platform-rename jtop[4892]: [ERROR] jtop.core.jetson_clocks - [0] [errno:1] Error process:
Jan 06 09:44:33 platform-rename jtop[4892]: [ERROR] jtop.core.jetson_clocks - [1] [errno:1] Error process:
Jan 06 09:44:35 platform-rename jtop[4892]: [ERROR] jtop.core.jetson_clocks - [2] [errno:1] Error process:
Jan 06 09:44:36 platform-rename jtop[4892]: [ERROR] jtop.core.jetson_clocks - [3] [errno:1] Error process:
Jan 06 09:44:37 platform-rename jtop[4892]: [ERROR] jtop.core.jetson_clocks - [4] [errno:1] Error process:
Jan 06 09:44:39 platform-rename jtop[4892]: Process JtopServer-1:
Jan 06 09:44:39 platform-rename jtop[4892]: Traceback (most recent call last):
Jan 06 09:44:39 platform-rename jtop[4892]:   File "/usr/lib/python3.6/multiprocessing/process.py", line 258, in _bootstrap
Jan 06 09:44:39 platform-rename jtop[4892]:     self.run()
Jan 06 09:44:39 platform-rename jtop[4892]:   File "/usr/local/lib/python3.6/dist-packages/jtop/service.py", line 188, in run
Jan 06 09:44:39 platform-rename jtop[4892]:     self.jetson_clocks.initialization(self.nvpmodel)
Jan 06 09:44:39 platform-rename jtop[4892]:   File "/usr/local/lib/python3.6/dist-packages/jtop/core/jetson_clocks.py", line 291, in init
Jan 06 09:44:39 platform-rename jtop[4892]:     raise JtopException("I cannot initialize jetson_clocks controller")
Jan 06 09:44:39 platform-rename jtop[4892]: jtop.core.exceptions.JtopException: I cannot initialize jetson_clocks controller
Jan 06 09:44:39 platform-rename jtop[4892]: [INFO] jtop.service - Service closed
lines 1-24/24 (END)...skipping...

rbonghi, 2023-01-06:
There is something wrong with jetson_clock initialization.

Can you try this for me command:

```
sudo /usr/bin/jetson_clocks --show
```

Thank you for your help! I really appreciate

harshad-0302, 2023-01-06:
In previous version , I was able to run this issue is happening after installing again.

admin@platform-rename:/$ sudo /usr/bin/jetson_clocks --show
SOC family:tegra194  Machine:jetson-xavier
Online CPUs: 0-7
cpu0: Online=1 Governor=schedutil MinFreq=115200 MaxFreq=2265600 CurrentFreq=1497600 IdleStates: C1=1 c6=1 
cpu1: Online=1 Governor=schedutil MinFreq=115200 MaxFreq=2265600 CurrentFreq=2265600 IdleStates: C1=1 c6=1 
cpu2: Online=1 Governor=schedutil MinFreq=115200 MaxFreq=2265600 CurrentFreq=2265600 IdleStates: C1=1 c6=1 
cpu3: Online=1 Governor=schedutil MinFreq=115200 MaxFreq=2265600 CurrentFreq=2265600 IdleStates: C1=1 c6=1 
cpu4: Online=1 Governor=schedutil MinFreq=115200 MaxFreq=2265600 CurrentFreq=2265600 IdleStates: C1=1 c6=1 
cpu5: Online=1 Governor=schedutil MinFreq=115200 MaxFreq=2265600 CurrentFreq=2265600 IdleStates: C1=1 c6=1 
cpu6: Online=1 Governor=schedutil MinFreq=115200 MaxFreq=2265600 CurrentFreq=2265600 IdleStates: C1=1 c6=1 
cpu7: Online=1 Governor=schedutil MinFreq=115200 MaxFreq=2265600 CurrentFreq=2265600 IdleStates: C1=1 c6=1 
Error: Unknown GPU Type!

rbonghi, 2023-01-06:
I think something happened on your jetson_clock file.
I don't know the reason is lost on your Jetson.

Maybe you can recover from the SDK or try to open a post on the https://forums.developer.nvidia.com/

rbonghi, 2023-01-07:
I close this issue. Please reopen a new one if something goes wrong after fixing the jetson_clock call.

closed 2023-01-07 by rbonghi