rbonghi/jetson_stats #291: Support request for Nvidia Clara Agx

engrjislam, 2022-11-02:
A prompt request to support **`jtop`** in Nvidia Clara Agx Dev Kit. Some info is detected correctly and is showing well in **`jtop`** already but some are not (as if it can't find information as expected, e.g., CUDA). It would be nice to have support for Nvidia Clara Agx Dev Kit in its future release.

rbonghi, 2023-01-03:
Hi @engrjislam ,

Good question. I never tried a Clara Agx, but I will let you know :-)
Just for curiosity, if you can try my jetson-stats, can you share the output, just to have an idea about the effort to add this hardware.

Best,
Raffaello

engrjislam, 2023-11-13:
Hello @rbonghi, I am getting the following console warning for `sudo jtop`:

```
[WARN] jetson-stats not supported for [L4T 34.1.2]
Please, try: sudo pip3 install -U jetson-stats or
open a Github issue (press CTRL + Click)
```

Anyway, `jetson_stats` successfully installed into the system. Some of the `jtop` outputs are presented below:

[image]
[image]
[image]
[image]
[image]
[image]

`Jetpack NOT DETECTED` seems quite reasonable (Clara is an `AGX` system but not a `Jetson` system). As a result, `jetson_stats` package is not fully compliant with the AGX system. 

[SORRY for the delay ... ]

rbonghi, 2024-01-07:
thank you for your post! I was thinking to never add this feature on jtop :-)

Do you know if Clara AGX have also `tegrastats`?

To figure out also how a Clara AGX works, can you also provide me this output:

When you install, please share the output from the following:
```
sudo pip3 install --no-cache-dir -U jetson-stats
```
```
journalctl -u jtop.service -n 100 --no-pager
```
```
jetson_release -v
```
And if you can attach the output from
```
jtop --error-log
```

engrjislam, 2024-01-08:
Thanks for your consideration. I am highlighting the outputs that I am getting from my Clara AGX.

> Do you know if Clara AGX also has `tegrastats`?

[image]

> **journalctl -u jtop.service -n 100 --no-pager**

```
-- Logs begin at Fri 2023-12-08 09:14:31 PST, end at Sun 2024-01-07 15:48:55 PST. --
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.service - Running on Python: 3.8.10
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.cpu - Found 8 CPU
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.gpu - NVIDIA SMI exist!
Jan 07 15:44:41 ubuntu jtop[2972181]: [WARNING] jtop.core.gpu - No NVIDIA GPU available
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.processes - Process service started
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.memory - Found EMC!
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.memory - Memory service started
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.engine - Special Engine group found: [dlaX]
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.engine - Special Engine group found: [pvaX]
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.engine - Engines found: [APE CVNAS DLA0 DLA1 NVDEC NVENC NVJPG PVA0 PVA1 SE VIC]
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.temperature - Found thermal "AUX" in thermal_zone2
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.temperature - Found thermal "CPU" in thermal_zone0
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.temperature - Found thermal "mlx5" in thermal_zone9
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.temperature - Found thermal "PCIe" in thermal_zone7
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.temperature - Found thermal "Tboard" in thermal_zone5
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.temperature - Found thermal "AO" in thermal_zone3
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.temperature - Found thermal "GPU" in thermal_zone1
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.temperature - Found thermal "thermal" in thermal_zone8
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.temperature - Found thermal "Tdiode" in thermal_zone6
Jan 07 15:44:41 ubuntu jtop[2972181]: [WARNING] jtop.core.temperature - Skipped PMIC
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.power - Alarms CV - {'crit_alarm': 0, 'max_alarm': 0}
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.power - Alarms VDDRQ - {'crit_alarm': 0, 'max_alarm': 0}
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.power - Alarms SYS5V - {'crit_alarm': 0, 'max_alarm': 0}
Jan 07 15:44:41 ubuntu jtop[2972181]: [WARNING] jtop.core.power - Skipped "sum of shunt voltages" /sys/bus/i2c/devices/1-0041/hwmon/hwmon4/in7_label
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.power - Alarms GPU - {'crit_alarm': 0, 'max_alarm': 0}
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.power - Alarms CPU - {'crit_alarm': 0, 'max_alarm': 0}
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.power - Alarms SOC - {'crit_alarm': 0, 'max_alarm': 0}
Jan 07 15:44:41 ubuntu jtop[2972181]: [WARNING] jtop.core.power - Skipped "sum of shunt voltages" /sys/bus/i2c/devices/1-0040/hwmon/hwmon3/in7_label
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.power - Found I2C power monitor
Jan 07 15:44:41 ubuntu jtop[2972181]: [WARNING] jtop.core.power - Skipped usb-charger type=USB in=usb-charger
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.fan - Fan pwmfan(1) found in /sys/class/hwmon/hwmon6
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.fan - Fan pwmfan(1) found in /sys/class/hwmon/hwmon5
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.fan - RPM pwm_tach found in /sys/class/hwmon/hwmon2
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.fan - Found nvfancontrol.service
Jan 07 15:44:41 ubuntu jtop[2972181]: [INFO] jtop.core.jetson_clocks - jetson_clocks found in /usr/bin/jetson_clocks
Jan 07 15:44:41 ubuntu jtop[2972181]: [WARNING] jtop.core.nvpmodel - nvpmodel not available
Jan 07 15:44:41 ubuntu jtop[2972198]: [INFO] jtop.service - Initialization service
Jan 07 15:44:41 ubuntu jtop[2972198]: [INFO] jtop.core.fan - Initialization pwmfan
Jan 07 15:44:41 ubuntu jtop[2972198]: [WARNING] jtop.core.fan - Fan pwmfan profile manual already active
Jan 07 15:44:42 ubuntu jtop[2972198]: [INFO] jtop.service - service ready
Jan 07 15:44:49 ubuntu jtop[2972198]: [INFO] jtop.service - jtop timer thread started 1000ms
Jan 07 15:44:53 ubuntu jtop[2972198]: [INFO] jtop.service - jtop timer thread close
Jan 07 15:45:30 ubuntu systemd[1]: Stopping jtop service...
Jan 07 15:45:30 ubuntu jtop[2972198]: [INFO] jtop.__main__ - Close service by signal 15
Jan 07 15:45:30 ubuntu jtop[2972198]: [WARNING] jtop.service - KeyboardInterrupt, SystemExit interrupt
Jan 07 15:45:30 ubuntu jtop[2972198]: [INFO] jtop.service - FORCE jtop timer thread close
Jan 07 15:45:30 ubuntu jtop[2972190]: [INFO] jtop.__main__ - Close service by signal 15
Jan 07 15:45:30 ubuntu jtop[2972181]: [INFO] jtop.__main__ - Close service by signal 15
Jan 07 15:45:30 ubuntu jtop[2972181]: [INFO] jtop.service - Service closed
Jan 07 15:45:30 ubuntu systemd[1]: jtop.service: Succeeded.
Jan 07 15:45:30 ubuntu systemd[1]: Stopped jtop service.
Jan 07 15:45:31 ubuntu systemd[1]: Started jtop service.
Jan 07 15:45:31 ubuntu jtop[2972418]: [INFO] jtop.service - jetson_stats 4.2.4 - server loaded
[... cut ...]
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF    ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ÿÿ..............
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 33    ...............3

------------------
I2C-1:
FAIL
------------------
I2C-2:
FAIL
------------------
I2C-7:
FAIL

Log from jtop 4.2.4
```

rbonghi, 2024-01-08:
Excellent, thank you!
I would like to figure out where is locate the GPU. Can you share the output from this script?
This script (I hope) helps me to understand and fix jtop :-)

```python
import os
igpu_path = "/sys/class/devfreq/"

for item in os.listdir(igpu_path):
    item_path = os.path.join(igpu_path, item)
    if os.path.isfile(item_path) or os.path.islink(item_path):
        # Check name device
        name_path = "{item}/device/of_node/name".format(item=item_path)
        if os.path.isfile(name_path):
            # Decode name
            with open(name_path, 'r') as f:
                name = f.readline().rstrip('\x00')
            # path and file
            print("Path: {}".format(name_path))
            print("{}".format(name))
```

Thank you in advance

engrjislam, 2024-01-12:
Hi @rbonghi,

Sorry for the delay! I found the following output for the above code:

[//]: # "[image]"

[image]