rbonghi/jetson_stats #292: AGX Orin and JetPack 5.0.2 (rev. 1) has tiny issues

rowen9, 2022-11-03:
**Describe the bug**
1. JetPack version is not correct. (Jetpack UNKNOWN)
According to nvidia developer forum, I install jetpack 5.0.2 (rev.1) and **AGX Orin overlay image**
Maybe it is because of this(It's unlikely),
Device name and L4T version is OK, buf Jetpack is not

- refer
- [https://developer.nvidia.com/embedded/jetson-linux-r351](https://developer.nvidia.com/embedded/jetson-linux-r351)
- [https://docs.nvidia.com/jetson/archives/r35.1/DeveloperGuide/text/IN/QuickStart.html](https://docs.nvidia.com/jetson/archives/r35.1/DeveloperGuide/text/IN/QuickStart.html)

[image]

2. It's not bug, but a good part to improve on the UI.
CPU order is strange (not 1, 2, 3 ... , but 1, 10, 2, 3, ..)

3. When I ran jetson samples(using CUDA GPU), There is no change in GPU gauge and temperature.
The GPU tab also looks the same, so it doesn't seem to be recognized.

[image]

4. In "CTRL" tab, there is no response when the Power buttons(such as MAXN, 15W, 30W, 50W) is pressed.
I'm sad abecasuse it was a very useful function for me to control a large amount of boards from remote location.

I hope my report will help you.
Thank you for creating a useful program.

Bobby-88, 2022-12-30:
I have the same issue. I currently have agx orin and can provide any info needed from it to help in fixing.

rbonghi, 2023-01-03:
Hi @rowen9 and @Bobby-88 

Sorry for my late reply; I am still fixing all these bugs! I really appreciate you collecting all these bugs. Thank you.

1. Will be fixed soon (Easy issue :-) )
2. It will be fixed for the next release.
3. I think is fixed, but can you help me? Try to run in another terminal `sudo tegrastats` and check if there are changes on on of these variables: **GR3D** or **GR3D2**, you should read something like: 

> GR3D X%@ Y
>           GR3D is the GPU engine.
>           X = Percent of the GR3D that is being used, relative to the current running frequency.
>           Y = GR3D frequency in megahertz

4. I notice on Jetson AGX Orin, if is already selected MAXN you cannot change without a reboot other nvpmodels, I will add a message in this case.
Try to change the `nvpmodel` from from terminal and check if you have the same issue and let me know :-)

rbonghi, 2023-01-04:
I made a new release candidate

```
sudo -H pip install jetson-stats==4.0.0rc3
```

check if all your bugs are fixed (I hope so :-) )

Bobby-88, 2023-01-04:
Thanks Raffaello. 
I tried installing it right away and it hung on 

`   Building wheel for jetson-stats (setup.py) ... -`
i tried running it verbose:

`sudo -H pip install jetson-stats==4.0.0rc3 --verbose`

and here is where it hangs:

```
 [skipped]
  running build_scripts
  creating build/scripts-3.8
  copying scripts/jetson_swap -> build/scripts-3.8
  copying scripts/jetson_release -> build/scripts-3.8
  copying scripts/jetson_config -> build/scripts-3.8
  installing to build/bdist.linux-aarch64/wheel
  running install
```

not sure, maybe it's smth on my side tho.

rbonghi, 2023-01-04:
Need a bit of time (2-3min) to uninstall and reinstall jtop, but should work.

Let me know if is installed now

Bobby-88, 2023-01-04:
It didn't work in auto mode - i've waited for 20+ min.
Instead, i've uninstalled jetson_stats manually and installed via your command - and it worked. 
Everything seems to work except jetson_clocks status. Even tho i've enabled it - jtop shows it inactive. Not a big issue tho. 
Thanks once more.

rbonghi, 2023-01-05:
Thank you! 
Good to hear now is working; I will take a look at the reason why the installation crashed.

About jetson_clock are you working with an AGX Orin?

Bobby-88, 2023-01-05:
Yes, AGX Orin dev kit 32GB

closed 2023-01-05 by rbonghi