rbonghi/jetson_stats #325: jtop 4.0.0 does not work (l4t 32.3.1)

theofficialgman, 2023-01-05:
**Describe the bug**
crashes on launch of jtop
```
jtop
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/jtop/__main__.py", line 157, in main
    loop=args.loop, seconds=LOOP_SECONDS, color_filter=color_filter)
  File "/usr/lib/python3.6/curses/__init__.py", line 94, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopgui.py", line 109, in __init__
    self.run(loop, seconds)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopgui.py", line 136, in run
    while not self.events() and self.jetson.ok(spin=True):
  File "/usr/local/lib/python3.6/dist-packages/jtop/jtop.py", line 1061, in ok
    raise ex_value
  File "/usr/local/lib/python3.6/dist-packages/jtop/jtop.py", line 149, in _load_jetson_libraries
    libraries = import_os_variables(JTOP_FOLDER + "/jetson_libraries", "JETSON_")
  File "/usr/local/lib/python3.6/dist-packages/jtop/core/common.py", line 98, in import_os_variables
    value = tup[1].strip()
IndexError: list index out of range
```

**To Reproduce**
```
sudo -H pip3 install -U jetson-stats
Collecting jetson-stats
Installing collected packages: jetson-stats
Successfully installed jetson-stats-4.0.0
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
 - Nintendo Switch (2017)
   * Jetpack 4.3 [L4T 32.3.1]
   * NV Power Mode: OC_All - Type: 4
   * jetson_stats.service: active
 - Board info:
   * SOC Family: tegra210
   * Chip ID:33
   * Module: UNKNOWN - Carrier: UNKNOWN
   * Code Name: nx
   * Boardids: 2595:0000:A0
   * CUDA GPU architecture (ARCH_BIN): NONE
   * Serial Number: NXO-347B4EFA-640A10
 - Libraries:
   * CUDA: 10.0.326
   * cuDNN: 7.6.3.28
   * TensorRT: 6.0.1.10
   * Visionworks: 1.6.0.500n
   * OpenCV: NOT_INSTALLED compiled CUDA: NO
   * VPI: NOT_INSTALLED
   * Vulkan: 1.1.70
 - jetson-stats:
   * Version 4.0.0
   * Works on Python 3.6.9

**Jetpack**
 - Jetpack: [e.g. 4.3]
 - L4T: [e.g. 5.2.1]

**Jetson-Stats**
jtop 4.0.0

theofficialgman, 2023-01-05:
works after reboot strangely

rbonghi, 2023-01-05:
make sense, I changed some env and need a reboot!
If everything works, please close this issue :-)

closed 2023-01-05 by theofficialgman