rbonghi/jetson_stats #328: Error connection in the newest release (4.0)

AmrElGendyy, 2023-01-08:
After installing jetson_stats using `sudo -H pip install -U jetson-stats` and rebooting the board, running `jtop` produces the error message 'Error connection'.

Steps to reproduce the behavior:
1 - Connect to Jetson via ssh
2 - Install jetson_stats using the above command 
3 - Reboot
4 - Run jtop

This is the log coming from `journalctl -u jetson_stats.service`

[jtop_log.txt](https://github.com/rbonghi/jetson_stats/files/10368112/jtop_log.txt)

My Setup:
NVIDIA Jetson Xavier NX 
Jetpack 4.5.1 [L4T 32.5.1]
Jetson-Stats version: 4.0.0

rbonghi, 2023-01-08:
Hi @AmrElGendyy ,

thank you for your message! I catch the issue and I fix it soon!
Just for curiosity can share me the output from:
```
sudo jetson_clocks --show
```

AmrElGendyy, 2023-01-08:
@rbonghi 

This is the output 

`SOC family:tegra194  Machine:NVIDIA Jetson Xavier NX Developer Kit
Online CPUs: 0-5
cpu0: Online=1 Governor=schedutil MinFreq=1190400 MaxFreq=1420800 CurrentFreq=1190400 IdleStates: C1=1 c6=1 
cpu1: Online=1 Governor=schedutil MinFreq=1190400 MaxFreq=1420800 CurrentFreq=1190400 IdleStates: C1=1 c6=1 
cpu2: Online=1 Governor=schedutil MinFreq=1190400 MaxFreq=1420800 CurrentFreq=1190400 IdleStates: C1=1 c6=1 
cpu3: Online=1 Governor=schedutil MinFreq=1190400 MaxFreq=1420800 CurrentFreq=1190400 IdleStates: C1=1 c6=1 
cpu4: Online=1 Governor=schedutil MinFreq=1190400 MaxFreq=1420800 CurrentFreq=1190400 IdleStates: C1=1 c6=1 
cpu5: Online=1 Governor=schedutil MinFreq=1190400 MaxFreq=1420800 CurrentFreq=1190400 IdleStates: C1=1 c6=1 
GPU MinFreq=114750000 MaxFreq=1109250000 CurrentFreq=114750000
cat: /sys/kernel/debug/bpmp/debug/clk/emc/max_rate: No such file or directory
/usr/bin/jetson_clocks: line 384: [: : integer expression expected
cat: /sys/kernel/debug/bpmp/debug/clk/emc/min_rate: No such file or directory
cat: /sys/kernel/debug/bpmp/debug/clk/emc/max_rate: No such file or directory
cat: /sys/kernel/debug/bpmp/debug/clk/emc/rate: No such file or directory
cat: /sys/kernel/debug/bpmp/debug/clk/emc/mrq_rate_locked: No such file or directory
EMC MinFreq= MaxFreq= CurrentFreq= FreqOverride=
Fan: PWM=0
NV Power Mode: MODE_15W_6CORE
`

rbonghi, 2023-01-08:
I see a bug on jetson_clocks, and doesn't read the status from the EMC.
I add an extra check before uploading this data!
Thank you!

commit 948a3487 (2023-01-08): Fix #328
closed 2023-01-25 by rbonghi