rbonghi/jetson_stats #330: Error with latest release version of jtop

harshad-0302, 2023-01-10:
Hi,

After updating jetson-stats to the latest version using commands:
 sudo -H pip install -U jetson-stats  Or
 sudo -H pip3 install -U jetson-stats. I am getting issues, in one of mine nvidia jetson I got issue with jetson_clocks and then I tried to install it in another board and in that when I rebooted after installing jetson-stats My board is not booting now.

Request you to please check the latest version, there was no such issues in the previous versions. Also please test the release version before rolling it out to others.

rbonghi, 2023-01-10:
Hi @harshad-0302 ,
Thank you for your issue, and I refer this issue also with #290 

I have some questions for you to better understand and fix for the next releases:
* Which ubuntu distribution are you using?
* Do you have access to: 
  * `/usr/bin/tegrastats`
  * `/usr/bin/jetson_clocks`

harshad-0302, 2023-01-10:
This issues are only faced in the latest version 4.0.0 by me . Previously i was using version 3.1.4 but there was no such issue in that just i was not able to run jtop without sudo there.

rbonghi, 2023-01-10:
I will investigate the reason for your issue to fix in future release

Meanwhile to downgrade
```
sudo -H pip3 install --no-cache-dir jetson-stats==3.1.4
```
And reboot your board

harshad-0302, 2023-01-10:
Hi,

I am connecting to the target remotely via ssh using vs code tool on my windows laptop.

harshad-0302, 2023-01-10:
I have downgraded to version 3.1.4 , Now it is working fine just one issue that I am unable to run jtop without sudo.
Please check this things.

commit 2941a5f1 (2023-01-17): Test fix #330
closed 2023-01-25 by rbonghi