rbonghi/jetson_stats #344: Missing / incorrect power values on AGX Orin (dev kit)

vebjornbl, 2023-01-24:
**Describe the bug**
The "ALL" value seems to be the VIN_SYS_5V0 value, which is the I/O ports according to [this](https://forums.developer.nvidia.com/t/tegrastats-monitoring/217088/5?u=user62045). It should be a separate entry and "ALL" needs to be updated. 

**Screenshots**
Screenshot showing values in Jetson Power GUI and jtop. 
[image]

**Board**
 - SOC: tegra23x
 - ID: Not sure where to find this
 - Board(s): Module: P3701-0000 - Carrier: P3737-0000
 - Code Name: concord

**Jetpack**
 - Jetpack: 5.0.2 GA
 - L4T: 35.1.0

**Jetson-Stats**
 - Version: 4.0.0

rbonghi, 2023-01-24:
Hi @vebjornbl ,

Thank you for this message and for pointing me to this issue.
I check the reason why doesn't appear on jtop and I keep you posted

rbonghi, 2023-01-24:
Find out why it was missing (ref #51)
It will be fixed in my next coming release! Thank you for noticing this bug!

[image]

commit 7da935e9 (2023-01-24): Fix #344
closed 2023-01-25 by rbonghi