rbonghi/jetson_stats #348: Impossible to upgrade jetson-stats latest release

JNaranjo-Alcazar, 2023-01-26:
**Describe the bug**
Impossible to install

**To Reproduce**
```
sudo -H pip3 install --upgrade jetson-stats
[sudo] password for orin: 
Collecting jetson-stats
  Downloading jetson-stats-4.1.0.tar.gz (93 kB)
     |████████████████████████████████| 93 kB 644 kB/s 
ERROR: Exception:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/_internal/cli/base_command.py", line 186, in _main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/_internal/commands/install.py", line 357, in run
    resolver.resolve(requirement_set)
  File "/usr/lib/python3/dist-packages/pip/_internal/legacy_resolve.py", line 177, in resolve
    discovered_reqs.extend(self._resolve_one(requirement_set, req))
  File "/usr/lib/python3/dist-packages/pip/_internal/legacy_resolve.py", line 333, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "/usr/lib/python3/dist-packages/pip/_internal/legacy_resolve.py", line 270, in _get_abstract_dist_for
    skip_reason = self._check_skip_installed(req)
  File "/usr/lib/python3/dist-packages/pip/_internal/legacy_resolve.py", line 228, in _check_skip_installed
    req_to_install.check_if_exists(self.use_user_site)
  File "/usr/lib/python3/dist-packages/pip/_internal/req/req_install.py", line 443, in check_if_exists
    self.satisfied_by = pkg_resources.get_distribution(str(no_marker))
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 514, in get_distribution
    dist = get_provider(dist)
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 386, in get_provider
    return working_set.find(moduleOrReq) or require(str(moduleOrReq))[0]
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 956, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 815, in resolve
    dist = self._resolve_dist(
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 844, in _resolve_dist
    env = Environment(self.entries)
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 1044, in __init__
    self.scan(search_path)
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 1077, in scan
    self.add(dist)
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 1096, in add
    dists.sort(key=operator.attrgetter('hashcmp'), reverse=True)
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 2631, in hashcmp
    self.parsed_version,
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 2678, in parsed_version
    self._parsed_version = parse_version(self.version)
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/_vendor/packaging/version.py", line 266, in __init__
    raise InvalidVersion(f"Invalid version: '{version}'")
```

### Board

- SOC: Orin

### Jetpack
NVIDIA NVIDIA Orin Jetson-Small Developer Kit
 L4T 34.1.1 [ JetPack 5.0.1 DP ]
   Ubuntu 20.04.5 LTS
   Kernel Version: 5.10.65-tegra
 CUDA NOT_INSTALLED
   CUDA Architecture: NONE
 OpenCV version: 4.5.4
   OpenCV Cuda: NO
 CUDNN: 8.3.2.49
 TensorRT: 8.4.0.11
 Vision Works: NOT_INSTALLED
 VPI: 2.0.14
 Vulcan: 1.3.203

rbonghi, 2023-01-26:
Hi @JNaranjo-Alcazar 

Thank you for your issue,
can you try again using the command below if fail show me the error

```
sudo pip3 install --no-cache-dir -v -U jetson-stats
```

JNaranjo-Alcazar, 2023-01-26:
```
ERROR: Exception:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/_internal/cli/base_command.py", line 186, in _main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/_internal/commands/install.py", line 357, in run
    resolver.resolve(requirement_set)
  File "/usr/lib/python3/dist-packages/pip/_internal/legacy_resolve.py", line 177, in resolve
    discovered_reqs.extend(self._resolve_one(requirement_set, req))
  File "/usr/lib/python3/dist-packages/pip/_internal/legacy_resolve.py", line 333, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "/usr/lib/python3/dist-packages/pip/_internal/legacy_resolve.py", line 270, in _get_abstract_dist_for
    skip_reason = self._check_skip_installed(req)
  File "/usr/lib/python3/dist-packages/pip/_internal/legacy_resolve.py", line 228, in _check_skip_installed
    req_to_install.check_if_exists(self.use_user_site)
  File "/usr/lib/python3/dist-packages/pip/_internal/req/req_install.py", line 443, in check_if_exists
    self.satisfied_by = pkg_resources.get_distribution(str(no_marker))
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 514, in get_distribution
    dist = get_provider(dist)
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 386, in get_provider
    return working_set.find(moduleOrReq) or require(str(moduleOrReq))[0]
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 956, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 815, in resolve
    dist = self._resolve_dist(
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 844, in _resolve_dist
    env = Environment(self.entries)
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 1044, in __init__
    self.scan(search_path)
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 1077, in scan
    self.add(dist)
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 1096, in add
    dists.sort(key=operator.attrgetter('hashcmp'), reverse=True)
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 2631, in hashcmp
    self.parsed_version,
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 2678, in parsed_version
    self._parsed_version = parse_version(self.version)
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/_vendor/packaging/version.py", line 266, in __init__
    raise InvalidVersion(f"Invalid version: '{version}'")
pkg_resources.extern.packaging.version.InvalidVersion: Invalid version: '0.23ubuntu1'
```

rbonghi, 2023-01-26:
seems this error is coming from pip and not from jetson-stats. I found an issue here https://github.com/pypa/setuptools/issues/3772

I think if you downgrade setuptools from version 66 to 65 should works

rbonghi, 2023-01-26:
meanwhile I investigate how to fix it

JNaranjo-Alcazar, 2023-01-26:
Now, it works. Following your guidelines

rbonghi, 2023-01-26:
Great!
I pin this issue, I think jetson-stats need some updates for the future

closed 2023-01-26 by rbonghi