rbonghi/jetson_stats #350: [AGX Orin] GPU temperature below 45C is shown as -256.0C

vebjornjr, 2023-01-30:
**Describe the bug**
GPU temperature below 45C is shown as -256.0C.

**Screenshots**
[image]

**Additional context**
### Board

- SOC: tegra23x
- Board(s): p3701-0000
- Code Name: Concord

### Jetpack

- Jetpack: 5.0.2 GA
- L4T: 35.1.0

### Jetson-Stats

- Version: 4.1.0

rbonghi, 2023-01-30:
Hi @vebjornjr ,

Thank you for your issue, but unfortunately, I cannot fix it. It's coming from **tegrastats**; if you run `sudo tegrastats`, you will read the same number.

I'm still working to avoid tegrastats, but it's not a feature that will be available soon.

vebjornjr, 2023-01-30:
Ah, so it's Nvidia's problem. I understand. By the way, thank you for this wonderful program!

rbonghi, 2023-01-30:
Yes, I found a way to avoid, but I'm still fixing other features.

I think, but I didn't try yet, on the Jetpack 5.1 should be fixed.

Ah, when you read the -256°C there is other strange stuff happening on tegratstas, like the frequency of the GPU is 0.
All these buggy errors disappear if the GPU is working.

closed 2023-01-30 by rbonghi