rbonghi/jetson_stats #357: Hardware Missing

nsercproject, 2023-02-02:
Please update jetson-stats with this board

### Board

 - Model: lanai-3636
 - BoardIDs: p3636-0001-A0
 - Module: **MISSING**
 - SoC: tegra186
 - CUDA Arch BIN: 6.2
 - L4T: 32.6.1
 - Jetpack: 4.6

### Jetson-Stats

 - Version: 4.1.2

### RAW Data

File from `--error-log` attached

rbonghi, 2023-02-02:
Hi @nsercproject 

Thank you for your issue! Can you share for me the output from:

```
i2cdetect -y 7
```

And

```
i2cdump -y 7 0x50
```

Thank you in advance

nsercproject, 2023-02-03:
***@***.***:~$ i2cdetect -y 7
Warning: Can't use SMBus Quick Write command, will skip some addresses
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:
10:
20:
30: -- -- -- -- -- -- -- --
40:
50: 50 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60:
70:

***@***.***:~$ i2cdump -y 7 0x50
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 01 00 0d 00 34 0e 01 00 03 48 00 00 00 00 00 00    ?.?.4??.?H......
10: 00 01 00 00 36 39 39 2d 31 33 36 33 36 2d 30 30    .?..699-13636-00
20: 30 31 2d 33 30 30 20 48 2e 30 00 00 00 00 00 00    01-300 H.0......
30: 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
40: ff ff ff ff 00 dd 55 2d b0 48 31 34 32 34 32 32    .....?U-?H142422
50: 31 30 31 35 32 31 36 00 00 00 00 00 00 00 00 00    1015216.........
60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
90: 00 00 00 00 00 00 4e 56 43 42 1c 00 4d 31 00 00    ......NVCB?.M1..
a0: ff ff ff ff ff ff ff ff ff ff ff ff 00 dd 55 2d    .............?U-
b0: b0 48 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ?H..............
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 c6    ...............?

Thanks,
Pablo

rbonghi, 2023-02-03:
Thanks!
I catch my bug! I will fix before the end of this week :-)

nsercproject, 2023-02-06:
[image]

Thanks for your quick response!

closed 2023-02-06 by nsercproject