rbonghi/jetson_stats #358: External dependencies required to compile smbus

robcowie, 2023-02-04:
Hi,

Am I correct in thinking that gcc is required on the device in order to build [smbus](https://pypi.org/project/smbus/)?

Build tools aren't installed by default on Jetson devices and I'd like to avoid installing a compiler if I can. One option would be to switch to [smbus2](https://pypi.org/project/smbus2/) which purports to be a pure python drop-in replacement for smbus.

In my limited testing jetson-stats does seem to work well with smbus2. Any thoughts on making the switch?

Cheers and thanks!

rbonghi, 2023-02-04:
Hi @robcowie ,

I never tried smbus2 , I quickly check and if works I move out on smbus2

rbonghi, 2023-02-04:
The first tests look promising :-)

closed 2023-02-05 by rbonghi

rbonghi, 2023-02-05:
New version released :-)
Thank you for your hint

robcowie, 2023-02-05:
Thanks very much for the swift work! This will simplify our device provisioning a little!

rbonghi, 2023-02-05:
You are welcome!
For next major release, I'm still improving the jtop API for CPU, memory, and GPU.
This new feature will optimize the analysis. Keep ready for a new release :-)