rbonghi/jetson_stats #362: jtop shows 2 GPUs

lakshanthad, 2023-02-07:
When running jtop on Jetson Orin NX 16GB, jtop shows 2 GPUs. Does it relate to the 2 GPC on the GPU?

[image]

rbonghi, 2023-02-07:
Hi @lakshanthad ,

Thank you for your issue!
I don't have an Orin NX, and I cannot test it either. I was looking at the documentation seems are the 2 GPCs, I don't have a confirmation (below is the reference that I found)

now jtop is working reading the output from `sudo tegrastats` and in this case one of the outputs you can see is something like: **GR3D_FREQ 0%@114 GR3D2_FREQ 0%@114**

Can I ask you an help? Can you share the output from the command below? (I'm working for a new jtop version tegrastatsless and this information are helpful to understand how works #360)

```
sudo tegrastats
```

```
cat /tmp./tstats.txt
```

Also a screenshot from page 5 and 7
( think you have some warning about the Jetson detection)

[image]
https://developer.download.nvidia.com/assets/embedded/secure/jetson/orin_nx/docs/Jetson_Orin_NX_DS-10712-001_v0.5.pdf?_sOvbfNIrREaSs3FzaY1aEfqXcUsB-63ICdM4M7IZ3s9rwyLpU0tieEMDZdueKTiPEreo2gq_x8ie2LnInl1UjN6ydcN_i5PNvLHO59YAE5eZnVvgeCjna3ajYHiDO-NuVbDD-7hduoQBWH_DCBM8XsmGs8R2nNULC3iqWgu_ztgbogTRceH7HekqPMMbQ==&t=eyJscyI6ImdzZW8iLCJsc2QiOiJodHRwczovL3d3dy5nb29nbGUuY29tLyJ9

lakshanthad, 2023-02-08:
Hello @rbonghi,

Thank you for your time in looking into this.

As you requested:

1, Output of **sudo tegrastats**

```sh
lakshanthad@lakshanthad-desktop:~$ sudo tegrastats
02-08-2023 20:23:24 RAM 12324/14486MB (lfb 298x4MB) SWAP 0/7243MB (cached 0MB) CPU [100%@1984,100%@1984,100%@1984,100%@1984,100%@1984,100%@1984,100%@1984,100%@1984] EMC_FREQ 24%@2133 GR3D_FREQ 99%@917 GR3D2_FREQ 99%@0 VIC_FREQ 115 APE 174 CV0@56.937C CPU@58.343C SOC2@55.625C SOC0@54.937C CV1@54.093C GPU@60.125C tj@60.125C SOC1@55.156C CV2@53.812C VDD_IN 24719mW/24719mW VDD_CPU_GPU_CV 18649mW/18649mW VDD_SOC 2083mW/2083mW
02-08-2023 20:23:25 RAM 12324/14486MB (lfb 298x4MB) SWAP 0/7243MB (cached 0MB) CPU [100%@1984,100%@988,100%@1984,100%@1984,100%@1984,100%@1984,100%@1984,100%@1984] EMC_FREQ 20%@3199 GR3D_FREQ 99%@917 GR3D2_FREQ 99%@0 VIC_FREQ 115 APE 174 CV0@57.093C CPU@58.968C SOC2@55.937C SOC0@55.281C CV1@54.343C GPU@60.156C tj@60.156C SOC1@55.562C CV2@54.093C VDD_IN 24719mW/24719mW VDD_CPU_GPU_CV 17089mW/17869mW VDD_SOC 3084mW/2583mW
02-08-2023 20:23:26 RAM 12325/14486MB (lfb 298x4MB) SWAP 0/7243MB (cached 0MB) CPU [100%@1984,100%@1984,100%@1984,100%@1984,100%@1984,100%@1984,100%@1984,100%@1984] EMC_FREQ 27%@2133 GR3D_FREQ 99%@917 GR3D2_FREQ 99%@0 VIC_FREQ 115 APE 174 CV0@57.875C CPU@58.906C SOC2@56.093C SOC0@55.375C CV1@54.593C GPU@60.562C tj@60.562C SOC1@55.75C CV2@54.562C VDD_IN 24719mW/24719mW VDD_CPU_GPU_CV 18490mW/18076mW VDD_SOC 2163mW/2443mW
```

2, Output of **cat /tmp./tstats.txt**

```sh
lakshanthad@lakshanthad-desktop:~$ cat /tmp./tstats.txt
cat: /tmp./tstats.txt: No such file or directory
```

3, Screenshot of Page 5
[image]

4, Screenshot of Page 7
[image]

Please let me know if you need further info from me.

I have done the above while running a CPU, GPU stress test.

lakshanthad, 2023-02-08:
> I will fix the main page to make it more readable with a Xavier NX :-)

You mean Orin NX?

lakshanthad, 2023-02-08:
> I mistyped a command, can you write again this command?
> 
> ```
> cat /tmp/.tstats.txt
> ```

There is no such file inside /tmp
[image]

rbonghi, 2023-02-09:
> You mean Orin NX?

yes, you are right!

> There is no such file inside /tmp

I notice this "magic" file appears after running tegrastats, but never mind, I figured out the expected output.

About the two GPU you see on jtop, I suspect they are the 2 GPC units. You can see on your tegrastats share shared with me:

```
lakshanthad@lakshanthad-desktop:~$ sudo tegrastats
02-08-2023 20:23:24 RAM 12324/14486MB (lfb 298x4MB) SWAP 0/7243MB (cached 0MB) CPU [100%@1984,100%@1984,100%@1984,100%@1984,100%@1984,100%@1984,100%@1984,100%@1984] EMC_FREQ 24%@2133 GR3D_FREQ 99%@917 GR3D2_FREQ 99%@0 VIC_FREQ 115 APE 174 CV0@56.937C CPU@58.343C SOC2@55.625C SOC0@54.937C CV1@54.093C GPU@60.125C tj@60.125C SOC1@55.156C CV2@53.812C VDD_IN 24719mW/24719mW VDD_CPU_GPU_CV 18649mW/18649mW VDD_SOC 2083mW/2083mW
```

There are: GR3D_FREQ 99%@917 GR3D2_FREQ 99%@0

these two are detected from jtop, such as a GPU or GPC. I will look into the NVIDIA Jetson documentation more and keep you posted.

Raffaello

lakshanthad, 2023-02-09:
Thank you very much for further digging into it!

I am a bit confused about the output though. It says one is at 99% running at 917MHz while the other one is at 99% running at 0.0kHz. I am not sure why the second one is 0.0kHz.

rbonghi, 2023-02-09:
It's a bug from _tegrastats_ that shows incorrect data. (for sure is not jtop, that is only reshaping this output)

Interesting story `jetson_clocks --show` show the right frequency number.

lakshanthad, 2023-02-10:
Here is an output for your reference:
[image]

rbonghi, 2023-02-10:
After a long investigation, I think all Orin doesn't really have any load from GPC but only from the GPU.

I write a hotfix for this issue. I keep you posted

lakshanthad, 2023-02-11:
Okay thanks. So it should only show one GPU right?

rbonghi, 2023-02-11:
I do a really long reverse engineering, and I'm sure there is a bug on tegrastats. All Orin have only one GPU Load and 

`GR3D_FREQ 99%@917 GR3D2_FREQ 99%@0`

Can be rewritten as:

`GR3D_FREQ <LOAD_GPU>%@<Freq_GPC1> GR3D2_FREQ <LOAD_GPU>%@<Freq_GPC2>`

Where:

LOAD_GPU = It's the same GPU load
Freq_GPC1 = Frequency GPC 1
Freq_GPC2 = Frequency GPC 2

The frequency GPCs are completely different from the GPU frequency.

I will fix all together :-)

closed 2023-02-11 by rbonghi

rbonghi, 2023-02-11:
Fixed from jtop **4.1.5**

```
sudo pip3 install -U jetson-stats
```

lakshanthad, 2023-02-12:
Thanks a lot for your efforts!

commit 8962ca88 (2023-02-11): Fix #362