rbonghi/jetson_stats #366: Jtop won't start

ooijenvangilbert, 2023-02-09:
when starting jtop i get the following message : 

The jtop.service is not active. Please run:
sudo systemctl restart jtop.service

when i enter 
sudo systemctl restart jtop.service

i get the following message : 
Failed to restart jtop.service: Unit jtop.service not found.

when i use sudo jtop --health i get Fix jetson-stats service but when i hit enter i get error : 

Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/jtop/__main__.py", line 130, in main
    jtop_config()
  File "/usr/local/lib/python3.8/dist-packages/jtop/jetson_config.py", line 226, in jtop_config
    curses.wrapper(JTOPCONFIG, JTOP_MENU)
  File "/usr/lib/python3.8/curses/__init__.py", line 105, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopguiconfig.py", line 73, in __init__
    self.loop()
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopguiconfig.py", line 153, in loop
    while not self.events():
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopguiconfig.py", line 166, in events
    status_keyboard = self.keyboard(event)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopguiconfig.py", line 200, in keyboard
    output = cmd()
  File "/usr/local/lib/python3.8/dist-packages/jtop/jetson_config.py", line 68, in fix_service
    install_service(folder, copy=copy)
  File "/usr/local/lib/python3.8/dist-packages/jtop/service.py", line 130, in install_service
    copyfile(service_package_path, service_install_path)
  File "/usr/lib/python3.8/shutil.py", line 264, in copyfile
    with open(src, 'rb') as fsrc, open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.8/dist-packages/services/jtop.service'

### Board

Jetson Xavier NX 16gb
on seeedstudio 206 carrierboard

### Jetpack

- Jetpack: 5.1
- L4T: 35.2.1 ?

### Jetson-Stats

- Version: latest

rbonghi, 2023-02-09:
Looks like, pip bad installed jetson-stats and lost a file

Can you try to:

```
sudo pip3 uninstall jetson-stats
```

```
sudo pip3 install --no-cache-dir -v -U jetson-stats
```

ooijenvangilbert, 2023-02-09:
console output for uninstall :

Found existing installation: jetson-stats 4.1.4
Uninstalling jetson-stats-4.1.4:
  Would remove:
    /usr/local/bin/jetson_config
    /usr/local/bin/jetson_release
    /usr/local/bin/jetson_swap
    /usr/local/bin/jtop
    /usr/local/jetson_stats/jtop.service
    /usr/local/jetson_stats/jtop_env.sh
    /usr/local/lib/python3.8/dist-packages/jetson_stats-4.1.4.dist-info/*
    /usr/local/lib/python3.8/dist-packages/jtop/*
Proceed (y/n)? Y
  Successfully uninstalled jetson-stats-4.1.4

Console output for install : 

Non-user install because site-packages writeable
Created temporary directory: /tmp/pip-ephem-wheel-cache-xp6z5033
Created temporary directory: /tmp/pip-req-tracker-tryvdi0c
Initialized build tracking at /tmp/pip-req-tracker-tryvdi0c
Created build tracker: /tmp/pip-req-tracker-tryvdi0c
Entered build tracker: /tmp/pip-req-tracker-tryvdi0c
Created temporary directory: /tmp/pip-install-ihxn1txg
1 location(s) to search for versions of jetson-stats:
* https://pypi.org/simple/jetson-stats/
Fetching project page and analyzing links: https://pypi.org/simple/jetson-stats/
Getting page https://pypi.org/simple/jetson-stats/
Found index url https://pypi.org/simple
Getting credentials from keyring for https://pypi.org/simple
Getting credentials from keyring for pypi.org
Starting new HTTPS connection (1): pypi.org:443
https://pypi.org:443 "GET /simple/jetson-stats/ HTTP/1.1" 200 6625
  Found link https://files.pythonhosted.org/packages/2d/d5/0db8f183b263300ac5a6f81c514b70ea8948c5f0db6d6a69ae4c30caeab3/jetson-stats-1.6.2.tar.gz#sha256=dee03f9b3a3cb3181e26ad54f72721460148169ee0f2db9eb97ad8e70e523688 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.2
  Found link https://files.pythonhosted.org/packages/ee/ef/f917a467a77762fea5aef431e96020e1677b5de202af621cc027a3b2d12e/jetson-stats-1.6.3.tar.gz#sha256=c6fef9e5cac1678c2f1e94431f34306080195c7d56485481776390f66ab25931 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.3
  Found link https://files.pythonhosted.org/packages/73/12/b6acd0f4fcd4427b9ff89b971f50da2cb0a25ced09192b5ea4aceb4e997c/jetson-stats-1.6.4.tar.gz#sha256=a15589fc8d5c2f43e1d1be36966f132a8e25adacef547a95c920107287fe7798 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.4
  Found link https://files.pythonhosted.org/packages/aa/f7/65931081ea159783c27933f1f41e2074067a23d0844e8904879e38b008db/jetson-stats-1.6.5.tar.gz#sha256=aec859a35d666fab9551853370dc2e34baa8b0ba9f397df688afe95db4e7a3f9 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.5
  Found link https://files.pythonhosted.org/packages/12/16/536096666ff809481b3b9ca85080e2011951412e75950b8d180c609c263d/jetson-stats-1.6.6.tar.gz#sha256=7789ebfdad42b10341d5c2aa3347f361d63e26f7a7a65a9c432a9fa9bbe1eb4b (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.6
  Found link https://files.pythonhosted.org/packages/70/e2/8dd2dbffa2b60fa7fc3936515a455c2208a9880f52d8c69bba09ec5b989d/jetson-stats-1.6.7.tar.gz#sha256=38724beeecdb709b9d707ad75ad145e6852516d47d1c7a21c955fbdea48384f6 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.7
  Found link https://files.pythonhosted.org/packages/a8/58/fd74cee39ff63869b11e11d22a0548a8a801b6c44d4712b2644a26c6d991/jetson-stats-1.6.8.tar.gz#sha256=dd4c216fc8adc28b66147cee30609202ab5ee643b7aee43ba3c4352dea8bc51e (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.8
  Found link https://files.pythonhosted.org/packages/00/b8/b5f8f9422d3178b884eed48df32e562212972c37afff4485ff28d0df35db/jetson-stats-1.6.9.tar.gz#sha256=96cb05deb19a221adb581107a8f6fc00592fac2e2726fa24f43bafaea6916f70 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.9
  Found link https://files.pythonhosted.org/packages/7f/88/e777cb4e02e3470b88531b58ffcfcc86adb440a09dc9ad4d6987e36466a4/jetson-stats-1.6.10.tar.gz#sha256=0ec5c97e844d995f41deb768a3879be0ed7789c22e5be7f134c4bd9f2b9ed4b5 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.10
  Found link https://files.pythonhosted.org/packages/22/de/8068116eae57f94fb7bcab86e98ec3cc2161d135ac9d51c4a8acad0e14f9/jetson-stats-1.6.11.tar.gz#sha256=48937efbaa6710c5cc9988434385c9b7231d6ac83a76627288a94f3365c8d413 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.11
  Found link https://files.pythonhosted.org/packages/a3/11/994c5d324736118f45c4afcf0a4d366ab1f1a0cd117977d14171535c7e4a/jetson-stats-1.6.12.tar.gz#sha256=0baa9ae32db9d9de33847d6642f546b68df982b7d7fa06b6a62e190ebdd5664b (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.12
  Found link https://files.pythonhosted.org/packages/42/1c/a54ce3e69422a1b627ac2e38a1b1a3d9e39f840ca5ed1d118d60c2e34aba/jetson-stats-1.6.13.tar.gz#sha256=fe81b7401ebf65388a521de56711e5108037d08366eea3497a2a39e347dd6d10 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7,  [... cut ...]
  Stored in directory: /tmp/pip-ephem-wheel-cache-xp6z5033/wheels/c5/5f/bc/2edcae9301fea21884b1e47c0cb6a3c65beddba9048f2027f9
Successfully built jetson-stats
Installing collected packages: jetson-stats
  Created temporary directory: /tmp/pip-unpacked-wheel-yv37u9cj

  changing mode of /usr/local/bin/jetson_config to 755
  changing mode of /usr/local/bin/jetson_release to 755
  changing mode of /usr/local/bin/jtop to 755
Successfully installed jetson-stats-4.1.4
Cleaning up...
  Removing source in /tmp/pip-install-ihxn1txg/jetson-stats
Removed build tracker: '/tmp/pip-req-tracker-tryvdi0c'

ooijenvangilbert, 2023-02-09:
output for journalctl -u jtop.service : 

Feb 09 16:43:46 gilbert-desktop systemd[1]: Started jtop service.
Feb 09 16:43:47 gilbert-desktop jtop[363]: [INFO] jtop.service - Running on Python: 3.8.10
Feb 09 16:43:47 gilbert-desktop jtop[363]: [INFO] jtop.core.common - fan loaded on /sys/devices/platform
Feb 09 16:43:47 gilbert-desktop jtop[363]: [INFO] jtop.core.common - jetson_clocks loaded on /usr/bin/jetson_clocks
Feb 09 16:43:48 gilbert-desktop jtop[363]: [INFO] jtop.core.engine - Special Engine group found: [dlaX]
Feb 09 16:43:48 gilbert-desktop jtop[363]: [INFO] jtop.core.engine - Special Engine group found: [pvaX]
Feb 09 16:43:48 gilbert-desktop jtop[363]: [INFO] jtop.core.engine - Engines found: [APE CVNAS DLA0 DLA1 NVDEC NVENC NVJPG PVA0 PVA1 SE VIC]
Feb 09 16:43:48 gilbert-desktop jtop[363]: [INFO] jtop.core.common - tegrastats loaded on /usr/bin/tegrastats
Feb 09 16:43:48 gilbert-desktop jtop[363]: [INFO] jtop.__main__ - jetson_stats 4.1.4 - server loaded
Feb 09 16:43:49 gilbert-desktop jtop[745]: [ERROR] jtop.core.jetson_clocks - [0] [errno:1] Error process:
Feb 09 16:43:50 gilbert-desktop jtop[745]: [ERROR] jtop.core.jetson_clocks - [1] [errno:1] Error process:
Feb 09 16:43:52 gilbert-desktop jtop[745]: [ERROR] jtop.core.jetson_clocks - [2] [errno:1] Error process:
Feb 09 16:43:53 gilbert-desktop jtop[745]: [ERROR] jtop.core.jetson_clocks - [3] [errno:1] Error process:
Feb 09 16:43:57 gilbert-desktop jtop[745]: [ERROR] jtop.core.command - Terminating process: ['/usr/bin/jetson_clocks', '--show']
Feb 09 16:43:57 gilbert-desktop jtop[745]: [WARNING] jtop.core.command - Process terminated: ['/usr/bin/jetson_clocks', '--show']
Feb 09 16:43:57 gilbert-desktop jtop[745]: [ERROR] jtop.core.jetson_clocks - [4] [errno:-1] Process does not replied in time
Feb 09 16:43:58 gilbert-desktop jtop[745]: Process JtopServer-1:
Feb 09 16:43:58 gilbert-desktop jtop[745]: Traceback (most recent call last):
Feb 09 16:43:58 gilbert-desktop jtop[745]:   File "/usr/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
Feb 09 16:43:58 gilbert-desktop jtop[745]:     self.run()
Feb 09 16:43:58 gilbert-desktop jtop[745]:   File "/usr/local/lib/python3.8/dist-packages/jtop/service.py", line 315, in run
Feb 09 16:43:58 gilbert-desktop jtop[745]:     self.jetson_clocks.initialization(self.nvpmodel)
Feb 09 16:43:58 gilbert-desktop jtop[745]:   File "/usr/local/lib/python3.8/dist-packages/jtop/core/jetson_clocks.py", line 293, in initialization
Feb 09 16:43:58 gilbert-desktop jtop[745]:     raise JtopException("I cannot initialize jetson_clocks controller")
Feb 09 16:43:58 gilbert-desktop jtop[745]: jtop.core.exceptions.JtopException: I cannot initialize jetson_clocks controller
Feb 09 16:43:59 gilbert-desktop jtop[363]: [INFO] jtop.service - Service closed
Feb 09 16:44:00 gilbert-desktop systemd[1]: jtop.service: Succeeded.
Feb 09 16:51:37 gilbert-desktop systemd[1]: Started jtop service.
Feb 09 16:51:37 gilbert-desktop systemd[4019]: jtop.service: Failed to execute command: No such file or directory
Feb 09 16:51:37 gilbert-desktop systemd[4019]: jtop.service: Failed at step EXEC spawning /usr/local/bin/jtop: No such file or directory
Feb 09 16:51:37 gilbert-desktop systemd[1]: jtop.service: Main process exited, code=exited, status=203/EXEC
Feb 09 16:51:37 gilbert-desktop systemd[1]: jtop.service: Failed with result 'exit-code'.
Feb 09 16:51:47 gilbert-desktop systemd[1]: jtop.service: Scheduled restart job, restart counter is at 1.
Feb 09 16:51:47 gilbert-desktop systemd[1]: Stopped jtop service.
Feb 09 16:51:47 gilbert-desktop systemd[1]: Started jtop service.
Feb 09 16:51:48 gilbert-desktop jtop[4027]: [INFO] jtop.service - Running on Python: 3.8.10
Feb 09 16:51:48 gilbert-desktop jtop[4027]: [INFO] jtop.core.common - fan loaded on /sys/devices/platform
Feb 09 16:51:48 gilbert-desktop jtop[4027]: [INFO] jtop.core.common - jetson_clocks loaded on /usr/bin/jetson_clocks
Feb 09 16:51:48 gilbert-desktop jtop[4027]: [INFO] jtop.core.engine - Special Engine group found: [dlaX]
Feb 09 16:51:48 gilbert-desktop jtop[4027]: [INFO] jtop.core.engine - Special Engine group found: [pvaX]
Feb 09 16:51:48 gilbert-desktop jtop[4027]: [INFO] jtop.core.engine - Engines found: [APE CVNAS DLA0 DLA1 NVDEC NVENC NVJPG PVA0 PVA1 SE VIC]
Feb 09 16:51:48 gilbert-desktop jtop[4027]: [INFO] jtop.core.common - tegrastats loaded on /usr/bin/tegrastats
Feb 09 16:51:48 gilbert-desktop jtop[4027]: [INFO] jtop.__main__ - jetson_stats 4.1.4 - server loaded
Feb 09 16:51:49 gilbert-desktop jtop[4046]: [INFO] jtop.core.fan - Mode set system status=False
~

ooijenvangilbert, 2023-02-10:
it turns out my linux environment was bad. Thank you for your quick responses Raffaello

closed 2023-02-10 by ooijenvangilbert

rbonghi, 2023-02-10:
Happy to hear :-)

ltremblay00, 2024-01-24:
> it turns out my linux environment was bad. Thank you for your quick responses Raffaello

@ooijenvangilbert did you find out why your linux environment was bad? Or, at least, what you did to fix it? I am having the same issues. Thanks.

rbonghi, 2024-01-24:
Did you update jetson-stats with the latest version, if you have the same bug, can you please me attach the output from:
```
sudo pip3 install --no-cache-dir -v -U jetson-stats
```

```
journalctl -u jtop.service -n 100 --no-pager
```

ltremblay00, 2024-01-24:
### Problem

Running `jtop` gives

> The jtop.service is not active. Please run:
> sudo systemctl restart jtop.service

And doing `sudo systemctl restart jtop.service` gives

> Failed to restart jtop.service: Unit jtop.service not found.

### Solution

For some reason, uninstalling and reinstalling jtop using `sudo pip3 uninstall jetson-stats` and `sudo pip3 install --no-cache-dir -v -U jetson-stats` doesn't fix the problem. Running `sudo jtop --health` shows all options with `FAIL`. I fixed it by:

1. Adding the jtop.service file manually from the [jtop.service file on GitHub](https://github.com/rbonghi/jetson_stats/blob/34fc4c1204af7aab664e0213573909f5c890cd8a/services/jtop.service#L4) to the (new) file `/etc/systemd/service/jtop.service`.

2. Creating a jtop group manually with `sudo groupadd jtop`.

3. Adding the user to the group with `sudo usermod -aG jtop *your_username*` where *your_username* should be replaced with your actual username.

4. Reloading the daemon with `sudo systemctl daemon-reload`.

5. And restarting the jtop service with `sudo systemctl restart jtop.service` (and enabling it on reboot with `sudo systemctl enable jtop.service`).

rbonghi, 2024-01-24:
Thank you, @ltremblay00. It's definitely the jetson-stats installer.
Doesn't install properly jtop service. (this issue is coming from an update that I made to check if jeton-stats is installed in a docker environment)

I'll fix it in a day, hopefully!

Thank you very much

rbonghi, 2024-01-25:
Can you test this release candidate for me? Looks fixed @ltremblay00

```
sudo -H pip3 install -v jetson-stats==4.2.6rc1
```

if you confirm I'll make a new version stable
Otherwise can you copy and paste the output of this installation

Thank you very much

rbonghi, 2024-01-25:
Fixed with new release **4.2.6**

Update with:
```
sudo -H pip3 install -U jetson-stats
```

ltremblay00, 2024-01-25:
> Can you test this release candidate for me? Looks fixed @ltremblay00
> 
> ```
> sudo -H pip3 install -v jetson-stats==4.2.6rc1
> ```
> 
> if you confirm I'll make a new version stable Otherwise can you copy and paste the output of this installation
> 
> Thank you very much

I'm sure you're aware since you've pushed the update but yes, it is now working with 4.2.6. Thank you.

gitthnx, 2025-02-15:
for to restart a `jtop` process denying to start, following procedure was successful here (Ubuntu 18.04, Bionic, aarch64):

- (maybe not always necessary: `sudo rm -rf /run/jtop.sock`)
- `sudo systemctl restart jetson_stats.service`
- `sudo jtop`