rbonghi/jetson_stats #372: Failed to restart jetson_stats.service: Unit jetson_stats.service not found.

happyEveryDay1005, 2023-02-16:
**Describe the bug**
Terminal:
$ sudo systemctl restart jetson_stats.service
Failed to restart jetson_stats.service: Unit jetson_stats.service not found.

when I reboot,a message pops up on the interface:
Error found when loading /etc/profile:
/usr/bin/python3:No module named jtop.core.jeston_variables

### Jetpack
 Nvidia agx orin
- Jetpack: [5.0.2]

### Jetson-Stats

- Version: [4.1.5]

rbonghi, 2023-02-16:
Hi @happyEveryDay1005 ,

Thank you for your issue! It looks like it's badly installed jetson-stats

Can you try to run:

```
sudo pip3 install --no-cache-dir -v -U jetson-stats
```

If this error appears again, can you also share it here:

```
journalctl -u jtop.service
```

happyEveryDay1005, 2023-02-16:
Successfully installed jetson-stats-4.1.5

Terminal:
$ jtop
This mode  does not exists

$ journalctl -u jtop.service
-- Logs begin at Thu 2022-09-08 17:58:15 CST, end at Thu 2023-02-16 21:29:52 CS>
2月 16 21:23:07 nvidia-desktop systemd[1]: Started jtop service.
2月 16 21:23:08 nvidia-desktop jtop[419]: [INFO] jtop.service - Running on Pyth>
2月 16 21:23:08 nvidia-desktop jtop[419]: [INFO] jtop.core.common - fan loaded >
2月 16 21:23:08 nvidia-desktop jtop[419]: [INFO] jtop.core.common - jetson_cloc>
2月 16 21:23:08 nvidia-desktop jtop[419]: [INFO] jtop.core.engine - Special Eng>
2月 16 21:23:08 nvidia-desktop jtop[419]: [INFO] jtop.core.engine - Special Eng>
2月 16 21:23:08 nvidia-desktop jtop[419]: [INFO] jtop.core.engine - Engines fou>
2月 16 21:23:08 nvidia-desktop jtop[419]: [INFO] jtop.core.common - tegrastats >
2月 16 21:23:08 nvidia-desktop jtop[419]: [INFO] jtop.__main__ - jetson_stats 4>
2月 16 21:23:10 nvidia-desktop jtop[1164]: [INFO] jtop.core.fan - Mode set syst>
2月 16 21:24:27 nvidia-desktop jtop[1164]: [INFO] jtop.service - tegrastats sta>
2月 16 21:24:30 nvidia-desktop jtop[1164]: [INFO] jtop.service - tegrastats clo>
2月 16 21:24:30 nvidia-desktop jtop[1164]: [INFO] jtop.service - jetson_clocks >
2月 16 21:28:11 nvidia-desktop systemd[1]: Stopping jtop service...
2月 16 21:28:11 nvidia-desktop jtop[1164]: [INFO] jtop.__main__ - Close service>
2月 16 21:28:11 nvidia-desktop jtop[1146]: [INFO] jtop.__main__ - Close service>
2月 16 21:28:11 nvidia-desktop jtop[419]: [INFO] jtop.__main__ - Close service >
2月 16 21:28:11 nvidia-desktop jtop[1164]: [INFO] jtop.service - Force tegrasta>
2月 16 21:28:11 nvidia-desktop jtop[419]: [INFO] jtop.service - Service closed
2月 16 21:28:11 nvidia-desktop systemd[1]: jtop.service: Succeeded.
2月 16 21:28:11 nvidia-desktop systemd[1]: Stopped jtop service.
2月 16 21:28:13 nvidia-desktop systemd[1]: Started jtop service.
lines 1-23

Restart has no effect.

happyEveryDay1005, 2023-02-16:
- Logs begin at Thu 2022-09-08 17:58:15 CST, end at Thu 2023-02-16 21:29:52 CST. --
2月 16 21:23:07 nvidia-desktop systemd[1]: Started jtop service.
2月 16 21:23:08 nvidia-desktop jtop[419]: [INFO] jtop.service - Running on Python: 3.8.10
2月 16 21:23:08 nvidia-desktop jtop[419]: [INFO] jtop.core.common - fan loaded on /sys/devices/platform
2月 16 21:23:08 nvidia-desktop jtop[419]: [INFO] jtop.core.common - jetson_clocks loaded on /usr/bin/jetson_>
2月 16 21:23:08 nvidia-desktop jtop[419]: [INFO] jtop.core.engine - Special Engine group found: [dlaX]
2月 16 21:23:08 nvidia-desktop jtop[419]: [INFO] jtop.core.engine - Special Engine group found: [pvaX]
2月 16 21:23:08 nvidia-desktop jtop[419]: [INFO] jtop.core.engine - Engines found: [APE DLA0 DLA1 NVDEC NVEN>
2月 16 21:23:08 nvidia-desktop jtop[419]: [INFO] jtop.core.common - tegrastats loaded on /usr/bin/tegrastats
2月 16 21:23:08 nvidia-desktop jtop[419]: [INFO] jtop.__main__ - jetson_stats 4.1.5 - server loaded
2月 16 21:23:10 nvidia-desktop jtop[1164]: [INFO] jtop.core.fan - Mode set system status=False
2月 16 21:24:27 nvidia-desktop jtop[1164]: [INFO] jtop.service - tegrastats started 500ms
2月 16 21:24:30 nvidia-desktop jtop[1164]: [INFO] jtop.service - tegrastats close
2月 16 21:24:30 nvidia-desktop jtop[1164]: [INFO] jtop.service - jetson_clocks show closed
2月 16 21:28:11 nvidia-desktop systemd[1]: Stopping jtop service...
2月 16 21:28:11 nvidia-desktop jtop[1164]: [INFO] jtop.__main__ - Close service by signal 15
2月 16 21:28:11 nvidia-desktop jtop[1146]: [INFO] jtop.__main__ - Close service by signal 15
2月 16 21:28:11 nvidia-desktop jtop[419]: [INFO] jtop.__main__ - Close service by signal 15
2月 16 21:28:11 nvidia-desktop jtop[1164]: [INFO] jtop.service - Force tegrastats close
2月 16 21:28:11 nvidia-desktop jtop[419]: [INFO] jtop.service - Service closed
2月 16 21:28:11 nvidia-desktop systemd[1]: jtop.service: Succeeded.
2月 16 21:28:11 nvidia-desktop systemd[1]: Stopped jtop service.
2月 16 21:28:13 nvidia-desktop systemd[1]: Started jtop service.
2月 16 21:28:13 nvidia-desktop systemd[5110]: jtop.service: Failed to execute command: No such file or direc>
2月 16 21:28:13 nvidia-desktop systemd[5110]: jtop.service: Failed at step EXEC spawning /usr/local/bin/jtop>
2月 16 21:28:13 nvidia-desktop systemd[1]: jtop.service: Main process exited, code=exited, status=203/EXEC
2月 16 21:28:13 nvidia-desktop systemd[1]: jtop.service: Failed with result 'exit-code'.

rbonghi, 2023-02-16:
On which platform have you installed jetson-stats?
I read from your log "nvidia-desktop"

rbonghi, 2023-02-16:
There is something strange in this installation, looks like now is completely undetected jtop.
Try to uninstall:

```
sudo pip3 uninstall jetson-stats
sudo pip uninstall jetson-stats
```

and after retry to install jetson-stats

```
sudo pip3 install --no-cache-dir -v -U jetson-stats
```
I hope will fix it

happyEveryDay1005, 2023-02-17:
The platform is Nvidia AGX orin.
I did it, but it didn't work.
Terminal:
$ jtop
This mode  does not exists
$ sudo systemctl restart jetson_stats.service
Failed to restart jetson_stats.service: Unit jetson_stats.service not found.

happyEveryDay1005, 2023-02-17:
$ cat /etc/nv_tegra_release
# R35 (release), REVISION: 1.0, GCID: 31346300, BOARD: t186ref, EABI: aarch64, DATE: Thu Aug 25 18:41:45 UTC 2022
$ sudo apt -a show nvidia-jetpack
Package: nvidia-jetpack
Version: 5.0.2-b231
Priority: standard
Section: metapackages
Maintainer: NVIDIA Corporation
Installed-Size: 199 kB
Depends: nvidia-jetpack-runtime (= 5.0.2-b231), nvidia-jetpack-dev (= 5.0.2-b231)
Homepage: http://developer.nvidia.com/jetson
Download-Size: 29.3 kB
APT-Manual-Installed: yes
APT-Sources: https://repo.download.nvidia.com/jetson/common r35.1/main arm64 Packages
Description: NVIDIA Jetpack Meta Package

happyEveryDay1005, 2023-02-17:
Thank you very much for your help, I appreciate it. 

$ sudo pip3 install --no-cache-dir -v -U jetson-stats
Using pip 23.0 from /usr/local/lib/python3.8/dist-packages/pip (python 3.8)
Collecting jetson-stats
  Downloading jetson-stats-4.1.5.tar.gz (93 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 93.4/93.4 kB 516.5 kB/s eta 0:00:00
  Running command python setup.py egg_info
  running egg_info
  creating /tmp/pip-pip-egg-info-6s3oc6i9/jetson_stats.egg-info
  writing /tmp/pip-pip-egg-info-6s3oc6i9/jetson_stats.egg-info/PKG-INFO
  writing dependency_links to /tmp/pip-pip-egg-info-6s3oc6i9/jetson_stats.egg-info/dependency_links.txt
  writing entry points to /tmp/pip-pip-egg-info-6s3oc6i9/jetson_stats.egg-info/entry_points.txt
  writing requirements to /tmp/pip-pip-egg-info-6s3oc6i9/jetson_stats.egg-info/requires.txt
  writing top-level names to /tmp/pip-pip-egg-info-6s3oc6i9/jetson_stats.egg-info/top_level.txt
  writing manifest file '/tmp/pip-pip-egg-info-6s3oc6i9/jetson_stats.egg-info/SOURCES.txt'
  INFO:root:Generating grammar tables from /usr/lib/python3.8/lib2to3/Grammar.txt
  INFO:root:Generating grammar tables from /usr/lib/python3.8/lib2to3/PatternGrammar.txt
  reading manifest file '/tmp/pip-pip-egg-info-6s3oc6i9/jetson_stats.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no previously-included files found matching 'tox.ini'
  warning: no previously-included files found matching 'Dockerfile'
  warning: no previously-included files found matching '.dockerignore'
  no previously-included directories found matching 'tests'
  no previously-included directories found matching 'docs'
  no previously-included directories found matching 'examples'
  writing manifest file '/tmp/pip-pip-egg-info-6s3oc6i9/jetson_stats.egg-info/SOURCES.txt'
  Preparing metadata (setup.py) ... done
Requirement already satisfied: distro in /usr/lib/python3/dist-packages (from jetson-stats) (1.4.0)
Requirement already satisfied: smbus2 in /usr/local/lib/python3.8/dist-packages (from jetson-stats) (0.4.2)
Building wheels for collected packages: jetson-stats
  Running command python setup.py bdist_wheel
  running bdist_wheel
  running build
  running build_py
  INFO:root:Generating grammar tables from /usr/lib/python3.8/lib2to3/Grammar.txt
  INFO:root:Generating grammar tables from /usr/lib/python3.8/lib2to3/PatternGrammar.txt
  creating build
  creating build/lib
  creating build/lib/jtop
  copying jtop/jtop.py -> build/lib/jtop
  copying jtop/service.py -> build/lib/jtop
  copying jtop/github.py -> build/lib/jtop
  copying jtop/__main__.py -> build/lib/jtop
  copying jtop/terminal_colors.py -> build/lib/jtop
  copying jtop/jetson_config.py -> build/lib/jtop
  copying jtop/jetson_release.py -> build/lib/jtop
  copying jtop/__init__.py -> build/lib/jtop
  creating build/lib/jtop/gui
  copying jtop/gui/pcpu.py -> build/lib/jtop/gui
  copying jtop/gui/jtopguiconfig.py -> build/lib/jtop/gui
  copying jtop/gui/pall.py -> build/lib/jtop/gui
  copying jtop/gui/jtopgui.py -> build/lib/jtop/gui
  copying jtop/gui/jtopguimenu.py -> build/lib/jtop/gui
  copying jtop/gui/pgpu.py -> build/lib/jtop/gui
  copying jtop/gui/pcontrol.py -> build/lib/jtop/gui
  copying jtop/gui/__init__.py -> build/lib/jtop/gui
  copying jtop/gui/pmem.py -> build/lib/jtop/gui
  copying jtop/gui/pengine.py -> build/lib/jtop/gui
  copying jtop/gui/pinfo.py -> build/lib/jtop/gui
  creating build/lib/jtop/core
  copying jtop/core/swap.py -> build/lib/jtop/core
  copying jtop/core/cpu.py -> build/lib/jtop/core
  copying jtop/core/tegrastats.py -> build/lib/jtop/core
  copying jtop/core/jetson_variables.py -> build/lib/jtop/core
  copying jtop/core/jetson_clocks.py -> build/lib/jtop/core
  copying jtop/core/common.py -> build/lib/jtop/core
  copying jtop/core/memory.py -> build/lib/jtop/core
  copying jtop/core/jetson_libraries.py -> build/lib/jtop/core
  copying jtop/core/command.py -> build/lib/jtop/core
[... 149 lines cut ...]
  adding 'jtop/gui/pengine.py'
  adding 'jtop/gui/pgpu.py'
  adding 'jtop/gui/pinfo.py'
  adding 'jtop/gui/pmem.py'
  adding 'jtop/gui/lib/__init__.py'
  adding 'jtop/gui/lib/button.py'
  adding 'jtop/gui/lib/chart.py'
  adding 'jtop/gui/lib/common.py'
  adding 'jtop/gui/lib/linear_gauge.py'
  adding 'jetson_stats-4.1.5.dist-info/LICENSE'
  adding 'jetson_stats-4.1.5.dist-info/METADATA'
  adding 'jetson_stats-4.1.5.dist-info/WHEEL'
  adding 'jetson_stats-4.1.5.dist-info/entry_points.txt'
  adding 'jetson_stats-4.1.5.dist-info/top_level.txt'
  adding 'jetson_stats-4.1.5.dist-info/RECORD'
  removing build/bdist.linux-aarch64/wheel
  Building wheel for jetson-stats (setup.py) ... done
  Created wheel for jetson-stats: filename=jetson_stats-4.1.5-py3-none-any.whl size=123566 sha256=e4d645b1638aa146a26957abf3eaeacf47647d400e36e7c47c5e2b1e272531d7
  Stored in directory: /tmp/pip-ephem-wheel-cache-6_7bxaqb/wheels/0f/5c/33/e403f3194de418f17fbd1118c4f19478ff097a784ab0202a39
Successfully built jetson-stats
Installing collected packages: jetson-stats
  changing mode of /usr/local/bin/jetson_config to 755
  changing mode of /usr/local/bin/jetson_release to 755
  changing mode of /usr/local/bin/jtop to 755
Successfully installed jetson-stats-4.1.5

SolaZn, 2023-02-21:
Hello @happyEveryDay1005 ! Have you tried setting the relevant power mode for your Jetson using "nvpmodel -m 0" for example ? 
I also get this kind of issue when it is not set before running jtop.

happyEveryDay1005, 2023-02-22:
Hello @SolaZn ! I do not set the power mode and only use the default. I can try to set the power mode.

happyEveryDay1005, 2023-02-22:
Hello @SolaZn ! Thank you very much, I appreciate your help.  I run successfully.

SolaZn, 2023-02-22:
Hello @happyEveryDay1005 !

Happy to know that it worked out : here's NVIDIA documentation on what power modes are about and how you can choose one for your device when it is not set, depending on what performance/power ratio you are looking for.

[NVP Model clock configuration for Jetson Xavier NX series - docs.nvidia.com](https://docs.nvidia.com/jetson/archives/r34.1/DeveloperGuide/text/SD/PlatformPowerAndPerformance/JetsonXavierNxSeriesAndJetsonAgxXavierSeries.html#supported-modes-and-power-efficiency
)

Since you seem to be on Jetpack 5, I provided the link for the Jetson Xavier NX (current Nanos are not compatible with Jetpack 5).

happyEveryDay1005, 2023-02-22:
Thank you @SolaZn !

rbonghi, 2023-02-22:
> Hello @SolaZn ! I do not set the power mode and only use the default. I can try to set the power mode.

Hi @SolaZn , which error is happening?
These days I'm just rewriting the nvpmodel controller. I would like to improve this part

SolaZn, 2023-02-22:
Hi @rbonghi ! 

Yes I got the hint of what was happening because of this from @happyEveryDay1005 : 

> The platform is Nvidia AGX orin. I did it, but it didn't work. 
Terminal: $ jtop This mode does not exists 
$ sudo systemctl restart jetson_stats.service
Failed to restart jetson_stats.service: Unit jetson_stats.service not found.

What seems to be happening is that jtop.service/jetson_stats.service is crashing (and not restarting) when it is not able to read the current power mode set for the device. What I always do before installing jetson_stats is setting the power mode using nvpmodel; It is often not set automatically (or maybe the information is not available) when a device has just been flashed in my case.

In this case, I think that the power mode wasn't set and that jetson_stats.service just wasn't available at all because it tried to read the power mode and crashed.

rbonghi, 2023-02-22:
From jetson-stats version **4.1** the new server is called: `jtop.service`

you should do:
```
sudo systemctl restart jtop.service
```

I don't know why you have this old message, this is the reason that seems jetson-stats is not installed/updated

SolaZn, 2023-02-22:
Oh I see yes ! 
It was @happyEveryDay1005 that was having the issue. 
I was just sharing what worked for me when I stumbled into the same kind of issue.

rbonghi, 2023-03-06:
I'm working on releasing a new version, now in the release candidate. I think I should fix all issues. Let me know if this works
```
sudo pip3 install jetson-stats==4.2.0rc0
```
Documented bugs in this release candidate: https://github.com/rbonghi/jetson_stats/issues/383

happyEveryDay1005, 2023-03-13:
Sorry to reply so late.
The new version is good.

closed 2023-03-17 by rbonghi