rbonghi/jetson_stats #374: [4.1.5] Missing jtop.board['hardware']['BoardsIDs'] field on Jetson Xavier

SolaZn, 2023-02-20:
Hello !

**Describe the bug**
I got KeyError while going through jtop.board dict.
The "BoardIDs" Key seems to have disappeared from the "hardware" dict

Here's the return of jtop.board on my board : 
`{'hardware': {'Model': 'NVIDIA Jetson Xavier NX Developer Kit', '699-level Part Number': '[****]', 'P-Number': 'p3668-0001', 'Module': 'NVIDIA Jetson Xavier NX', 'SoC': 'tegra194', 'CUDA Arch BIN': '7.2', 'Codename': 'Jakku', 'Serial Number': '[****]', 'L4T': '32.6.1', 'Jetpack': '4.6'}, 'platform': {'Machine': 'aarch64', 'System': 'Linux', 'Distribution': 'Ubuntu 18.04 Bionic Beaver', 'Release': '4.9.253-tegra', 'Python': '3.6.9'}, 'libraries': {'CUDA': '', 'OpenCV': '', 'OpenCV-Cuda': False, 'cuDNN': '8.2.1.32', 'TensorRT': '', 'VPI': ''}}
`

**To Reproduce**
Steps to reproduce the behavior:

Parse through jtop.board['hardware'] on jtop 4.1.5

**Expected behavior**
BoardIDs was here before and did not find mention of its removal in the release

**Additional context**
Model: NVIDIA Jetson Xavier NX Developer Kit - Jetpack 4.6 [L4T 32.6.1]
NV Power Mode: MODE_10W_4CORE - Type: 4
Hardware:
 - Module: Not available
 - SoC: tegra194
 - CUDA Arch BIN: 7.2
 - Codename: Jakku
 - Serial Number: 1421821014784
Platform:
 - Machine: aarch64
 - System: Linux
 - Distribution: Ubuntu 18.04 Bionic Beaver
 - Release: 4.9.253-tegra
 - Python: 3.6.9
jtop:
 - Version: 4.1.5
 - Service: Active
Libraries:
 - CUDA not installed!
 - cuDNN: 8.2.1.32
 - TensorRT: Not installed
 - VPI: Not installed
 - OpenCV not installed!

### Jetson-Stats

- Version: 4.1.5

rbonghi, 2023-02-20:
Hi @SolaZn ,

Thank you for working with jtop :-)

If you don't see `BoardIDs` on `jetson.board` means it is a variable not available on your configuration.
Jtop to fill this field, jtop read this file: `/proc/device-tree/nvidia,boardids` and I think this file is missing.

BoardIDs is an _optional_ field, the fields always available on jetson are:

- Model
- 699-level Part Number
- P-Number
- Module
- SoC
- CUDA Arch BIN
- Serial Number
- L4T (Linux for Tegra)
- Jetpack

I hope I helped you :-)

> **Note**: from jtop 4.2.0 I will make some changes to the API (I will share some posts on my social and will be in release candidate for few weeks before the official release)

SolaZn, 2023-02-20:
Hello @rbonghi ! 

Thank you for your work on jtop and your quick response.
I do understand how it works now, thank you very much.

I am closing the issue since it's not a bug but a feature.

closed 2023-02-20 by SolaZn