rbonghi/jetson_stats #376: jetson_release -v shows Module as MISSING

SolaZn, 2023-02-20:
Hi again @rbonghi !

**Describe the bug**
Just found something strange when using jetson_release -v on my Jetson Xavier NX using L4T 32.6.1
I do not know if it counts as a Hardware missing since the hardware is correctly shown on jtop GUI

**Screenshots**

[image]

**Expected behavior**
As shown in INFO tab of jtop 

[image]

**Additional context**

Model: NVIDIA Jetson Xavier NX Developer Kit - Jetpack 4.6 [L4T 32.6.1]
NV Power Mode: MODE_10W_4CORE - Type: 4
Hardware:
 - Module: Not available
 - SoC: tegra194
 - CUDA Arch BIN: 7.2
 - Codename: Jakku
 - Serial Number: [****]
Platform:
 - Machine: aarch64
 - System: Linux
 - Distribution: Ubuntu 18.04 Bionic Beaver
 - Release: 4.9.253-tegra
 - Python: 3.6.9
jtop:
 - Version: 4.1.5
 - Service: Active

Thank you !

rbonghi, 2023-02-20:
Thank you for opening all these issues :-)

Can you try running

```
sudo jetson_release -v
```

SolaZn, 2023-02-21:
[image]

Seems like it is only showing on superuser or using jtop GUI but yes, it works now !
I do not know if there is a need to investigate further since it is not a hardware "missing" issue anymore?
Thanks for the hindsight @rbonghi !

rbonghi, 2023-02-21:
I think I got the reason, and I will fix it in the next release.

rbonghi, 2023-03-03:
Can you share me the output from:

```
export | grep JETSON
```

SolaZn, 2023-03-06:
Hello, 

Here's the output of the command :

> declare -x JETSON_CODENAME="Jakku"
declare -x JETSON_CUDA_ARCH_BIN="7.2"
declare -x JETSON_JETPACK="4.6"
declare -x JETSON_L4T="32.6.1"
declare -x JETSON_MODEL="NVIDIA Jetson Xavier NX Developer Kit"
declare -x JETSON_MODULE=""
declare -x JETSON_SERIAL_NUMBER="(***)"
declare -x JETSON_SOC="tegra194"

rbonghi, 2023-03-06:
The Jetson_module is entirely unavailable. Only sudo user has access.
It's more complicated to fix it.

Meanwhile, I added a message in this specific case to use sudo, and in a future release, I will fix it if needed.
I notice from Jetpack 5 is always available this data

closed 2023-03-17 by rbonghi