rbonghi/jetson_stats #380: [4.1.5] jtop.service randomly stops sending data on Jetson Xavier

SolaZn, 2023-02-27:
**Describe the bug**
Hello @rbonghi :)

I've been using jtop 4.1.5 on Xavier for a few days now and I have some kind of strange issues with jtop.service that forces me to restart the service once every few seconds.
What happens is that around 30-40 minutes after starting the service, it would just block and stop sending information to both jtop GUI and every thread using jetson_stats in general. 
For example, I tried launching the _small jetson script_ to read jtop data and it would just be able to get data once (with jetson.ok) before closing.

Restarting the service fixes the issue, so it seems that some data makes jtop.service hang or crash and it is not able to recover its former state; all consumer processes are retrieving the same last data again and again (I think this is why everything is stuck in jtop GUI...)

**To Reproduce**
Steps to reproduce the behavior:

Launch jtop 4.1.5 on a Jetson Xavier L32.6.1 - Jetpack 4.6 and wait a few minutes for it to hang indefintely 

**Screenshots**

[image]
**jtop GUI works but data is stuck (except uptime)**

**Additional context**

> ● jtop.service - jtop service
   Loaded: loaded (/etc/systemd/system/jtop.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2023-02-24 18:03:46 UTC; 2 days ago
 Main PID: 4340 (jtop)
    Tasks: 16 (limit: 4915)
   CGroup: /system.slice/jtop.service
           ├─ 4340 /usr/bin/python3 /usr/local/bin/jtop --force
           ├─ 5462 /usr/bin/python3 /usr/local/bin/jtop --force
           ├─ 5498 /usr/bin/python3 /usr/local/bin/jtop --force
           ├─14254 /bin/bash /usr/bin/jetson_clocks --show
           ├─14339 /bin/bash /usr/bin/jetson_clocks --show
           ├─14340 nvpmodel -q
           └─14341 grep NV Power Mode
Feb 24 18:03:47 jetson-549 jtop[4340]: [WARNING] jtop.service - Fan is not available on this board in paths ['/sys/devices/platform']
Feb 24 18:03:47 jetson-549 jtop[4340]: [INFO] jtop.core.common - fan loaded on /sys/devices/pwm-fan
Feb 24 18:03:47 jetson-549 jtop[4340]: [INFO] jtop.core.common - jetson_clocks loaded on /usr/bin/jetson_clocks
Feb 24 18:03:47 jetson-549 jtop[4340]: [INFO] jtop.core.engine - Special Engine group found: [dlaX]
Feb 24 18:03:47 jetson-549 jtop[4340]: [INFO] jtop.core.engine - Special Engine group found: [pvaX]
Feb 24 18:03:47 jetson-549 jtop[4340]: [INFO] jtop.core.engine - Engines found: [APE CVNAS DLA0 DLA1 NVDEC NVENC NVJPG PVA0 PVA1 SE VIC]
Feb 24 18:03:47 jetson-549 jtop[4340]: [INFO] jtop.core.common - tegrastats loaded on /usr/bin/tegrastats
Feb 24 18:03:47 jetson-549 jtop[4340]: [INFO] jtop.__main__ - jetson_stats 4.1.5 - server loaded
Feb 24 18:03:48 jetson-549 jtop[4340]: [INFO] jtop.core.fan - Mode set default status=False
Feb 27 10:01:42 jetson-549 jtop[4340]: [INFO] jtop.service - tegrastats started 500ms

status of jtop.service when it hangs

> RAM 546/7773MB (lfb 1290x4MB) SWAP 0/3887MB (cached 0MB) CPU [19%@1190,20%@1190,16%@1190,17%@1190,15%@1190,11%@1190] EMC_FREQ 0% GR3D_FREQ 0% AO@44.5C GPU@44.5C PMIC@50C AUX@43.5C CPU@47.5C thermal@44.85C

output of tegrastats when jtop is still hanged

Model: NVIDIA Jetson Xavier NX Developer Kit - Jetpack 4.6 [L4T 32.6.1]
NV Power Mode: MODE_10W_4CORE - Type: 4
Hardware:
 - P-Number: p3668-0001
 - Module: NVIDIA Jetson Xavier NX
 - SoC: tegra194
 - CUDA Arch BIN: 7.2
 - Codename: Jakku
Platform:
 - Machine: aarch64
 - System: Linux
 - Distribution: Ubuntu 18.04 Bionic Beaver
 - Release: 4.9.253-tegra
 - Python: 3.6.9
jtop:
 - Version: 4.1.5
 - Service: Active
Libraries:
 - CUDA not installed!
 - cuDNN: 8.2.1.32
 - TensorRT: Not installed
 - VPI: Not installed
 - OpenCV: 4.1.1 - with CUDA: NO

Thank you for your help on this ! 
Please let me know if you need any additional context or information.

rbonghi, 2023-02-27:
Hi @SolaZn 

When this happens, can you share the output from:

```
journalctl -u jtop.service -n 100 --no-pager
```

Meanwhile, I'm working on a new jtop release. If you want to try it, watch this issue: https://github.com/rbonghi/jetson_stats/issues/378

I changed the API. If you are working on python, please take care before running. I didn't write the new documentation.

SolaZn, 2023-02-27:
Hello,

Here's the output :

> -- Logs begin at Fri 2021-12-10 09:15:49 UTC, end at Mon 2023-02-27 13:05:44 UTC. --
Feb 24 18:03:46 jetson-549 systemd[1]: Started jtop service.
Feb 24 18:03:47 jetson-549 jtop[4340]: [INFO] jtop.service - Running on Python: 3.6.9
Feb 24 18:03:47 jetson-549 jtop[4340]: [INFO] jtop.core.common - fan loaded on /sys/devices/platform
Feb 24 18:03:47 jetson-549 jtop[4340]: [WARNING] jtop.service - Fan is not available on this board in paths ['/sys/devices/platform']
Feb 24 18:03:47 jetson-549 jtop[4340]: [INFO] jtop.core.common - fan loaded on /sys/devices/pwm-fan
Feb 24 18:03:47 jetson-549 jtop[4340]: [INFO] jtop.core.common - jetson_clocks loaded on /usr/bin/jetson_clocks
Feb 24 18:03:47 jetson-549 jtop[4340]: [INFO] jtop.core.engine - Special Engine group found: [dlaX]
Feb 24 18:03:47 jetson-549 jtop[4340]: [INFO] jtop.core.engine - Special Engine group found: [pvaX]
Feb 24 18:03:47 jetson-549 jtop[4340]: [INFO] jtop.core.engine - Engines found: [APE CVNAS DLA0 DLA1 NVDEC NVENC NVJPG PVA0 PVA1 SE VIC]
Feb 24 18:03:47 jetson-549 jtop[4340]: [INFO] jtop.core.common - tegrastats loaded on /usr/bin/tegrastats
Feb 24 18:03:47 jetson-549 jtop[4340]: [INFO] jtop.__main__ - jetson_stats 4.1.5 - server loaded
Feb 24 18:03:48 jetson-549 jtop[4340]: [INFO] jtop.core.fan - Mode set default status=False
Feb 27 10:01:42 jetson-549 jtop[4340]: [INFO] jtop.service - tegrastats started 500ms

rbonghi, 2023-02-27:
mmm weird.
There is no output about this crash.

I don't want to investigate too much. The next release **4.2.0** completely and deeply changes the jtop backend, and I hope this bug is also fixed.

SolaZn, 2023-02-27:
Okay I see, seems like something strange.
Do you think it could happen because two threads could be trying to read data from jetson_stats at the same time?

At some point, I used the Python jtop reading script to get some specific data while jtop GUI was still running.
Maybe that could have put jtop.service in an incoherent state at some point ?

Thank you anyway for your help.
Also, do you know around which time you plan to release 4.2.0 please, so that I know when to start planning the transition?

rbonghi, 2023-03-05:
I attached on discord the new documentation https://discord.gg/BFbuJNhYzS
I think I will release the new version next week

closed 2024-01-05 by rbonghi