rbonghi/jetson_stats #391: Does not work in Docker when user is not root

jrb-cobot, 2023-04-04:
## Describe the bug

I have made a non-root user in my container. Jtop does not run. When running the container as root, it works fine. 

In my Dockerfile, before making the user, I have this line which runs as root: 
`RUN python3 -m pip install -U jetson-stats`. This is the behavior I want--install as root, run as non-root. 

## To Reproduce

`docker run ...`
```bash
I can't access jtop.service.
Please logout or reboot this board
```

`docker run --user root ...`
Jtop works as normal. 

## Expected behavior

After installation, jtop should run without `sudo`, just as in the non-containerized case. 

### Board

Output from `jetson_release -v`:
```
Software part of jetson-stats 4.2.1 - (c) 2023, Raffaello Bonghi
Model: Jetson AGX Orin - Jetpack 5.0.2 GA [L4T 35.1.0]
NV Power Mode[0]: MAXN
Serial Number: [XXX Show with: jetson_release -s XXX]
Hardware:
 - 699-level Part Number: 699-13701-0000-500 L.0
 - P-Number: p3701-0000
 - Module: NVIDIA Jetson AGX Orin
 - SoC: tegra23x
 - CUDA Arch BIN: 8.7
 - Codename: Concord
Platform:
 - Machine: aarch64
 - System: Linux
 - Distribution: Ubuntu 20.04 focal
 - Release: 5.10.104-tegra
 - Python: 3.8.10
jtop:
 - Version: 4.2.1
 - Service: Active
Libraries:
 - CUDA: 11.4.239
 - cuDNN: 8.4.1.50
 - TensorRT: 5.0.2
 - VPI: 2.1.6
 - Vulkan: 1.3.203
 - OpenCV: 4.5.4 - with CUDA: NO
```

rbonghi, 2023-04-11:
Hi @jrb-cobot ,

Try to add on your docker the jtop user.

```
sudo usermod -a -G jtop  $USER
```

jrb-cobot, 2023-04-21:
Hello all, the correct answer was to add `--group-add $JTOP_GID` to my `docker run` command. The command didn't work with `--group-add jtop` for whatever reason. You can get the `JTOP_GID` by running `getent group jtop | awk -F: '{print $3}'`.

closed 2023-04-21 by jrb-cobot

rbonghi, 2023-04-21:
Thank you! I add your fix to the documentation

rbonghi, 2023-04-21:
Updated https://rnext.it/jetson_stats/docker.html

Let me know if is correct for you as well

ambigus9, 2023-07-09:
Hi, @rbonghi I just trying to make the container starts after reboot the Jetson-nano, the current configuration is:

```
docker run --detach --restart always --group-add $JTOP_GID -it -v /home/dev01/Documents/code/:/home/ -v 
/run/jtop.sock:/run/jtop.sock my-image:v60 bash
```

However I getting the following error:

> level=error msg="failed to start container" container=f3bb2343d9dd4ff00ab146bfd0f5f88d222749dd7803ac2375410f78cd1ae280 error="failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error mounting \"/run/jtop.sock\" to rootfs at \"/run/jtop.sock\": mount /run/jtop.sock:/run/jtop.sock (via /proc/self/fd/7), flags: 0x5000: not a directory: unknown: Are you trying to mount a directory onto a file (or vice-versa)? Check if the specified host path exists and is the expected type" 

Any help will be appreciated, Thanks.