rbonghi/jetson_stats #393: jtop.service could not be found

volks73, 2023-04-10:
Hello! First off, this is a great utility and package for the Jetsons. Thank you for creating and maintaining it.

I have encountered the following errors when installing jetson-stats in a Docker container and trying to run the `jtop` utility from within the Docker container.

## Describe the bug

It looks like the `jtop.service` is not being installed.

I have the following line in my `Dockerfile`:

```
RUN python3 -m pip install --no-cache-dir -U -v jetson-stats
```

I then start the docker container and run `docker exec -it jetson bash`. I get the following output:

```
root@orin-dev-kit:/app# jtop
The jtop.service is not active. Please run:
sudo systemctl restart jtop.service
root@orin-dev-kit:/app# systemctl restart jtop.service
ERROR:systemctl:Unit jtop.service could not be found.
root@orin-dev-kit:/app#
```

Contents of the `/etc/systemd/system` folder, inside the container:

```
root@orin-dev-kit:/app# ls /etc/systemd/system
default.target.wants                    nv_nvsciipc_init.service    nvramoopsconfig.service
getty.target.wants                      nv_update_verifier.service  nvs-service.service
multi-user.target.wants                 nvfb-udev.service           nvwifibt.service
nv-l4t-bootloader-config.service        nvfb.service                nvzramconfig.service
nv-l4t-usb-device-mode-runtime.service  nvgetty.service             remote-fs.target.wants
nv-l4t-usb-device-mode.service          nvmemwarning.service        sockets.target.wants
nv-late-init.service                    nvphs.service               sysinit.target.wants
nv.service                              nvpower.service             timers.target.wants
```

## Expected behavior

Start and run the `jtop` utility without errors.

### Board

Output from `jetson_release -v`:

```
root@orin-dev-kit:/app# jetson_release -v
Software part of jetson-stats 4.2.1 - (c) 2023, Raffaello Bonghi
Model: Jetson AGX Orin - Jetpack 5.0.2 GA [L4T 35.1.0]
Serial Number: [XXX Show with: jetson_release -s XXX]
Hardware:
 - 699-level Part Number: 699-13701-0000-500 J.0
 - P-Number: p3701-0000
 - Module: NVIDIA Jetson AGX Orin
 - SoC: tegra23x
 - CUDA Arch BIN: 8.7
 - Codename: Concord
Platform:
 - Machine: aarch64
 - System: Linux
 - Distribution: Ubuntu 20.04 Focal Fossa
 - Release: 5.10.104-tegra
 - Python: 3.8.10
jtop:
 - Version: 4.2.1
ERROR:systemctl:Unit jtop.service could not be found.
 - Service: Inactive
Libraries:
[... 245 lines cut ...]
#22 2.899   INFO:wheel:adding 'jtop/gui/lib/chart.py'
#22 2.899   INFO:wheel:adding 'jtop/gui/lib/colors.py'
#22 2.900   INFO:wheel:adding 'jtop/gui/lib/common.py'
#22 2.900   INFO:wheel:adding 'jtop/gui/lib/linear_gauge.py'
#22 2.901   INFO:wheel:adding 'jtop/gui/lib/process_table.py'
#22 2.901   INFO:wheel:adding 'jtop/gui/lib/smallbutton.py'
#22 2.904   INFO:wheel:adding 'jetson_stats-4.2.1.dist-info/LICENSE'
#22 2.904   INFO:wheel:adding 'jetson_stats-4.2.1.dist-info/METADATA'
#22 2.905   INFO:wheel:adding 'jetson_stats-4.2.1.dist-info/WHEEL'
#22 2.905   INFO:wheel:adding 'jetson_stats-4.2.1.dist-info/entry_points.txt'
#22 2.905   INFO:wheel:adding 'jetson_stats-4.2.1.dist-info/top_level.txt'
#22 2.906   INFO:wheel:adding 'jetson_stats-4.2.1.dist-info/RECORD'
#22 2.906   INFO:wheel:removing build/bdist.linux-aarch64/wheel
#22 2.999   Building wheel for jetson-stats (setup.py): finished with status 'done'
#22 3.001   Created wheel for jetson-stats: filename=jetson_stats-4.2.1-py3-none-any.whl size=154252 sha256=ae0009b208977137516f43a6533618e767a5128bd50e2f684bec087c22ba73c3
#22 3.001   Stored in directory: /tmp/pip-ephem-wheel-cache-i_vym0m5/wheels/ba/c9/33/0cac8e299478174eb10280e003e8eccd2ae8a53c0dc8470060
#22 3.009 Successfully built jetson-stats
#22 4.324 Installing collected packages: smbus2, distro, jetson-stats
#22 4.370   changing mode of /usr/local/bin/distro to 755
#22 4.498   changing mode of /usr/local/bin/jetson_config to 755
#22 4.499   changing mode of /usr/local/bin/jetson_release to 755
#22 4.500   changing mode of /usr/local/bin/jetson_swap to 755
#22 4.500   changing mode of /usr/local/bin/jtop to 755
#22 4.563 Successfully installed distro-1.8.0 jetson-stats-4.2.1 smbus2-0.4.2
```

rbonghi, 2023-04-11:
Hi @volks73 ,

Thank you for your post.
Have you also installed jtop on your host?

Following this guide: https://rnext.it/jetson_stats/docker.html

1. Install jetson-stats on your host
2. Install jetson-stats on your container as well
3. Pass to your container `/run/jtop.sock:/run/jtop.sock`

volks73, 2023-04-15:
I was able to use the `jtop.core.Tegrastats` class to run the `tegrastats` utility within the Balena-based Docker container without needing to install systemd, the `jtop.service`, or `jetson-stats` in the host OS.

Thank you again for the project and guidance.

rbonghi, 2023-04-16:
I'm sorry for not getting back to you sooner. You can run jtop from your host without systems using this command:

```
sudo JTOP_SERVICE=True jtop --force
```

volks73, 2023-04-16:
No worries. Good to know. Thank you for the information.

SolaZn, 2023-04-24:
Hello ! 

I do have a similar problem; here's what happens when running jtop (with or without sudo)

`jetson@jetson-486:~$ jtop
I can't access jtop.service.
Please logout or reboot this board.`

with sudo : 
`The jtop.service is not active. Please run:
sudo systemctl restart jtop.service`

Here's what happens when I try to repair jtop through `sudo jtop --health` : 
[image]

jetson@jetson-486:~$ sudo jtop --health
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/jtop/__main__.py", line 134, in main
    jtop_config()
  File "/usr/local/lib/python3.6/dist-packages/jtop/jetson_config.py", line 227, in jtop_config
    curses.wrapper(JTOPCONFIG, JTOP_MENU)
  File "/usr/lib/python3.6/curses/__init__.py", line 94, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopguiconfig.py", line 74, in __init__
    self.loop()
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopguiconfig.py", line 154, in loop
    while not self.events():
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopguiconfig.py", line 167, in events
    status_keyboard = self.keyboard(event)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopguiconfig.py", line 201, in keyboard
    output = cmd()
  File "/usr/local/lib/python3.6/dist-packages/jtop/jetson_config.py", line 79, in fix_jtop_all
    fix_service()
  File "/usr/local/lib/python3.6/dist-packages/jtop/jetson_config.py", line 69, in fix_service
    install_service(folder, copy=copy)
  File "/usr/local/lib/python3.6/dist-packages/jtop/service.py", line 124, in install_service
    copyfile(service_package_path, service_install_path)
  File "/usr/lib/python3.6/shutil.py", line 120, in copyfile
    with open(src, 'rb') as fsrc:
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.6/dist-packages/services/jtop.service'

Restarting the service or rebooting does not do anything...

Here's what the client side on my docker container has to say about this : 
`Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/jtop/jtop.py", line 1110, in start
    self._broadcaster.connect()
  File "/usr/local/lib/python3.11/multiprocessing/managers.py", line 532, in connect
    conn = Client(self._address, authkey=self._authkey)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/multiprocessing/connection.py", line 501, in Client
    c = SocketClient(address)
        ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/multiprocessing/connection.py", line 629, in SocketClient
    s.connect(address)
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/push_metrics.py", line 220, in <module>
    registry = CustomCollector()
               ^^^^^^^^^^^^^^^^^
  File "/app/push_metrics.py", line 16, in __init__
    self._jetson.start()
  File "/usr/local/lib/python3.11/site-packages/jtop/jtop.py", line 1121, in start
    raise JtopException("The jtop.service is not active. Please run:\nsudo systemctl restart jtop.service")
jtop.core.exceptions.JtopException: The jtop.service is not active. Please run:
sudo systemctl restart jtop.service
Closing jetson-stats connection...
`

The command you provided `sudo JTOP_SERVICE=True jtop --force` does work and get jtop running, although it is still not available from the docker container (that doesn't want to start anyway since there's no jtop).
When I restart the docker I use manually (the one with the `/run/jtop.sock:/run/jtop.sock` parameter), here's what happens : 

`Error response from daemon: Cannot restart container 3a100371a041: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error mounting "/run/jtop.sock" to rootfs at "/run/jtop.sock": mount /run/jtop.sock:/run/jtop.sock (via /proc/self/fd/6), flags: 0x5000: not a directory: unknown: Are you trying to mount a directory onto a file (or vice-versa)? Check if the specified host path exists and is the expected type
`

Do you happen to know what might be happening please @rbonghi ?

SolaZn, 2023-04-24:
Okay I maybe found what was happening : 
here's the output of sudo journalctl -u jtop.service -n 600 --no-pager

```
avril 24 15:48:09 jetson-486 jtop[30531]: Traceback (most recent call last):
avril 24 15:48:09 jetson-486 jtop[30531]:   File "/usr/local/bin/jtop", line 11, in <module>
avril 24 15:48:09 jetson-486 jtop[30531]:     load_entry_point('jetson-stats==4.2.1', 'console_scripts', 'jtop')()
avril 24 15:48:09 jetson-486 jtop[30531]:   File "/usr/local/lib/python3.6/dist-packages/jtop/main.py", line 98, in main
avril 24 15:48:09 jetson-486 jtop[30531]:     locale.setlocale(locale.LC_ALL, '')
avril 24 15:48:09 jetson-486 jtop[30531]:   File "/usr/lib/python3.6/locale.py", line 598, in setlocale
avril 24 15:48:09 jetson-486 jtop[30531]:     return _setlocale(category, locale)
avril 24 15:48:09 jetson-486 jtop[30531]: locale.Error: unsupported locale setting
avril 24 15:48:09 jetson-486 systemd[1]: jtop.service: Main process exited, code=exited, status=1/FAILURE
avril 24 15:48:09 jetson-486 systemd[1]: jtop.service: Failed with result 'exit-code'.
avril 24 15:48:20 jetson-486 systemd[1]: jtop.service: Service hold-off time over, scheduling restart.
avril 24 15:48:20 jetson-486 systemd[1]: jtop.service: Scheduled restart job, restart counter is at 102.
```

My locale is set to LANG=fr_FR.UTF-8
Maybe it is the source of the problem.

Changing it back to en_US.UTF-8 seems to solve the issue in a way.
Is this on purpose that French locale doesn't work ?

ProxyScci, 2023-05-03:
Having a similar issue. There is no jtop.service after install and I've uninstall and reinstalled with reboots in between to get this to work. I'm running Jetpack 5.1.0 on an Nvidia Orin with a CTI Forge Board and getting this issue. 
sudo jtop --health gives me the fix all option but that errors out as well. 

  File "/usr/local/bin/jtop", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/jtop/__main__.py", line 134, in main
    jtop_config()
  File "/usr/local/lib/python3.8/dist-packages/jtop/jetson_config.py", line 227, in jtop_config
    curses.wrapper(JTOPCONFIG, JTOP_MENU)
  File "/usr/lib/python3.8/curses/__init__.py", line 105, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopguiconfig.py", line 74, in __init__
    self.loop()
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopguiconfig.py", line 154, in loop
    while not self.events():
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopguiconfig.py", line 167, in events
    status_keyboard = self.keyboard(event)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopguiconfig.py", line 201, in keyboard
    output = cmd()
  File "/usr/local/lib/python3.8/dist-packages/jtop/jetson_config.py", line 79, in fix_jtop_all
    fix_service()
  File "/usr/local/lib/python3.8/dist-packages/jtop/jetson_config.py", line 69, in fix_service
    install_service(folder, copy=copy)
  File "/usr/local/lib/python3.8/dist-packages/jtop/service.py", line 124, in install_service
    copyfile(service_package_path, service_install_path)
  File "/usr/lib/python3.8/shutil.py", line 264, in copyfile
    with open(src, 'rb') as fsrc, open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.8/dist-packages/services/jtop.service'

lehni, 2023-10-03:
I am seeing the same problem as @ProxyScci, fresh install of Jetpack 4.6.4 on a Jetson Nano. This used to work just fine on 4.6

rbonghi, 2024-01-06:
Hi all

I'm sorry for not getting back to you sooner. I figured out a bad design for my code. I also refer #466 that has the same error.
I will release a new feature soon.

colingoodman, 2024-05-27:
Has this been resolved? @rbonghi 

From inside of a docker container:

```
# python3
Python 3.8.10 (default, Nov 22 2023, 10:22:35)
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from jtop import jtop
>>> with jtop() as jetson:
...     if jetson.ok():
...         usage = jetson.stats
...
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/dist-packages/jtop/jtop.py", line 1277, in __enter__
    self.start()
  File "/usr/local/lib/python3.8/dist-packages/jtop/jtop.py", line 1146, in start
    raise JtopException("""Mismatch version jtop service: [{service_version}] and client: [{client_version}]. Please run:\n
jtop.core.exceptions.JtopException: Mismatch version jtop service: [4.2.7] and client: [4.2.8]. Please run:

sudo systemctl restart jtop.service
```

I do have jtop installed on both the docker container and the host,
```
# jtop --version
jtop 4.2.8
```

And I did run the container with `-v /run/jtop.sock:/run/jtop.sock`

I cannot restart the jtop.service as instructed from inside of the docker container:

```
# systemctl restart jtop.service
System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: Host is down
```

And as a sanity check, it works as expected on the host machine:

```
>>> from jtop import jtop
>>> with jtop() as jetson:
...     if jetson.ok():
...             usage = jetson.stats
...
>>> usage
{'time': datetime.datetime(2024, 5, 27, 8, 58, 33, 617344), ...}
```

wohali, 2024-06-21:
@rbonghi Just to add another data point, your fix in 1688c50 does not resolve this issue. I see the fix is already in 4.2.8 and the issue persists in installing there.

I am on Jetpack 4.6.4 as well.

rbonghi, 2024-07-26:
Hi @wohali 

Could you please provide the output of your console so that we can identify the cause of the bug?

Best,
Raffaello

wohali, 2024-08-02:
Sure, I need to power back up the Jetson for testing. I'll try and get to this over the weekend. Sorry for the delay.