rbonghi/jetson_stats #398: `jetson.stats` is not a valid JSON string

sheldonmaschmeyer, 2023-04-19:
## Describe the bug

The `jetson.stats` is almost a valid JSON string but I need to make tweaks to the string for it to pass validation.

The following is done in JavaScript.

```
jetsonStats = jetsonStats.replace(/'/g, '"');
jetsonStats = jetsonStats.replace(/: datetime/g, ': "datetime');
jetsonStats = jetsonStats.replace(/\),/g, ')",');
JSON.parse(jetsonStats) // Success
```
## To Reproduce

Steps to reproduce the behavior:

1. 
```
jetson = jtop()
jetson.start()
print(jetson.stats)
```
2. Copy string to an online validator such as https://jsonlint.com/
Notice it does not pass.

3. Pass string to Node.JS
```
jetsonStats = jetsonStats.replace(/'/g, '"');
jetsonStats = jetsonStats.replace(/: datetime/g, ': "datetime');
jetsonStats = jetsonStats.replace(/\),/g, ')",');
console.debug(jetsonStats)
JSON.parse(jetsonStats)
```
4. Notice it now passes the online validator https://jsonlint.com/

### Board
Ubuntu 18.04
jetson-stats==4.1.5

sheldonmaschmeyer, 2023-04-19:
Found article https://www.influxdata.com/blog/nvidia-jetson-series-part-1-jetson-stats/

closed 2023-04-19 by sheldonmaschmeyer

rbonghi, 2023-04-20:
There is now a new function

```
jetson.json()
```

https://rnext.it/jetson_stats/reference/jtop.html#jtop.jtop.json

SheldonWBM, 2023-06-19:
Thanks :+1:

sheldonmaschmeyer, 2023-06-19:
Thanks :+1: