rbonghi/jetson_stats #402: Installed offline, permission denied error, recommended fix does not work

nsercproject, 2023-05-01:
## Describe the bug

After installing from the downloaded package I get:

> **$ jtop
> bash: /usr/local/bin/jtop: Permission denied**
> 

The file jtop permissions look as follows:

> -rwx------ 1 root root 391 May  1 16:10 /usr/local/bin/jtop

 
I wrote a bash script to install jetson-stats from the downloaded module so it can be installed while offline (no internet)
After completing the installation satisfactorily jtop only runs as the root user.
After reading the FAQ section I first tried sudo jtop --health and FIX ALL (even though there was no indication of file permission problems in the menu). This gave me this error 
  FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.6/dist-packages/services/jtop.service'  
The path /usr/local/lib/python3.6/dist-packages/services does not exist on my jetson board.

I also added the second fix suggestion for this problem to my script:
   sudo usermod -a -G jtop "$USER"

However, this did not fix the problem either.   

## To Reproduce

Steps to reproduce the behavior:

1. sudo pip3 uninstall jetson-stats
2. cd /path/to/download/dir/
3. echo "jetson-stats==4.2.1" > jtop.requirement 
4. pip3 download -r jtop.requirement
5. pip3 install --no-index --find-links /path/to/download/dir/ -r jtop.requirement
6. sudo usermod -a -G jtop "$USER"

## Screenshots
[image]

[image]
[image]

## Expected behavior

I would expect that this procedure behaves the same way as running the installation of the latest when there is internet connection:

1. sudo pip3 install -U jetson-stats
2. jtop

The jtop program starts as expected

## Additional context

Running jtop from the docker image does not work for me because I need to install it in the host and that part is the one that is not working as expected.

### Board

Output from `jetson_release -v`:

$ sudo jetson_release -v 
Software part of jetson-stats 4.2.1 - (c) 2023, Raffaello Bonghi
Model: lanai-3636 - Jetpack 4.6 [L4T 32.6.1]
NV Power Mode[3]: MAXP_CORE_ARM
Serial Number: [XXX Show with: jetson_release -s XXX]
Hardware:
 - 699-level Part Number: 699-13636-0001-300 H.0
 - P-Number: p3636-0001
 - BoardIDs: p3636-0001-A0
 - Module: NVIDIA Jetson TX2 NX
 - SoC: tegra186
 - CUDA Arch BIN: 6.2
Platform:
 - Machine: aarch64
 - System: Linux
 - Distribution: Ubuntu 18.04 Bionic Beaver
 - Release: 4.9.253-tegra
 - Python: 3.6.9
jtop:
 - Version: 4.2.1
 - Service: Active
Libraries:
 - CUDA: 10.2.300
 - cuDNN: 8.2.1.32
 - TensorRT: 8.0.1.6
 - VPI: 1.1.12
 - Vulkan: 1.2.70
 - OpenCV: 4.1.1 - with CUDA: NO

* jetson-stats version: [e.g. 1.8]  
* P-Number: [e.g. pXXXX-XXXX]
* Module: [e.g. NVIDIA Jetson XXX]
* Jetpack: [e.g. 4.3]
* L4T: [e.g. 5.2.1]

### Log from jtop.service

Attach here the output from: `journalctl -u jtop.service -n 100 --no-pager`

[jtop.service.log](https://github.com/rbonghi/jetson_stats/files/11368533/jtop.service.log)

### Log from jetson-stats installation

Attach here the output from: `sudo -H pip3 install --no-cache-dir -U jetson-stats`

$ sudo -H pip3 install --no-cache-dir -U jetson-stats
Requirement already up-to-date: jetson-stats in /usr/local/lib/python3.6/dist-packages
Requirement already up-to-date: distro in /usr/local/lib/python3.6/dist-packages (from jetson-stats)
Requirement already up-to-date: smbus2 in /usr/local/lib/python3.6/dist-packages (from jetson-stats)

rbonghi, 2024-01-05:
Hi @nsercproject 

I apologize for late reply. First of all, I suggest to install the latest version of jetson-stats 4.2.4
```
sudo pip3 install -U jetson-stats
```

I think your issue is coming for a different user installation https://rnext.it/jetson_stats/troubleshooting.html#jtop-start-only-with-sudo

Try to add manually your user to the group:
```console
sudo usermod -a -G jtop  $USER
```

nsercproject, 2024-05-01:
Thanks @rbonghi  all good now! This was fixed by with `sudo jtop --health`

nsercproject, 2024-05-01:
An update was required before the fix worked! Thanks

closed 2024-05-01 by nsercproject