rbonghi/jetson_stats #406: Lot of problems with jtop on Commercial AGX Orin 32gb

hpetty3d, 2023-05-05:
## Describe the bug

I am using a commercial model of the Jetson AGX Orin 32GB and having some trouble getting jtop to work. After following all of the usual install steps and running jtop, I am met with the message: 
```
The jtop.service is not active. Please run: 
sudo systemctl restart jtop.service
```

After running the command I get:
`Failed to restart jtop.service: Unit jtop.service not found.`

When running `sudo jtop --health` I see "FAIL Fix jetson-stats service" but the other two entries say OK. If I try to "Fix all" or navigate to the failed one I get the following error:
```
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/jtop/__main__.py", line 134, in main
    jtop_config()
  File "/usr/local/lib/python3.8/dist-packages/jtop/jetson_config.py", line 227, in jtop_config
    curses.wrapper(JTOPCONFIG, JTOP_MENU)
  File "/usr/lib/python3.8/curses/__init__.py", line 105, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopguiconfig.py", line 74, in __init__
    self.loop()
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopguiconfig.py", line 154, in loop
    while not self.events():
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopguiconfig.py", line 167, in events
    status_keyboard = self.keyboard(event)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopguiconfig.py", line 201, in keyboard
    output = cmd()
  File "/usr/local/lib/python3.8/dist-packages/jtop/jetson_config.py", line 79, in fix_jtop_all
    fix_service()
  File "/usr/local/lib/python3.8/dist-packages/jtop/jetson_config.py", line 69, in fix_service
    install_service(folder, copy=copy)
  File "/usr/local/lib/python3.8/dist-packages/jtop/service.py", line 124, in install_service
    copyfile(service_package_path, service_install_path)
  File "/usr/lib/python3.8/shutil.py", line 264, in copyfile
    with open(src, 'rb') as fsrc, open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.8/dist-packages/services/jtop.service'
```

## To Reproduce

Steps to reproduce the behavior:

Not sure if it is reproducible since I'm on a commercial model. 

## Screenshots

[image]

## Expected behavior

Jetson GUI to pop up

## Additional context

I have also tried all of the fixes I could find online and none of them seem to work.

### Board

Output from `jetson_release -v`:

I get an error after running this command:
```
Software part of jetson-stats 4.2.1 - (c) 2023, Raffaello Bonghi
Model: Jetson AGX Orin - Jetpack 5.0.2 GA [L4T 35.1.0]
Traceback (most recent call last):
  File "/usr/local/bin/jetson_release", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/jtop/jetson_release.py", line 65, in main
    name=bcolors.bold(nvpmodel_now['name']),
TypeError: 'NoneType' object is not subscriptable
```

### Log from jtop.service

Attach here the output from: `journalctl -u jtop.service -n 100 --no-pager`

```
-- Logs begin at Fri 2023-05-05 22:06:13 CST, end at Fri 2023-05-05 22:42:22 CST. --
-- No entries --
```

### Log from jetson-stats installation

Attach here the output from: `sudo -H pip3 install --no-cache-dir -U jetson-stats`

```
Requirement already up-to-date: jetson-stats in /usr/local/lib/python3.8/dist-packages (4.2.1)
Requirement already satisfied, skipping upgrade: distro in /usr/lib/python3/dist-packages (from jetson-stats) (1.4.0)
Requirement already satisfied, skipping upgrade: smbus2 in /usr/local/lib/python3.8/dist-packages (from jetson-stats) (0.4.2)
```

shawnLang, 2023-06-05:
I set the mode with sudo nvpmodel -m 0 and now it's working after a reboot.

ztbupt, 2024-01-25:
I meet the same issue with hpetty3d. The hardware is Orin NX, L4T 35.4.1.
Is there any solution?

rbonghi, 2024-01-25:
Hi @ztbupt which kind of issue do you have?

have you updated to the latest jetson-stats version?

ztbupt, 2024-01-26:
> Hi @ztbupt which kind of issue do you have?
> 
> have you updated to the latest jetson-stats version?

yes, the latest version.
Finally, I copied jetson_stats/services/jtop.service to /usr/local/lib/python3.8/dist-packages. Then it works now.

rbonghi, 2024-01-26:
@ztbupt please update to new release **4.2.6** that fix this bug

Update with:
```
sudo -H pip3 install -U jetson-stats
```

ztbupt, 2024-01-26:
> @ztbupt please update to new release **4.2.6** that fix this bug
> 
> Update with:
> 
> ```
> sudo -H pip3 install -U jetson-stats
> ```

bad news. updated to 4.2.6. but meet new issue:
:~$ jtop
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/jtop/__main__.py", line 160, in main
    curses.wrapper(JTOPGUI, jetson, pages, init_page=args.page,
  File "/usr/lib/python3.8/curses/__init__.py", line 105, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopgui.py", line 100, in __init__
    self.run(loop, seconds)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopgui.py", line 129, in run
    self.draw()
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopgui.py", line 143, in draw
    page.draw(self.key, self.mouse)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/pall.py", line 190, in draw
    func(self.stdscr, pos_y_mini_menu, column_width * nline, column_width, column_height, self.jetson)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/pengine.py", line 115, in compact_engines
    size_eng = size_table // len(row) - 1
ZeroDivisionError: integer division or modulo by zero

rbonghi, 2024-01-26:
Strang bug, can you please share with me the output from:
```
journalctl -u jtop.service -n 100 --no-pager
```
and the file from
````
jtop --error-log
```
Thank you in advance

ztbupt, 2024-02-06:
> Strang bug, can you please share with me the output from:
> 
> ```
> journalctl -u jtop.service -n 100 --no-pager
> ```
> 
> and the file from
> 
> ```
> jtop --error-log
> ```
> 
> Thank you in advance

sorry for the late response. attaching the logs.
[Uploading log_jtop.txt…]()

ztbupt, 2024-02-26:
hi rbonghi, 
Do you have a solution to my bug? Thanks in advance!

Billy2001-shadow, 2024-03-21:
> > @ztbupt please update to new release **4.2.6** that fix this bug
> > Update with:
> > ```
> > sudo -H pip3 install -U jetson-stats
> > ```
> 
> bad news. updated to 4.2.6. but meet new issue: :~$ jtop Traceback (most recent call last): File "/usr/local/bin/jtop", line 8, in sys.exit(main()) File "/usr/local/lib/python3.8/dist-packages/jtop/**main**.py", line 160, in main curses.wrapper(JTOPGUI, jetson, pages, init_page=args.page, File "/usr/lib/python3.8/curses/**init**.py", line 105, in wrapper return func(stdscr, *args, **kwds) File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopgui.py", line 100, in **init** self.run(loop, seconds) File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopgui.py", line 129, in run self.draw() File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopgui.py", line 143, in draw page.draw(self.key, self.mouse) File "/usr/local/lib/python3.8/dist-packages/jtop/gui/pall.py", line 190, in draw func(self.stdscr, pos_y_mini_menu, column_width * nline, column_width, column_height, self.jetson) File "/usr/local/lib/python3.8/dist-packages/jtop/gui/pengine.py", line 115, in compact_engines size_eng = size_table // len(row) - 1 ZeroDivisionError: integer division or modulo by zero

change "pass_orin" function in  /usr/local/lib/python3.8/dist-packages/jtop/gui/pengine.py. change to the following code:
def pass_orin(engine):
    return [
        add_engine_in_list('APE', engine, 'APE', 'APE') + add_engine_in_list('PVA0a', engine, 'PVA0', 'PVA0_CPU_AXI'),
        add_engine_in_list('DLA0c', engine, 'DLA0', 'DLA0_CORE') + add_engine_in_list('DLA1c', engine, 'DLA1', 'DLA1_CORE') + add_engine_in_list('DLA1c', engine, 'DLA', 'DLA1_CORE'),
        add_engine_in_list('NVENC', engine, 'NVENC', 'NVENC') + add_engine_in_list('NVDEC', engine, 'NVDEC', 'NVDEC'),
        add_engine_in_list('NVJPG', engine, 'NVJPG', 'NVJPG') + add_engine_in_list('NVJPG1', engine, 'NVJPG', 'NVJPG1'),
        add_engine_in_list('SE', engine, 'SE', 'SE') + add_engine_in_list('VIC', engine, 'VIC', 'VIC'),
    ]

zgfbupt, 2024-03-21:
> bad news. updated to 4.2.6. but meet new issue: :~$ jtop Traceback (most recent call last): File "/usr/local/bin/jtop", line 8, in sys.exit(main()) File [...]

change the "pass_orin" function in /usr/local/lib/python3.8/dist-packages/jtop/gui/pengine.py file. and then replace with the following code:
def pass_orin(engine):
    return [
        add_engine_in_list('APE', engine, 'APE', 'APE') + add_engine_in_list('PVA0a', engine, 'PVA0', 'PVA0_CPU_AXI'),
        add_engine_in_list('DLA0c', engine, 'DLA0', 'DLA0_CORE') + add_engine_in_list('DLA1c', engine, 'DLA1', 'DLA1_CORE') + add_engine_in_list('DLA1c', engine, 'DLA', 'DLA1_CORE'),
        add_engine_in_list('NVENC', engine, 'NVENC', 'NVENC') + add_engine_in_list('NVDEC', engine, 'NVDEC', 'NVDEC'),
        add_engine_in_list('NVJPG', engine, 'NVJPG', 'NVJPG') + add_engine_in_list('NVJPG1', engine, 'NVJPG', 'NVJPG1'),
        add_engine_in_list('SE', engine, 'SE', 'SE') + add_engine_in_list('VIC', engine, 'VIC', 'VIC'),
    ]

ztbupt, 2024-03-25:
> change the "pass_orin" function in /usr/local/lib/python3.8/dist-packages/jtop/gui/pengine.py file. and then replace with the following code: def pass_orin(engine): return [ [...]

Thank you, zgfbupt.

Linda1226, 2024-03-31:
> /usr/local/lib/python3.8/dist-packages
您的回复很有用。谢谢。

daothanh2011, 2024-05-23:
for me I tried to fix the /usr/local/lib/python3.8/dist-packages/jtop/core/nvpmodel.py file: in NVPModelService class 
command: #except (OSError, Command.CommandException):
add: except:
and it works