rbonghi/jetson_stats #409: Fan speed shows it's zero

kais-bedioui, 2023-05-10:
## Describe the bug

Hi,
I am using jtop 4.1.5 installed directly on my system which is an Nvidia Jetson AGX Orin, with **Jetpack 5.0.2 GA**
I notice that on jtop, fan speed is always zero as you can see in the picture below
[image]
with nvpmodel = 0 and jetson clocks are ON

using the python API I tried to set it manually to an arbitrary value and watch its evolution over time. Here are the results:
[image]

## To Reproduce

Steps to reproduce the behavior:

`
from jtop import jtop
with jtop() as jetson:
  print(jetson.fan.speed) # 0.3921
  print('increasing fan speed')
  jetson.fan.speed = 20
  while jetson.ok():
    print(jetson.fan.speed)
#20.0
#7.0588
#7.0588
#7.0588
#7.0588
#2.745...
#2.745...
#2.745...
#2.745...
#1.1764..
#1.1764..
#1.1764..
#1.1764..
#0.7843..
#0.7843..
#0.7843.
#0.7843..
#0.3921..
#0.3921..
#0.3921..
#0.3921..
#0.3921..
#0.3921..
`

## Screenshots

Screenshots above.

## Expected behavior

Fan should be running at a higher speed.
When value set by python API, it should stay constant at that desired value.

## Additional context

Add any other context about the problem here.

### Board

Output from `jetson_release -v`:

kais-bedioui, 2023-05-25:
@rbonghi Hi, did you have time to have a look at this? Thanks ;)

rbonghi, 2024-01-04:
Hi @kais-bedioui 

Apologize for my late reply
With jetson-stats 4.2, I added significant updates to control the fan and monitor the NVIDIA Jetson.
I suggest to update to the latest version
```
sudo pip3 install -U jetson-stats
```

closed 2024-01-04 by rbonghi