rbonghi/jetson_stats #413: Wrong display when GPU memory is just a little bit over 1G

clogwog, 2023-05-23:
## Describe the bug

When GPU memory goes up to 1 G it shows it wrongly as 101M for a bit.. Once it gets higher i think I have seen it display 1G properly.. but it must be around the chang-over from M to G.
This confused me a bit because i thought i was running light on memory, until i saw the startup sequence as shown in the video below and confirmed it with the graph on page 2(GPU)

## To Reproduce

When starting a program that will take heaps of GPU memory it goes up to 900M then it suddenly shows 101M even though the memory has still gone up.

It shows it wrong in both the 1(All) screen next to the process as well in the 2(GPU) total GPU graph

See this example https://www.youtube.com/watch?v=R_rWsqLXMfw

## Expected behavior

just showing either 1000 M or 1 G 

### Board

Output from `jetson_release -v`:
```
$ sudo jetson_release -v
Software part of jetson-stats 4.2.1 - (c) 2023, Raffaello Bonghi
Model: lanai-3636 - Jetpack 4.6.3 [L4T 32.7.3]
NV Power Mode[3]: MAXP_CORE_ARM
Serial Number: [XXX Show with: jetson_release -s XXX]
Hardware:
 - BoardIDs: p3636-0001-A0
 - Module: Not available
 - SoC: tegra186
 - CUDA Arch BIN: 6.2
 - Codename: Lanai
Platform:
 - Machine: aarch64
 - System: Linux
 - Distribution: Ubuntu 18.04 Bionic Beaver
 - Release: 4.9.299-tegra
 - Python: 3.6.9
jtop:
 - Version: 4.2.1
 - Service: Active
Libraries:
 - CUDA: 10.2.300
 - cuDNN: 8.2.1.32
 - TensorRT: 8.2
 - VPI: 1.2.3
 - Vulkan: 1.2.70
 - OpenCV: 4.1.1 - with CUDA: NO
```

  You can find this data on:
   * jetson_release -v
   * jtop (page INFO)
-->

* jetson-stats version: [e.g. 1.8]
* P-Number: [e.g. pXXXX-XXXX]
* Module: [e.g. NVIDIA Jetson XXX]
* Jetpack: [e.g. 4.3]
* L4T: [e.g. 5.2.1]

### Log from jtop.service

Attach here the output from: `journalctl -u jtop.service -n 100 --no-pager`

```
$ journalctl -u jtop.service -n 100 --no-pager
-- Logs begin at Tue 2023-05-23 22:21:48 UTC, end at Tue 2023-05-23 22:36:26 UTC. --
May 23 22:22:05 jeteye systemd[1]: Started jtop service.
May 23 22:22:05 jeteye jtop[7633]: [INFO] jtop.service - jetson_stats 4.2.1 - server loaded
[... 72 lines cut ...]
May 23 22:22:21 jeteye jtop[9508]: [INFO] jtop.core.jetson_clocks - Starting jetson_clocks in: 24.78s
May 23 22:22:21 jeteye jtop[9508]: [INFO] jtop.core.jetson_clocks - Start jetson_clocks with booting
May 23 22:22:32 jeteye jtop[9508]: [INFO] jtop.service - jtop timer thread started 1000ms
May 23 22:22:46 jeteye jtop[9508]: [WARNING] jtop.core.fan - Fan tegra_pwmfan profile temp_control already active
May 23 22:22:46 jeteye jtop[9508]: [INFO] jtop.core.jetson_clocks - jetson_clocks started
```

### Log from jetson-stats installation

Attach here the output from: `sudo -H pip3 install --no-cache-dir -U jetson-stats`

```
$ sudo -H pip3 install --no-cache-dir -U jetson-stats
/usr/lib/python3/dist-packages/secretstorage/dhcrypto.py:15: CryptographyDeprecationWarning: Python 3.6 is no longer supported by the Python core team. Therefore, support for it is deprecated in cryptography. The next release of cryptography will remove support for Python 3.6.
  from cryptography.utils import int_from_bytes
Requirement already satisfied: jetson-stats in /usr/local/lib/python3.6/dist-packages (4.2.1)
Requirement already satisfied: smbus2 in /usr/local/lib/python3.6/dist-packages (from jetson-stats) (0.4.2)
Requirement already satisfied: distro in /usr/local/lib/python3.6/dist-packages (from jetson-stats) (1.8.0)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
```

```
$ jtop -v
jtop 4.2.1
```

clogwog, 2023-05-23:
just noticed that page 4MEM shows the same as well:
**GPU Sh:  101M**

```
Model: lanai-3636 - Jetpack 4.6.3 [L4T 32.7.3]
 RAM 2.9G/3.7GB - (lfb 0x4MB)                                         RAM
                                                         ├ 3.7G  Used:    2.9G
 ▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁├ 3.5G  GPU Sh:  101M
 ████████████████████████████████████████████████████████├ 3.3G  Buffers: 26.8M
 ████████████████████████████████████████████████████████├ 3.1G  Cached:  553M
 ▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁├ 2.9G  Free:    382M
 ████████████████████████████████████████████████████████├ 2.7G  TOT:     3.7G
 ████████████████████████████████████████████████████████├ 2.5G
 ████████████████████████████████████████████████████████├ 2.3G
 ████████████████████████████████████████████████████████├ 2.1G
 ▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁▁├ 1.9G
 ████████████████████████████████████████████████████████├ 1.8G
 ████████████████████████████████████████████████████████├ 1.6G
 ████████████████████████████████████████████████████████├ 1.4G
 ████████████████████████████████████████████████████████├ 1.2G
 ████████████████████████████████████████████████████████├ 1.0G
 ████████████████████████████████████████████████████████├ 0.8G
 ████████████████████████████████████████████████████████├ 0.6G
 ████████████████████████████████████████████████████████├ 0.4G
 ████████████████████████████████████████████████████████├ 0.2G
         └ -8s       └ -6s       └ -4s       └ -2s       0 time
 Emc [204MHz::::::::::::::::::::::::::::1.6GHz] 1.6GHz   0%
 SWAP 594M/9.9G (Cached 45.5M)                              [c| clear cache]
 zram0 [P5|||     148M/478M]  zram2 [P5|||     148M/478M]
 zram1 [P5|||     148M/478M]  zram3 [P5|||     148M/478M]   [Select swap]
 swapfile [P1                               0k/8.0G] Boot
                                                            [s| Create new]
                                                            [b| on boot]
                                                            [-]  1  GB [+]
                                                            New: /swfile

 1ALL  2GPU  3CPU  4MEM  5ENG  6CTRL  7INFO  Quit                  (c) 2023, RB
```

rbonghi, 2024-01-05:
Hi @clogwog (also reply for @artur-ag and @vesselofgod #469 )

I am aware of an issue where the output plotting in my jtop new version generates errors. I have worked to reduce the occurrence of this bug. The issue stems from the use of ASCII characters to plot the output, and there are cases where there is no small or big ASCII block available to perfectly plot the output.

I'm not sure if you have any experience with ASCII coding, but I was wondering if you could help me fix it. It would be really helpful for me! I wrote and updated this code some time ago, and the plot object is available at https://github.com/rbonghi/jetson_stats/blob/master/jtop/gui/lib/chart.py

artur-ag, 2024-04-14:
From what I understand, the issue is that there's only space for 4 characters in most places of the UI, but all values between `1000M` and `1023M` take 5 characters, and are truncated to `101M`. Values higher than 1024 get correctly converted to `1.0G`. This is not an issue just with the ASCII plot, but also in the text output.

One way of fixing it would be to start using the higher unit whenever the value goes above 999, even if the value in the higher unit is not above 1 of the larger unit. Like this:
- 999 MiB is rendered as `999M`
- 1000 MiB is converted to 1000/1024 = 0.9765 MiB, and rendered as `0.97M`
- 1023 MiB is converted to 1023/1024 = 0.9990234 MiB, and rendered as `0.99M`
- 1024 MiB is exactly 1 GiB, and rendered as `1.0G`
This guarantees that the value written in those 4 characters is always correct (even if a bit ugly).

@rbonghi I'm not familiar with the code, but from what I can tell, it's not [chart.py](https://github.com/rbonghi/jetson_stats/blob/master/jtop/gui/lib/chart.py) making these conversions between units, but some other code before this, right? I see [common.py](https://github.com/rbonghi/jetson_stats/blob/master/jtop/gui/lib/common.py) has some unit conversion code. Maybe that's where this new behavior needs to be implemented (if you agree that this solution is adequate, that is).