rbonghi/jetson_stats #417: jtop closes inexpectedly when used for docker (error 127)

SolaZn, 2023-06-01:
## Describe the bug

It seems that, for no particular reason, jtop (host side) closes itself and as such, ends up propagating the issue to any docker container that is listening to the /run/jtop.sock file;
In this case, the container will abruptly exit with the following error code : 

Exited (127) or Exited (0)
"Error": "failed to create task for container: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error mounting \"/run/jtop.sock\" to rootfs at \"/run/jtop.sock\": mount /run/jtop.sock:/run/jtop.sock (via /proc/self/fd/6), flags: 0x5000: not a directory: unknown: Are you trying to mount a directory onto a file (or vice-versa)? Check if the specified host path exists and is the expected type"

On the host side, there's close to no trace of it happening, other than jtop being unavailable
"jtop is not running on this board" and 

## To Reproduce

You must be listening to a jtop.sock from inside a Docker container that has /run/jtop.sock mounted as a volume
(as mentioned in the guidelines for Docker use)

## Additional context

Add any other context about the problem here.

### Board

Output from `jetson_release -v`:

Software part of jetson-stats 4.2.1 - (c) 2023, Raffaello Bonghi
Model: NVIDIA Jetson Nano Developer Kit - Jetpack 4.6 [L4T 32.6.1]
NV Power Mode[0]: MAXN
Serial Number: [XXX Show with: jetson_release -s XXX]
Hardware:
 - 699-level Part Number: (**)
 - P-Number: p3448-0002
 - BoardIDs: p3448
 - Module: NVIDIA Jetson Nano module (16Gb eMMC)
 - SoC: tegra210
 - CUDA Arch BIN: 5.3
 - Codename: Porg
Platform:
 - Machine: aarch64
 - System: Linux
 - Distribution: Ubuntu 18.04 Bionic Beaver
 - Release: 4.9.253-tegra
 - Python: 3.6.9
jtop:
 - Version: 4.2.1
 - Service: Active
Libraries:
 - CUDA: Not installed
 - cuDNN: 8.2.1.32
 - TensorRT: Not installed
 - VPI: Not installed
 - OpenCV: Not installed

Output when I tried "fix all" on jtop --health:

`Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/jtop/__main__.py", line 134, in main
    jtop_config()
  File "/usr/local/lib/python3.6/dist-packages/jtop/jetson_config.py", line 227, in jtop_config
    curses.wrapper(JTOPCONFIG, JTOP_MENU)
  File "/usr/lib/python3.6/curses/__init__.py", line 94, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopguiconfig.py", line 74, in __init__
    self.loop()
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopguiconfig.py", line 154, in loop
    while not self.events():
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopguiconfig.py", line 167, in events
    status_keyboard = self.keyboard(event)
  File "/usr/local/lib/python3.6/dist-packages/jtop/gui/jtopguiconfig.py", line 201, in keyboard
[... 105 lines cut ...]
Jun 01 15:48:09 jetson-486 jtop[6673]: [INFO] jtop.core.temperature - Found thermal "GPU" in thermal_zone2
Jun 01 15:48:09 jetson-486 jtop[6673]: [INFO] jtop.core.temperature - Found thermal "AO" in thermal_zone0
Jun 01 15:48:09 jetson-486 jtop[6673]: [INFO] jtop.core.temperature - Found thermal "thermal" in thermal_zone5
Jun 01 15:48:09 jetson-486 jtop[6673]: [INFO] jtop.core.power - Found I2C power monitor
Jun 01 15:48:09 jetson-486 jtop[6673]: [INFO] jtop.core.fan - Fan tegra_pwmfan(1) found in /sys/class/hwmon/hwmon1
Jun 01 15:48:09 jetson-486 jtop[6673]: [INFO] jtop.core.fan - RPM tegra_pwmfan(1) found in /sys/class/hwmon/hwmon1
Jun 01 15:48:09 jetson-486 jtop[6673]: [INFO] jtop.core.fan - Fan temp controller tegra_pwmfan found in /sys/class/hwmon/hwmon1/temp_control
Jun 01 15:48:09 jetson-486 jtop[6673]: [INFO] jtop.core.jetson_clocks - jetson_clocks found in /usr/bin/jetson_clocks
Jun 01 15:48:09 jetson-486 jtop[6673]: [INFO] jtop.core.nvpmodel - nvpmodel running in [0]MAXN - Default: 0
Jun 01 15:48:09 jetson-486 jtop[6673]: [INFO] jtop.service - Initialization service
Jun 01 15:48:09 jetson-486 jtop[6673]: [INFO] jtop.service - service ready
Jun 01 15:56:41 jetson-486 jtop[6673]: [INFO] jtop.service - jtop timer thread started 1000ms
Jun 01 16:10:42 jetson-486 jtop[6673]: [INFO] jtop.service - jtop timer thread close
Jun 01 16:10:50 jetson-486 jtop[6673]: [INFO] jtop.service - jtop timer thread started 1000ms
`

I also had this very weird problem where the jtop.service would progressively not survive powercycling : on the first few reboots, it would be fine but around the 3rd reboot, i would get this message from systemd : 

`warning: the unit file, source configuration file or drop-ins of jtop.service changed on disk. run 'systemctl daemon-reload' to reload units.`

When this happens, the jtop.service file would have disappeared and then the module would not reload at next startup (because the file is missing).

Will provide journalctl logs for when it happens.

Thank you !

hanyuImg, 2023-07-04:
I set the --restart=always in docker run args,and when i reboot my ubuntu ,the container can not restart because of the same reason

SheldonWBM, 2024-06-11:
JTOP seems to randomly stop working, causing the docker container to error until the board is restarted. 
Sometimes it is stable for a few days then stops.
This is a concern as it must be always on, including overnight. So stability is important.

shrijitsingh99, 2025-01-19:
I have had a similar issue too, any solutions?

LucasJSch, 2025-03-26:
In my case, I was having the same issue, but because of the error `jtop.core.exceptions.JtopException: Lost connection with jtop server`.

I realized that `start_http_server()` calls `threading.Thread(target=httpd.serve_forever)`, which **I'm guessing** might not end the server upon an exception. And on top of that, the server runs in a different thread than the main. So the main thread is not seeing the jtop-related exceptions, and the server is continuously serving despite the exceptions. So the main process is not being terminated on these kind of exceptions.

My solution was to read in my main program the collector's output. This way, the main program ends on an exception when having issues related to jtop:

```python
if __name__ == '__main__':

    start_http_server(8000, addr='::')
    collector = CustomCollector()
    REGISTRY.register(collector)
    while True:
        time.sleep(1)
        # This generates an exception if jtop connection is lost
        next(collector.collect())
```

After the exception being thrown, my container restarted as expected.