rbonghi/jetson_stats #423: jetson-stats not supported for [L4T 32.7.4]

Deyvidone, 2023-06-22:
Please update jetson-stats with new jetpack

### Linux for Tegra

 - L4T: 32.7.4

### Jetson-Stats

 - Version: 4.2.1

Jen-Hung-Ho, 2023-06-23:
I have same issue after upgrade to L4T32.7.4.  Below is my Jetson Nano information.

sudo tegrastats
RAM 1588/3963MB (lfb 170x4MB) SWAP 0/4096MB (cached 0MB) IRAM 0/252kB(lfb 252kB) CPU [19%@1479,23%@1479,20%@1479,19%@1479] EMC_FREQ 2%@1600 APE 25 PLL@37C CPU@40C PMIC@50C GPU@39.5C AO@47C thermal@40C POM_5V_IN 2765/2384 POM_5V_GPU 0/0 POM_5V_CPU 866/536

sudo jtop --health
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/jtop/__main__.py", line 134, in main
    jtop_config()
  File "/usr/local/lib/python3.8/dist-packages/jtop/jetson_config.py", line 227, in jtop_config
    curses.wrapper(JTOPCONFIG, JTOP_MENU)
  File "/usr/lib/python3.8/curses/__init__.py", line 105, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopguiconfig.py", line 74, in __init__
    self.loop()
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopguiconfig.py", line 154, in loop
    while not self.events():
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopguiconfig.py", line 167, in events
    status_keyboard = self.keyboard(event)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopguiconfig.py", line 201, in keyboard
    output = cmd()
  File "/usr/local/lib/python3.8/dist-packages/jtop/jetson_config.py", line 69, in fix_service
    install_service(folder, copy=copy)
  File "/usr/local/lib/python3.8/dist-packages/jtop/service.py", line 124, in install_service
    copyfile(service_package_path, service_install_path)
  File "/usr/lib/python3.8/shutil.py", line 264, in copyfile
    with open(src, 'rb') as fsrc, open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.8/dist-packages/services/jtop.service'

rbonghi, 2023-06-28:
Meanwhile I fixed the missing Jetpack with the latest release **4.2.2** try to install and check again
```
sudo -H pip install -U jetson-stats
```
I investigate the other bug

closed 2024-01-04 by rbonghi