rbonghi/jetson_stats #425: jetson-stats not supported for [L4T 32.7.4]

MuizL, 2023-06-23:
Please update jetson-stats with new jetpack

### Linux for Tegra

 - L4T: 32.7.4

### Jetson-Stats

 - Version: 4.2.1

SheldonWBM, 2023-06-24:
JTOP seems to be mostly compatible with Jetpack 5.1.1 and L4T 35.3.1 however, on my machine, it currently does not seem to detect the GPU.
The (new) Jetson Power GUI (using latest tegrastats) seems to detect the GPU. 

[image]

sheldonmaschmeyer, 2023-06-24:
JTOP seems to be mostly compatible with Jetpack 5.1.1 and L4T 35.3.1
On my machine, on boot, it does not detect the GPU until `sudo systemctl restart jetson_stats.service`
[image]

JJLimmm, 2023-06-26:
Facing the same issue here. 
Have Jetpack 4.6.3 [L4T 32.7.4] on my Jetson Nano 4Gb device.

rbonghi, 2023-06-28:
Hi all,
Strange issue, I hope to check the reason.

I'm looking which Jetpack is L4T: 32.7.4. Looks not logged https://developer.nvidia.com/embedded/jetpack-archive

rbonghi, 2023-06-28:
@sheldonmaschmeyer  or @JJLimmm 

Can you attach here the jetson_clocks file (it's located in **/usr/bin/jetson_clocks**)

Raffaello

closed 2024-01-04 by rbonghi